tpaviot/oce #551: Meta-issue for OCE 0.18

dbarbier, 2015-03-11:
This issue will be fixed when OCE 0.18 is released so that this milestone can get closed.

jf---, 2015-03-12:
ETA for [OCC 7.0.0](http://www.opencascade.org/org/forum/thread_26918/?forum=3) is expected in May

spirritus, 2015-05-18:
And recently released was 6.9.0 ;-)
http://www.opencascade.org/getocc/whatsnew/

jf---, 2015-06-27:
kind of looking fwd for 0.18 -- specifically, on OSX OpenGL improvements have been made ( landing GLSL / shader support *finally* ). usually branches upstream land on OCE pretty quickly, any ideas why this is different for 6.9.0?

greyltc, 2016-06-18:
@thatcadguy,
Am I correct in assuming that you closed #613 because OCE 0.18 will support opencascade version 7.0.0?

thatcadguy, 2016-06-18:
@greyltc closed because we already have this issue open for 0.18, and because we don't really need an issue open to support the next release of OCCT. The goal of the project is to keep up-to-date with OCCT already.

mat127, 2016-07-15:
Is there any progress related to support for OCCT 7.0.0? Can I participate somehow to make it happen? I am interested mainly in the up-to-date boolean operations and fuzzy boolean operations.

jf---, 2016-07-15:
@mat127 IIRC the fuzzy bools were released in OCCT 6.8.1, so you can expect these to be present in [this branch](https://github.com/tpaviot/oce/pull/589)