tpaviot/oce #553: SO versions

cshorler, 2015-04-07:
Can the SOVERSION be set to 6.8 (rather than 10) in CMakeLists.txt by reading out the value from  src/Standard/Standard_Version.hxx (or inc/Standard_Version.hxx in OCCT)

Is there some specific reason to maintain SOVERSION 10?

(plenty of other cmake modules do this in a few lines of code).

example from libxml2
    file(STRINGS "${LIBXML2_INCLUDE_DIR}/libxml/xmlversion.h" libxml2_version_str
         REGEX "^#define[\t ]+LIBXML_DOTTED_VERSION[\t ]+\".*\"")

    string(REGEX REPLACE "^#define[\t ]+LIBXML_DOTTED_VERSION[\t ]+\"([^\"]*)\".*" "\\1"
           LIBXML2_VERSION_STRING "${libxml2_version_str}")
    unset(libxml2_version_str)

dbarbier, 2015-04-08:
Hello, this does not make sense to me, SOVERSION is bumped only when incompatible changes are introduced.  And AFAICT libxml2 is not built with CMake ;)

cshorler, 2015-04-08:
you're right!

I had spent the afternoon building OCCT 6.8.0 - after fixing the build, discovering no SOVERSION set in the library, I added one that seemed logical given the release notes (incompatible changes reflected by minor version changes at 6.5, 6.6, 6.7 and 6.8).  Then I discovered OCE was different, and wanted to understand why?

I guess since 6.8.0 doesn't have one, neither did previous releases and you just picked one?  I suppose what you're doing is only what many other libraries do, e.g. glibc libraries.

w.r.t. the cmake stuff - you've misunderstood my intent.
I copied an example of extracting version information from a header found in the cmake installation, nothing to do with building libxml2.  Now irrelevant!

closed 2015-04-13 by cshorler