tpaviot/oce #555: math_Vector error help please!!

Nesty92, 2015-05-18:
Hello everyone!!   when I try to run a example get the following error
<b>/opt/oce/include/oce/math_Vector.hxx:273: error: expected unqualified-id before numeric constant
    Opposite Standard_EXPORT math_Vector ();</b>
with the libs compiled by the team freecad  same thing happens  any idea that it can be thank you very much

dbarbier, 2015-05-19:
Hello,

Maybe something defines a cpp macro named Opposite to a numeric value (likely -1), there is not much we can do.

Nesty92, 2015-05-19:
Thank you for responding this happens me to use the example occQt https://github.com/eryar/occQt available in linuxmint 17 based on ubuntu 14.04

dbarbier, 2015-05-19:
Okay, `X11/Xlib.h` and `X11/X.h` define a bunch of macros with generic names which conflict with OCC methods (Opposite, but also Status, and maybe others).
You can try to add
```
#undef Status
#undef Opposite
```
between Qt includes and OCC ones; does anyone have a better idea?

Nesty92, 2015-05-19:
Thank you for answering then I try your suggestion and posting the results

Nesty92, 2015-05-19:
anyone have any other idea?

Nesty92, 2015-05-21:
@dbarbier
#undef Opposite this is a solution!!! thanks you  ;)

dbarbier, 2015-05-21:
Glad it worked for you.  This is not ideal, but unfortunately I do not see how to fix that on our side.

closed 2015-05-21 by dbarbier