tpaviot/oce #559: oce does not build on ppc64le aarch

leitao, 2015-06-03:
Currently oce does not pass tests on new POWER architecture with the following error:

Errors while running CTest
48/48 Test #28: BRepMeshTestSuite.testMeshTorus .................   Passed    2.61 sec

98% tests passed, 1 tests failed out of 48

Total Test time (real) =   3.15 sec

The following tests FAILED:
	 24 - BRepOffsetAPITestSuite.testEvolvedShape (Failed)
make[1]: *** [test] Error 8
dh_auto_test: make -j8 test ARGS+=-j8 returned exit code 2
make: *** [build-arch] Error 2
Makefile:140: recipe for target 'test' failed
make[1]: Leaving directory '/«PKGBUILDDIR»/obj-powerpc64le-linux-gnu'
debian/rules:14: recipe for target 'build-arch' failed

We would like to have it ported and full building in ppc64el architecture

edelsohn, 2015-06-04:
IBM has created a $200 bounty on Bountysource to fix the failing test on PPC64LE Linux.
https://www.bountysource.com/issues/20145612-oce-does-not-build-on-ppc64le-aarch

ansani, 2015-07-21:
Are you interested in a specific distro ?
I'm working on Debian 8 or Ubuntu 15.04

hernan-erasmo, 2015-07-29:
I've been playing with this for the last few days, and I'm having trouble understanding what the problem is exactly.

Following the instructions in [INSTALL.Unix](https://github.com/tpaviot/oce/blob/master/INSTALL.Unix) I'm able to run the following commands without any error:
- Creating a build dir, and running **cmake ..** inside it
- Running **make**
- Running **make install**

When it comes down to testing however, running **make test** gives me the following results:

```
hernan@debianle:~/desarrollo/oce/build$ make test
Running tests...
Test project /home/desarrollo/oce/build
      Start  1: BRepOffsetAPITestSuite.testEvolvedShape
 1/48 Test  #1: BRepOffsetAPITestSuite.testEvolvedShape .........***Failed    0.03 sec
      Start  2: BRepOffsetAPITestSuite.testOffsetBox
 2/48 Test  #2: BRepOffsetAPITestSuite.testOffsetBox ............   Passed    0.04 sec
      Start  3: IGESImportTestSuite.testImportIGES_1
 3/48 Test  #3: IGESImportTestSuite.testImportIGES_1 ............   Passed    0.36 sec
      Start  4: gpTestSuite.testgp_PntConstruct
 4/48 Test  #4: gpTestSuite.testgp_PntConstruct .................   Passed    0.01 sec
      Start  5: gpTestSuite.testgp_PntDistance
 5/48 Test  #5: gpTestSuite.testgp_PntDistance ..................   Passed    0.01 sec
      Start  6: gpTestSuite.testBuildThousandsPoints
 6/48 Test  #6: gpTestSuite.testBuildThousandsPoints ............   Passed    0.01 sec
      Start  7: gpTestSuite.testGCC42gp_Trsf
 7/48 Test  #7: gpTestSuite.testGCC42gp_Trsf ....................   Passed    0.01 sec
      Start  8: OCAFExportTestSuite.testExportAscii
 8/48 Test  #8: OCAFExportTestSuite.testExportAscii .............   Passed    0.04 sec
      Start  9: OCAFExportTestSuite.testExportNonAscii
 9/48 Test  #9: OCAFExportTestSuite.testExportNonAscii ..........   Passed    0.03 sec
      Start 10: OCAFExportTestSuite.testOverflow
10/48 Test #10: OCAFExportTestSuite.testOverflow ................   Passed    0.03 sec
      Start 11: BRepMeshTestSuite.testMeshBox
11/48 Test #11: BRepMeshTestSuite.testMeshBox ...................   Passed    0.02 sec
      Start 12: BRepMeshTestSuite.testMeshSphere
12/48 Test #12: BRepMeshTestSuite.testMeshSphere ................   Passed    0.26 sec
      Start 13: BRepMeshTestSuite.testMeshTorus
13/48 Test #13: BRepMeshTestSuite.testMeshTorus .................   Passed    3.45 sec
      Start 14: BRepImportTestSuite.testImportBRep1
14/48 Test #14: BRepImportTestSuite.testImportBRep1 .............   Passed    0.33 sec
      Start 15: BRepImportTestSuite.testImportBRep2
15/48 Test #15: BRepImportTestSuite.testImportBRep2 .............   Passed    0.03 sec
      Start 16: BRepImportTestSuite.testImportBRep3
16/48 Test #16: BRepImportTestSuite.testImportBRep3 .............   Passed    0.03 sec
      Start 17: BRepImportTestSuite.testImportBRep4
17/48 Test #17: BRepImportTestSuite.testImportBRep4 .............   Passed    0.05 sec
      Start 18: TCollectionTestSuite.testSequenceOfPnt
18/48 Test #18: TCollectionTestSuite.testSequenceOfPnt ..........   Passed    0.13 sec
      Start 19: Standard_StdAllocatorTestSuite.testTraits
19/48 Test #19: Standard_StdAllocatorTestSuite.testTraits .......   Passed    0.01 sec
      Start 20: Standard_StdAllocatorTestSuite.testContainers
20/48 Test #20: Standard_StdAllocatorTestSuite.testContainers ...   Passed    0.01 sec
      Start 21: BRepExtremaTestSuite.testDistSphereSphere
21/48 Test #21: BRepExtremaTestSuite.testDistSphereSphere .......   Passed    0.01 sec
      Start 22: BRepExtremaTestSuite.testDistSphereVertex
22/48 Test #22: BRepExtremaTestSuite.testDistSphereVertex .......   Passed    0.01 sec
      Start 23: BRepExtremaTestSuite.testDistVertexEdge
23/48 Test #23: BRepExtremaTestSuite.testDistVertexEdge .........   Passed    0.01 sec
      Start 24: BRepExtremaTestSuite.testDistEdgeVertex
24/48 Test #24: BRepExtremaTestSuite.testDistEdgeVertex .........   Passed    0.01 sec
      Start 25: BRepExtremaTestSuite.testDistVertexFace
25/48 Test #25: BRepExtremaTestSuite.testDistVertexFace .........   Passed    0.01 sec
      Start 26: BRepExtremaTestSuite.testDistFaceVertex
26/48 Test #26: BRepExtremaTestSuite.testDistFaceVertex .........   Passed    0.01 sec
      Start 27: BRepGPropTestSuite.testComputeBoxVolume
27/48 Test #27: BRepGPropTestSuite.testComputeBoxVolume .........   Passed    0.01 sec
      Start 28: BRepGPropTestSuite.testComputeBoxSurface
28/48 Test #28: BRepGPropTestSuite.testComputeBoxSurface ........   Passed    0.01 sec
      Start 29: BRepGPropTestSuite.testComputeSphereVolume
[... 29 lines cut ...]
      Start 44: TestSuite.testNullPointer
44/48 Test #44: TestSuite.testNullPointer .......................   Passed    0.00 sec
      Start 45: TestSuite.testFloatEq
45/48 Test #45: TestSuite.testFloatEq ...........................   Passed    0.00 sec
      Start 46: TestSuite.testFloatNeq
46/48 Test #46: TestSuite.testFloatNeq ..........................   Passed    0.00 sec
      Start 47: TestSuite.testBoolean
47/48 Test #47: TestSuite.testBoolean ...........................   Passed    0.00 sec
      Start 48: TestSuite.testIntegerLighter
48/48 Test #48: TestSuite.testIntegerLighter ....................   Passed    0.00 sec

98% tests passed, 1 tests failed out of 48

Total Test time (real) =   6.46 sec

The following tests FAILED:
          1 - BRepOffsetAPITestSuite.testEvolvedShape (Failed)
Errors while running CTest
Makefile:137: recipe for target 'test' failed
make: *** [test] Error 8
```

Note that while the failing test is the same, the error message and number is different from the one @leitao posted.

I'm able to _build and install_ oce on my setup (with ppc64le arch), but this test keeps failing. Is it right to assume that this issue will be closed when the test is fixed, then?

edelsohn, 2015-07-29:
Are you consistently seeing the same (different) failure or does oce run the tests in a randomized order?  Is the real problem some state left by a previous test?

We would like to achieve clean testsuite on ppc64le.

hernan-erasmo, 2015-07-29:
Yes, I'm consistenly seeing the following error:
```
The following tests FAILED:
          1 - BRepOffsetAPITestSuite.testEvolvedShape (Failed)
Errors while running CTest
Makefile:137: recipe for target 'test' failed
make: *** [test] Error 8
```

No randomized order. I've already performed the build proces a couple of times (with and without the flag -DOCE_TESTING:BOOL=ON) and the test order is always the same.

I doubt it fails because a previous test leaving the environment unstable. At least in my case, BRepOffsetAPITestSuite.testEvolvedShape is always the first test that runs (notice that there's another test inside the same suit, and it passes).

The most I've been able to do is track down [the place where it blows up to this line](https://github.com/tpaviot/oce/blob/master/test/BRepOffsetAPI_test/BRepOffsetAPI_test.cpp#L19).
I know that because I've been modifying, rebuilding and running the tests several times, commenting out every line and replacing
```
ASSERT_TRUE(S.IsDone());
```
with
```
ASSERT_EQ (1,1);
```
on each run.

If the test is run with this code, then it passes:
```
TEST(BRepOffsetAPITestSuite, testEvolvedShape)
{
    BRepBuilderAPI_MakePolygon P;
    P.Add(gp_Pnt(0.,0.,0.));
    P.Add(gp_Pnt(200.,0.,0.));
    P.Add(gp_Pnt(200.,200.,0.));
    P.Add(gp_Pnt(0.,200.,0.));
    P.Add(gp_Pnt(0.,0.,0.));
    BRepBuilderAPI_MakePolygon wprof(gp_Pnt(0.,0.,0.),gp_Pnt(-60.,-60.,-200.));
    //BRepOffsetAPI_MakeEvolved S(P.Wire(),wprof.Wire(),GeomAbs_Arc,Standard_True,Standard_False,Standard_True,0.0001);
    //S.Build();
    //ASSERT_TRUE(S.IsDone());
    ASSERT_EQ(1,1);
}
```
(notice all the commented lines)

And if you change that code just a little, like this:
```
TEST(BRepOffsetAPITestSuite, testEvolvedShape)
{
    BRepBuilderAPI_MakePolygon P;
    P.Add(gp_Pnt(0.,0.,0.));
    P.Add(gp_Pnt(200.,0.,0.));
    P.Add(gp_Pnt(200.,200.,0.));
    P.Add(gp_Pnt(0.,200.,0.));
    P.Add(gp_Pnt(0.,0.,0.));
    BRepBuilderAPI_MakePolygon wprof(gp_Pnt(0.,0.,0.),gp_Pnt(-60.,-60.,-200.));
    BRepOffsetAPI_MakeEvolved S(P.Wire(),wprof.Wire(),GeomAbs_Arc,Standard_True,Standard_False,Standard_True,0.0001);
    //S.Build();
    //ASSERT_TRUE(S.IsDone());
    ASSERT_EQ(1,1);
}
```
and run only this test with the following command
```
env GTEST_CATCH_EXCEPTIONS=0 ctest -R testEvolvedShape -V
```

then you get
```
hernan@debianle:~/desarrollo/oce/build$ env GTEST_CATCH_EXCEPTIONS=0 ctest -R testEvolvedShape -V
UpdateCTestConfiguration  from :/home/hernan/desarrollo/oce/build/DartConfiguration.tcl
Parse Config file:/home/hernan/desarrollo/oce/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/hernan/desarrollo/oce/build/DartConfiguration.tcl
Parse Config file:/home/hernan/desarrollo/oce/build/DartConfiguration.tcl
Test project /home/hernan/desarrollo/oce/build
Constructing a list of tests
Done constructing a list of tests
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: BRepOffsetAPITestSuite.testEvolvedShape

1: Test command: /home/hernan/desarrollo/oce/build/test/BRepOffsetAPI_test/BRepOffsetAPI_test "--gtest_filter=BRepOffsetAPITestSuite.testEvolvedShape"
1: Test timeout computed to be: 1500
1: Note: Google Test filter = BRepOffsetAPITestSuite.testEvolvedShape
1: [==========] Running 1 test from 1 test case.
1: [----------] Global test environment set-up.
1: [----------] 1 test from BRepOffsetAPITestSuite
1: [ RUN      ] BRepOffsetAPITestSuite.testEvolvedShape
1: terminate called after throwing an instance of 'Standard_NotImplemented'
1/1 Test #1: BRepOffsetAPITestSuite.testEvolvedShape ...***Exception: Other  0.03 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.04 sec

The following tests FAILED:
          1 - BRepOffsetAPITestSuite.testEvolvedShape (OTHER_FAULT)
Errors while running CTest

```
So appearently, calling 
```
BRepOffsetAPI_MakeEvolved S(P.Wire(),wprof.Wire(),GeomAbs_Arc,Standard_True,Standard_False,Standard_True,0.0001);
``` 
throws an Standard_NotImplemented exception.

I still couldn't figure out why, but I believe that's what causing the test to fail.

edelsohn, 2015-07-29:
Can you find out where it's failing within the oce runtime for the BRepOffsetAPI_MakeEvolved call?  This implies that something is not implemented in the Power code, which could be a configuration failure not finding something in the math library.

hernan-erasmo, 2015-07-31:
I'm probably reaching the limits of my technical skills right now. I'll keep trying but in the meantime here's what I could find out so far:

I tried to debug the failing test using gdb. Set a breakpoint [on this line](https://github.com/tpaviot/oce/blob/master/test/BRepOffsetAPI_test/BRepOffsetAPI_test.cpp#L19) and when I tried to step in, the following occurred:

```
Program received signal SIGSEGV, Segmentation fault.
0x00003fffb696b5c0 in vtable for TopoDS_TCompound ()
   from /home/hernan/desarrollo/oce/build/Unix/ppc64le-Release-64/libTKBRep.so.10
```

At that point I used the bt full command (shows the backtrace of the current thread), and this is the output:
```
Starting program: /home/hernan/desarrollo/oce/build/test/BRepOffsetAPI_test/BRepOffsetAPI_test 
warning: Could not load shared library symbols for linux-vdso64.so.1.
Do you need "set solib-search-path" or "set sysroot"?
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/powerpc64le-linux-gnu/libthread_db.so.1".
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.
dl-debug.c:74: No such file or directory.

Breakpoint 1, BRepOffsetAPITestSuite_testEvolvedShape_Test::TestBody (this=<optimized out>)
    at /home/hernan/desarrollo/oce/test/BRepOffsetAPI_test/BRepOffsetAPI_test.cpp:19

Program received signal SIGSEGV, Segmentation fault.
0x00003fffb696b5c0 in vtable for TopoDS_TCompound () from /home/hernan/desarrollo/oce/build/Unix/ppc64le-Release-64/libTKBRep.so.10
#0  0x00003fffb696b5c0 in vtable for TopoDS_TCompound () from /home/hernan/desarrollo/oce/build/Unix/ppc64le-Release-64/libTKBRep.so.10
No symbol table info available.
#1  0x00000000100194b4 in ~Handle_Standard_Transient (this=0x3fffffffea80, __in_chrg=<optimized out>)
    at /home/hernan/desarrollo/oce/src/Standard/Handle_Standard_Transient.hxx:71
No locals.
#2  ~Handle_MMgt_TShared (this=0x3fffffffea80, __in_chrg=<optimized out>) at /home/hernan/desarrollo/oce/inc/Handle_MMgt_TShared.hxx:18
No locals.
#3  ~Handle_TopLoc_SListNodeOfItemLocation (this=0x3fffffffea80, __in_chrg=<optimized out>)
    at /home/hernan/desarrollo/oce/inc/Handle_TopLoc_SListNodeOfItemLocation.hxx:18
No locals.
#4  ~TopLoc_SListOfItemLocation (this=0x3fffffffea80, __in_chrg=<optimized out>) at /home/hernan/desarrollo/oce/inc/TopLoc_SListOfItemLocation.hxx:67
No locals.
#5  ~TopLoc_Location (this=0x3fffffffea80, __in_chrg=<optimized out>) at /home/hernan/desarrollo/oce/inc/TopLoc_Location.hxx:28
No locals.
#6  ~TopoDS_Shape (this=0x3fffffffea78, __in_chrg=<optimized out>) at /home/hernan/desarrollo/oce/inc/TopoDS_Shape.hxx:43
No locals.
#7  ~TopoDS_Edge (this=0x3fffffffea78, __in_chrg=<optimized out>) at /home/hernan/desarrollo/oce/inc/TopoDS_Edge.hxx:24
No locals.
#8  ~BRepLib_MakePolygon (this=0x3fffffffe9f0, __in_chrg=<optimized out>) at /home/hernan/desarrollo/oce/inc/BRepLib_MakePolygon.hxx:39
No locals.
#9  BRepBuilderAPI_MakePolygon::~BRepBuilderAPI_MakePolygon (this=0x3fffffffe9b8, __in_chrg=<optimized out>)
    at /home/hernan/desarrollo/oce/inc/BRepBuilderAPI_MakePolygon.hxx:40
No locals.
[... 96 lines cut ...]
        S = <incomplete type>
#8  0x000000001004dd04 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) ()
No symbol table info available.
#9  0x00000000100406a0 in testing::Test::Run() ()
No symbol table info available.
#10 0x0000000010040958 in testing::TestInfo::Run() [clone .part.375] ()
No symbol table info available.
#11 0x0000000010040b94 in testing::TestCase::Run() [clone .part.376] ()
No symbol table info available.
#12 0x00000000100414b4 in testing::internal::UnitTestImpl::RunAllTests() [clone .part.377] ()
No symbol table info available.
#13 0x0000000010041828 in testing::UnitTest::Run() ()
No symbol table info available.
#14 0x0000000010019090 in RUN_ALL_TESTS () at /home/hernan/desarrollo/oce/test/gtest-1.7.0/include/gtest/gtest.h:2288
No locals.
#15 0x0000000010017e40 in main (argc=1, argv=0x3ffffffff2a8) at /home/hernan/desarrollo/oce/test/BRepOffsetAPI_test/BRepOffsetAPI_test.cpp:35
No locals.
Single stepping until exit from function _ZTV18Draft_Modification,
which has no line number information.

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.
```

Hope this helps.

edelsohn, 2015-07-31:
Does the testcase fail if you build the package without optimization?  -O0 -g.  It's a lot easier to debug without optimization.

hernan-erasmo, 2015-07-31:
Yes. When building without optimization (-O0) the test case still fails.

I'm sorry, perhaps my last post was too long and wasn't clear enough. I posted two gdb outputs, the first one corresponds to the test case compiled with -O3, and the second with -O0.

edelsohn, 2015-07-31:
The second one still shows frames that state "optimized out", so it cannot be -O0.

slp, 2015-10-07:
Definitely, this looks like a compiler bug (I'm running *g++ (Ubuntu 4.9.2-10ubuntu13) 4.9.2*). The conflicting chunk of code is at https://github.com/tpaviot/oce/blob/master/src/Adaptor3d/Adaptor3d_SurfaceOfRevolution.cxx#L550

```
  switch ( myBasisCurve->GetType()) {
  case GeomAbs_Line:    {
    const gp_Ax1& Axe = (myBasisCurve->Line()).Position();
    
    if (myAxis.IsParallel(Axe, TolAng)) {
```

Here, when building with optimizations (that is, every -O* value except -O0), the call to Position() gets somehow optimized out, and Axe ends uninitialized.

As a workaround, both changing the code to use a direct value instead of a reference:

```
-    const gp_Ax1& Axe = (myBasisCurve->Line()).Position();
+    const gp_Ax1 Axe = (myBasisCurve->Line()).Position();
```

or referencing gp_Lin first:

```
-    const gp_Ax1& Axe = (myBasisCurve->Line()).Position();
+    const gp_Lin& Lin = myBasisCurve->Line();
+    const gp_Ax1& Axe = Lin.Position();
```

will do the trick.

If you would consider applying one of those workarounds, please let me know and I'll generate the corresponding PR.

a-betenev, 2015-10-07:
Hello,

Thank you for reporting your findings!
This is definitely not a compiler bug, but bad code: here, 
myBasisCurve->Line() returns temporary object (gp_Lin), and its method 
Position() returns reference to its field.
Thus, Axe is a reference to field of temporary object that gets deleted 
immediately.
I have registered issue in OCCT bug tracker on this, with ID #26755 
(http://tracker.dev.opencascade.org/view.php?id=26755)

Andrey

slp, 2015-10-07:
@a-betenev You're right, the fact that this issue didn't appear when running with other compilers/platforms confused me. So, using a direct value instead of a reference is the way to go, isn't it? Should I summit a PR?

tpaviot, 2015-10-08:
@sip Yes please submit a PR if you want this bug to be fixed and merged into master before the next release.

PR #582 (2015-10-10): Replace gp_Ax1 reference with direct value. Fixes #559
Replace gp_Ax1 reference with direct value. Fixes #559
commit b7acce7f (2015-10-10): Merge pull request #582 from slp/master

Replace gp_Ax1 reference with direct value. Fixes #559
commit e48e1bce (2015-10-10): Adaptor3d_SurfaceOfRevolution.cxx: Replace gp_Ax1 reference with direct value. Fixes #559
closed 2015-10-10 by tpaviot