tpaviot/oce #561: Upgrade to OCCT 6.9

fw0stahl, 2015-06-24:
Hello,

i'm a FreeCAD user and have a big problem with the export of round parts to STEP. So there are many improvements in occt 6.9 for the export. We need the OCE edition :-) maybe the problem could be solved... 

Post with the issue of step exp.: http://forum.freecadweb.org/viewtopic.php?f=3&t=11420

Thank you for the great work!

tpaviot, 2015-10-10:
We recently had a chat about moving up oce to latest upstream release. See https://gitter.im/tpaviot/oce. Path to oce support for occt691 is a work-in-progress.

blobfish, 2015-10-10:
I have a branch that can be found at:
https://github.com/blobfish/oce/tree/691patched

It seems to be functioning good. It passes the tests* and I loaded a
pretty complicated model in FreeCAD and did a full recompute without
incidence. I have attached my build and test output.

If we want to run with this branch, I think we can 'back port' these
changes into a patch set without too much trouble. I don't want to do it
for fun, though.

Some notes:
don't need. occt already fixed: d06c63e Fix Standard_Construction_Error
exception in BOP

don't need. occt removed function: be2e9b3 Disambiguate gp_Pnt
constructor when copying a Select3D_Pnt instance

need to prove out resource manager change: a35a11b
get-rid-of-environment-variables

don't need. occt already fixed: 759dc8e In Standard_MMgrFactory class,
make myFMMgr private, it does not have to be public

isCorrectPolyData is gone: 5bda7cf submitted/thread_20105/add-openmp-support

* Apparently the OCAF export tests either pass or fail dependent on
whether oce is installed or not????

jf---, 2015-10-12:
Cool @blobfish ... amazing to all tests passing on travis, awesome... :+1: 

I've managed to built on osx 10.10 with clang++ 6.1.0
There's a fair number of compiler warning showing up ( a lot more, but all the same error on dereferenced null pointer ).  See warnings at the end of this post.

This does seem to have an effect, since the test suite fails pretty hard ( see at the end of the post ).

Happy OCE is moving towards 6.9.1 / 0.18 :sunny: 

### compiler warnings

#### dereferenced null pointer

---

/Users/jelleferinga/miniconda/conda-bld/work/src/BSplSLib/BSplSLib.cxx:2441:8: warning: reference cannot be bound to
      dereferenced null pointer in well-defined C++ code; comparison may be assumed to always evaluate to true
      [-Wtautological-undefined-compare]
  if (&WeightsArray != NULL) {
       ^~~~~~~~~~~~    ~~~~
/Users/jelleferinga/miniconda/conda-bld/work/src/BSplSLib/BSplSLib.cxx:2570:8: warning: reference cannot be bound to
      dereferenced null pointer in well-defined C++ code; comparison may be assumed to always evaluate to true
      [-Wtautological-undefined-compare]
  if (&WeightsArray != NULL) {
       ^~~~~~~~~~~~    ~~~~
/Users/jelleferinga/miniconda/conda-bld/work/src/BSplSLib/BSplSLib_BzSyntaxes.cxx:71:10: warning: reference cannot be
      bound to dereferenced null pointer in well-defined C++ code; comparison may be assumed to always evaluate to
      false [-Wtautological-undefined-compare]
    if (&Weights == NULL) {
         ^~~~~~~    ~~~~
1 warning generated.

---

#### A nr of compiler warnings when compiling OpenGL:

/Users/jelleferinga/miniconda/conda-bld/work/src/OpenGl/OpenGl_View_2.cxx:671:8: warning: reference cannot be bound
      to dereferenced null pointer in well-defined C++ code; comparison may be assumed to always evaluate to false
      [-Wtautological-undefined-compare]
  if (&ACLayer == NULL
       ^~~~~~~    ~~~~
1 warning generated.
/Users/jelleferinga/miniconda/conda-bld/work/src/OpenGl/OpenGl_View_Raytrace.cxx:380:39: warning: comparison of array
      'theStructure->Transformation()->mat' not equal to a null pointer is always true
      [-Wtautological-pointer-compare]
  if (theStructure->Transformation()->mat != NULL)
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~    ~~~~
/Users/jelleferinga/miniconda/conda-bld/work/src/OpenGl/OpenGl_View_Raytrace.cxx:392:37: warning: address of array
      'theStructure->Transformation()->mat' will always evaluate to 'true' [-Wpointer-bool-conversion]
    theStructure->Transformation()->mat ? aStructTransform : NULL, theGlContext);
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~ ~
/Users/jelleferinga/miniconda/conda-bld/work/src/OpenGl/OpenGl_View_Raytrace.cxx:400:39: warning: address of array
      'theStructure->Transformation()->mat' will always evaluate to 'true' [-Wpointer-bool-conversion]
      theStructure->Transformation()->mat ? aStructTransform : NULL, theGlContext);
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~ ~

---
#### OCE test suite ouput

10% tests passed, 43 tests failed out of 48

Total Test time (real) =   3.38 sec

The following tests FAILED:
	  1 - BRepAlgoAPITestSuite.testCutBox (OTHER_FAULT)
	  2 - BRepAlgoAPITestSuite.testCutCylSphere (OTHER_FAULT)
	  3 - BRepExtremaTestSuite.testDistSphereSphere (OTHER_FAULT)
	  4 - BRepExtremaTestSuite.testDistSphereVertex (OTHER_FAULT)
	  5 - BRepExtremaTestSuite.testDistVertexEdge (OTHER_FAULT)
	  6 - BRepExtremaTestSuite.testDistEdgeVertex (OTHER_FAULT)
	  7 - BRepExtremaTestSuite.testDistVertexFace (OTHER_FAULT)
	  8 - BRepExtremaTestSuite.testDistFaceVertex (OTHER_FAULT)
	  9 - BRepGPropTestSuite.testComputeBoxVolume (OTHER_FAULT)
	 10 - BRepGPropTestSuite.testComputeBoxSurface (OTHER_FAULT)
	 11 - BRepGPropTestSuite.testComputeSphereVolume (OTHER_FAULT)
	 12 - BRepImportTestSuite.testImportBRep1 (OTHER_FAULT)
	 13 - BRepImportTestSuite.testImportBRep2 (OTHER_FAULT)
	 14 - BRepImportTestSuite.testImportBRep3 (OTHER_FAULT)
	 15 - BRepImportTestSuite.testImportBRep4 (OTHER_FAULT)
	 16 - BRepMeshTestSuite.testMeshBox (OTHER_FAULT)
	 17 - BRepMeshTestSuite.testMeshSphere (OTHER_FAULT)
	 18 - BRepMeshTestSuite.testMeshTorus (OTHER_FAULT)
	 19 - BRepOffsetAPITestSuite.testEvolvedShape (OTHER_FAULT)
	 20 - BRepOffsetAPITestSuite.testOffsetBox (OTHER_FAULT)
	 21 - BRepPrimAPITestSuite.testCreateBox (OTHER_FAULT)
	 22 - BRepPrimAPITestSuite.testCreateFlatBox (OTHER_FAULT)
	 23 - BRepPrimAPITestSuite.testCreateSmallestBox (OTHER_FAULT)
	 24 - gpTestSuite.testgp_PntConstruct (OTHER_FAULT)
	 25 - gpTestSuite.testgp_PntDistance (OTHER_FAULT)
	 26 - gpTestSuite.testBuildThousandsPoints (OTHER_FAULT)
	 27 - gpTestSuite.testGCC42gp_Trsf (OTHER_FAULT)
	 28 - IGESImportTestSuite.testImportIGES_1 (OTHER_FAULT)
	 29 - OCAFExportTestSuite.testExportAscii (OTHER_FAULT)
	 30 - OCAFExportTestSuite.testExportNonAscii (OTHER_FAULT)
	 31 - OCAFExportTestSuite.testOverflow (OTHER_FAULT)
	 32 - SortToolsTestSuite.testQuickSortOfInteger (OTHER_FAULT)
	 33 - Standard_StdAllocatorTestSuite.testTraits (OTHER_FAULT)
	 34 - Standard_StdAllocatorTestSuite.testContainers (OTHER_FAULT)
	 35 - STEPImportTestSuite.testImportAP203_1 (OTHER_FAULT)
	 36 - STEPImportTestSuite.testImportAP203_2 (OTHER_FAULT)
	 37 - STEPImportTestSuite.testImportAP214_1 (OTHER_FAULT)
	 38 - STEPImportTestSuite.testImportAP214_2 (OTHER_FAULT)
	 39 - STEPImportTestSuite.testImportAP214_3 (OTHER_FAULT)
	 40 - TCollectionTestSuite.testSequenceOfPnt (OTHER_FAULT)
	 41 - TDataXtdTestSuite.testPattern (OTHER_FAULT)
	 42 - UnitsAPITestSuite.testCurrentUnits (OTHER_FAULT)
	 43 - UnitsAPITestSuite.testMDTVCurrentUnits (OTHER_FAULT)

blobfish, 2015-10-12:
I recompiled with clang and can confirm your warnings. I see that some
of the "tricks" I used to silence gcc didn't work with clang. However, I
CAN'T confirm the failing tests. Mine still all pass. Assuming I am
running them correctly. It wasn't obvious too me how to run them all at
once, so I launched them with "find . -type f -iname "*_test" -exec {}
\;" Build and test results attached for reference.

IMHO: the warning you point out([-Wtautological-pointer-compare]) is a
good warning. This should be addressed. By who and how, is a bigger
discussion with more participants.

jf---, 2015-10-12:
Hi @blobfish , correct, the failing tests are due to a poor choice of CMake options, failing to link to libTkKernel ( that's why I deleted the post... ). 

On OSX these warnings lead to failing tests, see #576 
There is [OCCT patch upstream](http://tracker.dev.opencascade.org/view.php?id=26042) as @a-betenev pointed out

tpaviot, 2015-10-13:
@blobfish Are you sure you applied all patches ? I have the feeling that recent updates are not available. For instance, the appveyor build is not available from this branch, this would be very helpfull in order to check compilation on Windows.

tpaviot, 2015-10-13:
@jf--- Travis build is related to master branch of tpaviot/oce repository, not @blobfish fork. @blobfish could you please create a pull request for this branch, just to trigger a travis build ?

dbarbier, 2015-10-13:
@blobfish  Great work, thanks.  But I do not understand how you got these commits, did you use `git am` instead of quilt?  With quilt, you upgrade patches and commit them into the OCCT-patches branch.

I will take care of adding our changes posterior to f6d3095.

blobfish, 2015-10-13:
Thomas,
	I am sure you are right. I have only applied the changes from the
patches branch.

blobfish, 2015-10-13:
Yeah, I will see if I can fumble through it. I use github as little as I
can get away with.

blobfish, 2015-10-13:
Denis,
	Sorry, I went a different path. I just couldn't accept the idea of
editing diffs. Here is basically what I did.

set up quilt, per your instructions.
created a new branch called 680patched from 680official.
ran 'git quiltimport'. this builds a commit for each patch.
updated official branch to 691, per your instructions.
created a new branch called 691patched from 691official.
git cherry-pick each commit from 680patched and fixed any conflicts.

the quiltimport ran smooth except for the fact that it wasn't
recognizing the author of the commit. So I was doing a lot of copy/paste
during the operation. I want to double check the commit authors, if we
decide to use these commits in some capacity.

My first try I used git rebase --onto instead of the cherry-picks. This
was a little too automated and I made some mistakes. Second try I used
cherry-picks and was pretty happy with the outcome, but wasn't automated
enough for me. Next time, I want to see if it is possible to do
something like git rebase --onto --interactive and have a chance to
inspect each commit before it happens.

I did a very little research and I think it would be possible to get the
changes back to a patch set without too much head-ache. Maybe
http://linux.die.net/man/7/guilt

dbarbier, 2015-10-14:
@blobfish No problem, experimenting new ways is always a good idea.

FWIW I tried topgit and stgit years ago and found that, after few months, git commit tree was utterly cluttered by patches history, this is why I kept using quilt.

blobfish, 2015-10-15:
Denis,
	I have no experience with topgit or stgit, but I think your experience
is tied to git itself and not the GUI's. Trying to keep a clean
history(git rebase -i) with public branches can be challenging for all
involved parties. Just to be clear, I am not trying to change any
established process. If/when more parties weigh in and the consensus is
my branch will work, I will invest more time to get it back to patches.
I have no expectations. I will be at peace if my branch 'dies on the
vine'. Thanks.

thatcadguy, 2015-12-14:
Related to #583 and #587

thatcadguy, 2015-12-18:
Ad #589

greyltc, 2016-04-13:
7.0.0 is out. Can we hope for oce to be updated to use that release?

jf---, 2016-04-13:
@greyltc would you please start another issue to discuss that? this thread is 6.9.0 specific
thanks!

greyltc, 2016-04-13:
Sorry. I put this here because I thought maybe the new release made this issue obsolete.

jf---, 2016-04-13:
no worries, porting to 6.9 is still ongoing. 7.0.0 might be a pretty disruptive release, for application maintainers, so it will take some time for projects to adapt.

thatcadguy, 2016-11-05:
Just merged 6.9.1 branch, closing

closed 2016-11-05 by thatcadguy