tpaviot/oce #564: Building OCE as a static library fails on VS 2013

Stinkfist0, 2015-07-20:
Building OCE as a static library on VS 2013(.4) and Windows SDK 8.1 fails due to unfortunate precompiler definitions leaking from `windows.h`.

As a quick fix one can do the following additions:
in inc\Visual3d_Layer.hxx:
```c
#ifdef DrawText
#undef DrawText
#endif
```

in src\AIS\AIS_ConcentricRelation.cxx:
```c
#ifdef rad1
#undef rad1
#endif
#ifdef rad2
#undef rad2
#endif
```

in src\Voxel\Voxel_Writer.cxx:
```c
#ifdef small
#undef small
#endif
```

I think a good proper fix could be having a file similar to this:
https://github.com/juj/MathGeoLib/blob/master/src/Math/InclWindows.h
In this file the necessary precautions could be taken (defining WIN32_MEAN_AND_LEAN, NOMINMAX, NOGDI etc. before including `windows.h` and undefining possible code breaking definitions (DrawText. small, arg1 and arg2 in this case). This file would be used to include windows.h included throughout the whole codebase.

QbProg, 2015-07-24:
Thanks I'm fixing it this night in the branch qb/improvements
When I'm ready I'll ask you to kindly test it and give a feedback :)

Stinkfist0, 2015-07-27:
Great!

Stinkfist0, 2015-08-18:
FYI, these same errors occur also on VS 2015. VS 2015 dynamic lib build also has some problems, I will provide a PR to fix these.

QbProg, 2015-08-19:
The branch qb/improvements should already contain some fixes, you could
cherry-pick them.

tpaviot, 2015-08-19:
@QbProg If you consider some changes could be of interest, could you please rebase qb/improvements and create a pull request ?

QbProg, 2015-08-19:
The main purpose of that branch was to separate the the bundle from the OCE
CMakeLists.txt and make sure it works in all windows config variants. It's
almost ready I need to finish mingw support which takes forever to build
and merge some commits. Also I would like to integrate samples and enable
vtk, but these could be done in a second PR.
Also, I would like to enable the tcl test suite that doesn't run with the
current approach
Qb

Stinkfist0, 2015-08-26:
Also one notable warning that can be fixed e.g. by using `#define NOGDI`:
```
oce\src\Resource\Resource_Manager.cxx(42): warning C4005: 'ERROR': macro redefinition
C:\Program Files (x86)\Windows Kits\8.1\Include\um\wingdi.h(114): note: see previous definition of 'ERROR' (compiling source file C:\Dev\GitHub\Tridify\oce\src\Resource\Resource_Manager.cxx)
```

QbProg, 2015-09-21:
Fixed by PR #575

closed 2015-09-21 by QbProg

Stinkfist0, 2015-09-22:
Great!