tpaviot/oce #566: OCE Binaries - Broken Links

lribeirogomes, 2015-08-17:
While trying to access OCE binaries, I've found out both links leads to error pages:
 - http://code.google.com/p/oce-binaries/downloads/ -> "Your client does not have permission to get URL /p/oce-binaries/downloads/list from this server."
 - http://bit.ly/1kvDoTN -> "Not Found"

I have no idea if that is intentional or not, but both of them are misleading me into accessing something that doesn't exist. Could this issue be fixed or, at least, clarified and removed?

jf---, 2015-08-17:
Hi @lribeirogomes , so that ( and like google code itself :wink: ) is a remnant of the past... 
Are you looking for the [releases](https://github.com/tpaviot/oce/releases)?
It would be helpful to know on which platform you work...

QbProg, 2015-08-17:
I deleted both these two old download pages. Binaries will be hosted
directly on github (in the page of each github release). If there are inks
to these old pages then should be removed. Where did you find the
reference?

lribeirogomes, 2015-08-17:
Currently, I'm using Windows 10. If this helps, I'm not using any IDE. Just a text editor and powershell. The OCE - version 16.1, according to releases - will be used with pythonOCC 16.0 and python 3.4, both for Windows 32bit.

jf---, 2015-08-18:
@lribeirogomes , good find, the README is confusing indeed, with pointing to ancient releases ( <= OCE 0.14 ) and a 404 for newer releases...

tpaviot, 2015-08-18:
@lribeirogomes README and other documentation files have been updated a couple of days ago. Can you please have a look at PR #568 ?

lribeirogomes, 2015-08-18:
@tpaviot Yes, I can. It's great, actually, although I still can see the http://code.google.com/p/oce-binaries/downloads/ broken link.

tpaviot, 2015-08-19:
What do you mean with "broken link"? I can access this link, it provides ole binaries for oce 0.11 and 0.12. Anyway, this page is scheduled for deletion, I will remove the link from our README.

lribeirogomes, 2015-08-20:
@tpaviot My apologies. I may have misused the term incorrectly. The page exists, but I don't have any access, even after logging in. But that is not even relevant now since, a you said, the page will no longer exist.

@jf--- @QbProg @tpaviot Thank you all for addressing this issue - and creating new ones in the process - in short time. Keep the good work you have made so far.

tpaviot, 2015-08-22:
Removed google page link from the "ressource pages".

Issue fixed by PR #568

closed 2015-08-22 by tpaviot