tpaviot/oce #571: Appveyor support

tpaviot, 2015-08-21:
I started working on a appveyor support for oce. Appveyor is similar to travis, but supports compilation on Windows for both MinGW and Visual Studio. The oce travis-ci service (available at https://travis-ci.org/tpaviot/oce) really helped the development and the way we ensure pull requests don't create any regression; It should be the same for Windows, since many users usually complain about Windows support, whether it comes from the platform (32/64), the copiler itself (msvc, mingw) or its version (VS2012, 2013, 2015, gcc4.8, gcc5 etc.). Appveyor should be able to help us. Moreover, appveyor also provides a deployment service that would enable us to provide windows binaries for most platforms/compilers, in an automated way, as soon as a release is tagged. In other words, it's worth trying to investigate this service.

I opened an account for oce, available at: https://ci.appveyor.com/project/tpaviot/oce. Appveyor tests are pushed to tp/appveyor. It just adds an appveyor.yml configuration file at root. As soon as a commit is pushed to this branch, a new build is triggered on the appveyor server side. It's a bit of experimental right now, exactly the same way I set up the travis-ci support.

@QbProg and @Bihlerben I think I will need your help with that work. I use to work with mingw but I'm not familiar with msvc.

I will update this issue as the work is going on.

QbProg, 2015-08-21:
That's very good news!
Recently on my home computer I wrote a script (based on remobjects train
which is basically javascript) to compile all the MSVC configurations (24
in total), comprending 2010/12/13/15 , static/shared, win32/64 ,
debug/release ,etc... and also a few mingw32/64 ones.
I'll take a look at appveyor and see if we could use that script as
starting point, or if we need to implement something more specific.
Qb

tpaviot, 2015-09-07:
lot of work has been achieved. See #574

tpaviot, 2015-10-10:
Closing issue. Appveyor support added to 0.17.1 release. Please open another ticket for any issue related to appveyor CI support.

closed 2015-10-10 by tpaviot