tpaviot/oce #576: oce 0.17 tests fail on mac os X

samitbasu, 2015-09-09:
Building OCE v 0.17 on Mac OS X, with clang (claimed version is Apple LLVM Version 6.1.0).  The cake and build steps work fine, however "make test" fails the following tests:

Test project /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld
      Start  1: BRepAlgoAPITestSuite.testCutBox
 1/48 Test  #1: BRepAlgoAPITestSuite.testCutBox .................   Passed    0.02 sec
      Start  2: BRepAlgoAPITestSuite.testCutCylSphere
 2/48 Test  #2: BRepAlgoAPITestSuite.testCutCylSphere ...........   Passed    0.02 sec
      Start  3: BRepExtremaTestSuite.testDistSphereSphere
 3/48 Test  #3: BRepExtremaTestSuite.testDistSphereSphere .......   Passed    0.01 sec
      Start  4: BRepExtremaTestSuite.testDistSphereVertex
 4/48 Test  #4: BRepExtremaTestSuite.testDistSphereVertex .......   Passed    0.01 sec
      Start  5: BRepExtremaTestSuite.testDistVertexEdge
 5/48 Test  #5: BRepExtremaTestSuite.testDistVertexEdge .........   Passed    0.01 sec
      Start  6: BRepExtremaTestSuite.testDistEdgeVertex
 6/48 Test  #6: BRepExtremaTestSuite.testDistEdgeVertex .........   Passed    0.01 sec
      Start  7: BRepExtremaTestSuite.testDistVertexFace
 7/48 Test  #7: BRepExtremaTestSuite.testDistVertexFace .........   Passed    0.01 sec
      Start  8: BRepExtremaTestSuite.testDistFaceVertex
 8/48 Test  #8: BRepExtremaTestSuite.testDistFaceVertex .........   Passed    0.01 sec
      Start  9: BRepGPropTestSuite.testComputeBoxVolume
 9/48 Test  #9: BRepGPropTestSuite.testComputeBoxVolume .........   Passed    0.01 sec
      Start 10: BRepGPropTestSuite.testComputeBoxSurface
10/48 Test #10: BRepGPropTestSuite.testComputeBoxSurface ........   Passed    0.01 sec
      Start 11: BRepGPropTestSuite.testComputeSphereVolume
11/48 Test #11: BRepGPropTestSuite.testComputeSphereVolume ......   Passed    0.01 sec
      Start 12: BRepImportTestSuite.testImportBRep1
12/48 Test #12: BRepImportTestSuite.testImportBRep1 .............***Exception: SegFault  0.58 sec
      Start 13: BRepImportTestSuite.testImportBRep2
13/48 Test #13: BRepImportTestSuite.testImportBRep2 .............***Exception: SegFault  0.28 sec
      Start 14: BRepImportTestSuite.testImportBRep3
14/48 Test #14: BRepImportTestSuite.testImportBRep3 .............***Exception: SegFault  0.28 sec
      Start 15: BRepImportTestSuite.testImportBRep4
15/48 Test #15: BRepImportTestSuite.testImportBRep4 .............***Exception: SegFault  0.30 sec
      Start 16: BRepMeshTestSuite.testMeshBox
16/48 Test #16: BRepMeshTestSuite.testMeshBox ...................   Passed    0.01 sec
      Start 17: BRepMeshTestSuite.testMeshSphere
17/48 Test #17: BRepMeshTestSuite.testMeshSphere ................   Passed    0.18 sec
      Start 18: BRepMeshTestSuite.testMeshTorus
18/48 Test #18: BRepMeshTestSuite.testMeshTorus .................***Exception: SegFault  0.29 sec
      Start 19: BRepOffsetAPITestSuite.testEvolvedShape
19/48 Test #19: BRepOffsetAPITestSuite.testEvolvedShape .........***Exception: SegFault  0.22 sec
      Start 20: BRepOffsetAPITestSuite.testOffsetBox
20/48 Test #20: BRepOffsetAPITestSuite.testOffsetBox ............***Exception: SegFault  0.24 sec
      Start 21: BRepPrimAPITestSuite.testCreateBox
21/48 Test #21: BRepPrimAPITestSuite.testCreateBox ..............   Passed    0.01 sec
      Start 22: BRepPrimAPITestSuite.testCreateFlatBox
22/48 Test #22: BRepPrimAPITestSuite.testCreateFlatBox ..........   Passed    0.01 sec
      Start 23: BRepPrimAPITestSuite.testCreateSmallestBox
23/48 Test #23: BRepPrimAPITestSuite.testCreateSmallestBox ......   Passed    0.01 sec
      Start 24: gpTestSuite.testgp_PntConstruct
24/48 Test #24: gpTestSuite.testgp_PntConstruct .................   Passed    0.00 sec
      Start 25: gpTestSuite.testgp_PntDistance
25/48 Test #25: gpTestSuite.testgp_PntDistance ..................   Passed    0.00 sec
      Start 26: gpTestSuite.testBuildThousandsPoints
26/48 Test #26: gpTestSuite.testBuildThousandsPoints ............   Passed    0.00 sec
      Start 27: gpTestSuite.testGCC42gp_Trsf
27/48 Test #27: gpTestSuite.testGCC42gp_Trsf ....................   Passed    0.00 sec
      Start 28: IGESImportTestSuite.testImportIGES_1
28/48 Test #28: IGESImportTestSuite.testImportIGES_1 ............***Exception: SegFault  0.30 sec
      Start 29: OCAFExportTestSuite.testExportAscii
29/48 Test #29: OCAFExportTestSuite.testExportAscii .............***Exception: SegFault  0.47 sec
      Start 30: OCAFExportTestSuite.testExportNonAscii
30/48 Test #30: OCAFExportTestSuite.testExportNonAscii ..........***Exception: SegFault  0.37 sec
      Start 31: OCAFExportTestSuite.testOverflow
31/48 Test #31: OCAFExportTestSuite.testOverflow ................   Passed    0.02 sec
      Start 32: SortToolsTestSuite.testQuickSortOfInteger
32/48 Test #32: SortToolsTestSuite.testQuickSortOfInteger .......   Passed    0.00 sec
      Start 33: Standard_StdAllocatorTestSuite.testTraits
33/48 Test #33: Standard_StdAllocatorTestSuite.testTraits .......   Passed    0.01 sec
      Start 34: Standard_StdAllocatorTestSuite.testContainers
[... 43 lines cut ...]
	 29 - OCAFExportTestSuite.testExportAscii (SEGFAULT)
	 30 - OCAFExportTestSuite.testExportNonAscii (SEGFAULT)
	 35 - STEPImportTestSuite.testImportAP203_1 (SEGFAULT)
	 36 - STEPImportTestSuite.testImportAP203_2 (SEGFAULT)
	 37 - STEPImportTestSuite.testImportAP214_1 (SEGFAULT)
	 38 - STEPImportTestSuite.testImportAP214_2 (SEGFAULT)
	 39 - STEPImportTestSuite.testImportAP214_3 (SEGFAULT)

Furthermore, running STEPImport_test through valgrind shows the following:
==49147== Invalid read of size 4
==49147==    at 0x101BEA3CE: BSplCLib::LocateParameter(int, TColStd_Array1OfReal const&, TColStd_Array1OfInteger const&, double, bool, int&, double&) (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKMath.10.0.0.dylib)
==49147==    by 0x101A445DC: Geom_BSplineCurve::ValidateCache(double) (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKG3d.10.0.0.dylib)
==49147==    by 0x101A44AD2: Geom_BSplineCurve::D0(double, gp_Pnt&) const (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKG3d.10.0.0.dylib)
==49147==    by 0x101A591CE: Geom_Curve::Value(double) const (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKG3d.10.0.0.dylib)
==49147==    by 0x100A96AB6: ShapeAnalysis_Curve::Project(Handle_Geom_Curve const&, gp_Pnt const&, double, gp_Pnt&, double&, double, double, bool) const (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKShHealing.10.0.0.dylib)
==49147==    by 0x10012E730: StepToTopoDS_TranslateEdge::MakeFromCurve3D(Handle_StepGeom_Curve const&, Handle_StepShape_EdgeCurve const&, Handle_StepShape_Vertex const&, double, TopoDS_Edge&, TopoDS_Vertex&, TopoDS_Vertex&, StepToTopoDS_Tool&) (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKSTEP.10.0.0.dylib)
==49147==    by 0x10012DD02: StepToTopoDS_TranslateEdge::Init(Handle_StepShape_Edge const&, StepToTopoDS_Tool&, StepToTopoDS_NMTool&) (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKSTEP.10.0.0.dylib)
==49147==    by 0x100130A5F: StepToTopoDS_TranslateEdgeLoop::Init(Handle_StepShape_FaceBound const&, TopoDS_Face const&, Handle_Geom_Surface const&, Handle_StepGeom_Surface const&, bool, StepToTopoDS_Tool&, StepToTopoDS_NMTool&) (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKSTEP.10.0.0.dylib)
==49147==    by 0x100133D2E: StepToTopoDS_TranslateFace::Init(Handle_StepShape_FaceSurface const&, StepToTopoDS_Tool&, StepToTopoDS_NMTool&) (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKSTEP.10.0.0.dylib)
==49147==    by 0x10013642B: StepToTopoDS_TranslateShell::Init(Handle_StepShape_ConnectedFaceSet const&, StepToTopoDS_Tool&, StepToTopoDS_NMTool&) (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKSTEP.10.0.0.dylib)
==49147==    by 0x100120BF6: StepToTopoDS_Builder::Init(Handle_StepShape_ManifoldSolidBrep const&, Handle_Transfer_TransientProcess const&) (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKSTEP.10.0.0.dylib)
==49147==    by 0x100156176: STEPControl_ActorRead::TransferEntity(Handle_StepGeom_GeometricRepresentationItem const&, Handle_Transfer_TransientProcess const&, bool) (in /Users/sbasu/Devel/trimesh/oce-OCE-0.17/bld/Darwin/x86_64-Release-64/libTKSTEP.10.0.0.dylib)
==49147==  Address 0x0 is not stack'd, malloc'd or (recently) free'd

Is this a known issue?  I haven't been able to find anyone else reporting this issue, but I may have missed it.

chrisidefix, 2015-09-15:
I am having similarly severe issues under Mac OS X.  Which OS version are you running on?
Even running some basic example code SEGFAULT (I listed some stack traces here: https://gist.github.com/chrisidefix/f35c3fb0572e6bfc9ba1).

These problems all appear to be related to code execute inside `libTKMath.10.0.0.dylib`, especially in `BSplCLib::LocateParameter`  `BSplCLib::BuildKnots`  `BSplCLib::IncreaseDegree` (calling `PLib::SetPoles`). Have you tried to debug the issue?

chrisidefix, 2015-09-15:
side note: @tpaviot appears to use OS X himself (if you look through other issues) I can't imagine such basic problems would have not been discovered. Since I am running on OS X 10.11 (GM release), I am starting to think it might be related to some recent changes in OS X.

chrisidefix, 2015-09-15:
I tested everything under Linux Mint (17.2) first and didn't see any problems occurring with the same basic example / test code.

I then also compiled everything in `DEBUG` mode to see what is really going on. Of course now all issues are gone. Might be some race-condition in the code or the compiler optimising something that causes issues, but so far I couldn't re-create the issues in `DEBUG` mode. Obviously this isn't very satisfactory, but at least there is a work-around for now.

To compile your code in debug mode first add the following flag to your cmake setup:
`CMAKE_BUILD_TYPE:STRING=Debug`

I also added the `-ggdb` flag to the compiler flags: `CMAKE_CXX_FLAGS_DEBUG` and `CMAKE_C_FLAGS_DEBUG`

Apparently there used to be a variable `OCE_BUILD_TYPE:STRING=Debug` that had to be set for debug compilation in the past, but from what I can see it has been removed.

samitbasu, 2015-09-20:
Sorry - was off the grid for a while.  I will try and debug the issue.  I was using OS X 10.10.4 - now on 10.10.5 and Xcode 7.  I also get the same issues when building open cascade 6.9 direct from sources.

samitbasu, 2015-09-20:
Update: building master in debug mode (i.e., manually setting the CMAKE_BUILD_TYPE to Debug) on OS X 10.10.5 w/ Xcode 7, almost all tests pass:

Running tests...
Test project /Users/sbasu/Devel/trimesh/oce/bld/test
      Start  1: BRepAlgoAPITestSuite.testCutBox
 1/48 Test  #1: BRepAlgoAPITestSuite.testCutBox .................   Passed    0.06 sec
      Start  2: BRepAlgoAPITestSuite.testCutCylSphere
 2/48 Test  #2: BRepAlgoAPITestSuite.testCutCylSphere ...........   Passed    0.05 sec
      Start  3: BRepExtremaTestSuite.testDistSphereSphere
 3/48 Test  #3: BRepExtremaTestSuite.testDistSphereSphere .......   Passed    0.02 sec
      Start  4: BRepExtremaTestSuite.testDistSphereVertex
 4/48 Test  #4: BRepExtremaTestSuite.testDistSphereVertex .......   Passed    0.01 sec
      Start  5: BRepExtremaTestSuite.testDistVertexEdge
 5/48 Test  #5: BRepExtremaTestSuite.testDistVertexEdge .........   Passed    0.01 sec
      Start  6: BRepExtremaTestSuite.testDistEdgeVertex
 6/48 Test  #6: BRepExtremaTestSuite.testDistEdgeVertex .........   Passed    0.01 sec
      Start  7: BRepExtremaTestSuite.testDistVertexFace
 7/48 Test  #7: BRepExtremaTestSuite.testDistVertexFace .........   Passed    0.01 sec
      Start  8: BRepExtremaTestSuite.testDistFaceVertex
 8/48 Test  #8: BRepExtremaTestSuite.testDistFaceVertex .........   Passed    0.01 sec
      Start  9: BRepGPropTestSuite.testComputeBoxVolume
 9/48 Test  #9: BRepGPropTestSuite.testComputeBoxVolume .........   Passed    0.02 sec
      Start 10: BRepGPropTestSuite.testComputeBoxSurface
10/48 Test #10: BRepGPropTestSuite.testComputeBoxSurface ........   Passed    0.01 sec
      Start 11: BRepGPropTestSuite.testComputeSphereVolume
11/48 Test #11: BRepGPropTestSuite.testComputeSphereVolume ......   Passed    0.01 sec
      Start 12: BRepImportTestSuite.testImportBRep1
12/48 Test #12: BRepImportTestSuite.testImportBRep1 .............   Passed    0.56 sec
      Start 13: BRepImportTestSuite.testImportBRep2
13/48 Test #13: BRepImportTestSuite.testImportBRep2 .............   Passed    0.04 sec
      Start 14: BRepImportTestSuite.testImportBRep3
14/48 Test #14: BRepImportTestSuite.testImportBRep3 .............   Passed    0.03 sec
      Start 15: BRepImportTestSuite.testImportBRep4
15/48 Test #15: BRepImportTestSuite.testImportBRep4 .............   Passed    0.07 sec
      Start 16: BRepMeshTestSuite.testMeshBox
16/48 Test #16: BRepMeshTestSuite.testMeshBox ...................   Passed    0.02 sec
      Start 17: BRepMeshTestSuite.testMeshSphere
17/48 Test #17: BRepMeshTestSuite.testMeshSphere ................   Passed    0.55 sec
      Start 18: BRepMeshTestSuite.testMeshTorus
18/48 Test #18: BRepMeshTestSuite.testMeshTorus .................   Passed    9.17 sec
      Start 19: BRepOffsetAPITestSuite.testEvolvedShape
19/48 Test #19: BRepOffsetAPITestSuite.testEvolvedShape .........   Passed    0.05 sec
      Start 20: BRepOffsetAPITestSuite.testOffsetBox
20/48 Test #20: BRepOffsetAPITestSuite.testOffsetBox ............***Exception: Other  3.84 sec
      Start 21: BRepPrimAPITestSuite.testCreateBox
21/48 Test #21: BRepPrimAPITestSuite.testCreateBox ..............   Passed    0.02 sec
      Start 22: BRepPrimAPITestSuite.testCreateFlatBox
22/48 Test #22: BRepPrimAPITestSuite.testCreateFlatBox ..........   Passed    0.01 sec
      Start 23: BRepPrimAPITestSuite.testCreateSmallestBox
23/48 Test #23: BRepPrimAPITestSuite.testCreateSmallestBox ......   Passed    0.01 sec
      Start 24: gpTestSuite.testgp_PntConstruct
24/48 Test #24: gpTestSuite.testgp_PntConstruct .................   Passed    0.01 sec
      Start 25: gpTestSuite.testgp_PntDistance
25/48 Test #25: gpTestSuite.testgp_PntDistance ..................   Passed    0.00 sec
      Start 26: gpTestSuite.testBuildThousandsPoints
26/48 Test #26: gpTestSuite.testBuildThousandsPoints ............   Passed    0.00 sec
      Start 27: gpTestSuite.testGCC42gp_Trsf
27/48 Test #27: gpTestSuite.testGCC42gp_Trsf ....................   Passed    0.01 sec
      Start 28: IGESImportTestSuite.testImportIGES_1
28/48 Test #28: IGESImportTestSuite.testImportIGES_1 ............   Passed    0.37 sec
      Start 29: OCAFExportTestSuite.testExportAscii
29/48 Test #29: OCAFExportTestSuite.testExportAscii .............   Passed    0.04 sec
      Start 30: OCAFExportTestSuite.testExportNonAscii
30/48 Test #30: OCAFExportTestSuite.testExportNonAscii ..........   Passed    0.03 sec
      Start 31: OCAFExportTestSuite.testOverflow
31/48 Test #31: OCAFExportTestSuite.testOverflow ................   Passed    0.02 sec
      Start 32: SortToolsTestSuite.testQuickSortOfInteger
32/48 Test #32: SortToolsTestSuite.testQuickSortOfInteger .......   Passed    0.01 sec
      Start 33: Standard_StdAllocatorTestSuite.testTraits
33/48 Test #33: Standard_StdAllocatorTestSuite.testTraits .......   Passed    0.02 sec
[... 17 lines cut ...]
42/48 Test #42: UnitsAPITestSuite.testCurrentUnits ..............   Passed    0.01 sec
      Start 43: UnitsAPITestSuite.testMDTVCurrentUnits
43/48 Test #43: UnitsAPITestSuite.testMDTVCurrentUnits ..........   Passed    0.04 sec
      Start 44: TestSuite.testNullPointer
44/48 Test #44: TestSuite.testNullPointer .......................   Passed    0.00 sec
      Start 45: TestSuite.testFloatEq
45/48 Test #45: TestSuite.testFloatEq ...........................   Passed    0.00 sec
      Start 46: TestSuite.testFloatNeq
46/48 Test #46: TestSuite.testFloatNeq ..........................   Passed    0.00 sec
[... cut ...] stBoolean
47/48 Test #47: TestSuite.testBoolean ...........................   Passed    0.00 sec
      Start 48: TestSuite.testIntegerLighter
48/48 Test #48: TestSuite.testIntegerLighter ....................   Passed    0.00 sec

98% tests passed, 1 tests failed out of 48

Total Test time (real) =  16.93 sec

The following tests FAILED:
	 20 - BRepOffsetAPITestSuite.testOffsetBox (OTHER_FAULT)
Errors while running CTest

Will try rebuilding in release mode to see if the problem reappears.

Samit

samitbasu, 2015-09-20:
Building in release mode causes the seg faults again:

Test project /Users/sbasu/Devel/trimesh/oce/bld/test
      Start  1: BRepAlgoAPITestSuite.testCutBox
 1/48 Test  #1: BRepAlgoAPITestSuite.testCutBox .................   Passed    0.03 sec
      Start  2: BRepAlgoAPITestSuite.testCutCylSphere
 2/48 Test  #2: BRepAlgoAPITestSuite.testCutCylSphere ...........   Passed    0.02 sec
      Start  3: BRepExtremaTestSuite.testDistSphereSphere
 3/48 Test  #3: BRepExtremaTestSuite.testDistSphereSphere .......   Passed    0.02 sec
      Start  4: BRepExtremaTestSuite.testDistSphereVertex
 4/48 Test  #4: BRepExtremaTestSuite.testDistSphereVertex .......   Passed    0.01 sec
      Start  5: BRepExtremaTestSuite.testDistVertexEdge
 5/48 Test  #5: BRepExtremaTestSuite.testDistVertexEdge .........   Passed    0.01 sec
      Start  6: BRepExtremaTestSuite.testDistEdgeVertex
 6/48 Test  #6: BRepExtremaTestSuite.testDistEdgeVertex .........   Passed    0.01 sec
      Start  7: BRepExtremaTestSuite.testDistVertexFace
 7/48 Test  #7: BRepExtremaTestSuite.testDistVertexFace .........   Passed    0.01 sec
      Start  8: BRepExtremaTestSuite.testDistFaceVertex
 8/48 Test  #8: BRepExtremaTestSuite.testDistFaceVertex .........   Passed    0.01 sec
      Start  9: BRepGPropTestSuite.testComputeBoxVolume
 9/48 Test  #9: BRepGPropTestSuite.testComputeBoxVolume .........   Passed    0.01 sec
      Start 10: BRepGPropTestSuite.testComputeBoxSurface
10/48 Test #10: BRepGPropTestSuite.testComputeBoxSurface ........   Passed    0.01 sec
      Start 11: BRepGPropTestSuite.testComputeSphereVolume
11/48 Test #11: BRepGPropTestSuite.testComputeSphereVolume ......   Passed    0.01 sec
      Start 12: BRepImportTestSuite.testImportBRep1
12/48 Test #12: BRepImportTestSuite.testImportBRep1 .............***Exception: SegFault  0.56 sec
      Start 13: BRepImportTestSuite.testImportBRep2
13/48 Test #13: BRepImportTestSuite.testImportBRep2 .............***Exception: SegFault  0.29 sec
      Start 14: BRepImportTestSuite.testImportBRep3
14/48 Test #14: BRepImportTestSuite.testImportBRep3 .............***Exception: SegFault  0.29 sec
      Start 15: BRepImportTestSuite.testImportBRep4
15/48 Test #15: BRepImportTestSuite.testImportBRep4 .............***Exception: SegFault  0.31 sec
      Start 16: BRepMeshTestSuite.testMeshBox
16/48 Test #16: BRepMeshTestSuite.testMeshBox ...................   Passed    0.02 sec
      Start 17: BRepMeshTestSuite.testMeshSphere
17/48 Test #17: BRepMeshTestSuite.testMeshSphere ................   Passed    0.18 sec
      Start 18: BRepMeshTestSuite.testMeshTorus
18/48 Test #18: BRepMeshTestSuite.testMeshTorus .................***Exception: SegFault  0.31 sec
      Start 19: BRepOffsetAPITestSuite.testEvolvedShape
19/48 Test #19: BRepOffsetAPITestSuite.testEvolvedShape .........***Exception: SegFault  0.25 sec
      Start 20: BRepOffsetAPITestSuite.testOffsetBox
20/48 Test #20: BRepOffsetAPITestSuite.testOffsetBox ............***Exception: SegFault  0.24 sec
      Start 21: BRepPrimAPITestSuite.testCreateBox
21/48 Test #21: BRepPrimAPITestSuite.testCreateBox ..............   Passed    0.01 sec
      Start 22: BRepPrimAPITestSuite.testCreateFlatBox
22/48 Test #22: BRepPrimAPITestSuite.testCreateFlatBox ..........   Passed    0.01 sec
      Start 23: BRepPrimAPITestSuite.testCreateSmallestBox
23/48 Test #23: BRepPrimAPITestSuite.testCreateSmallestBox ......   Passed    0.01 sec
      Start 24: gpTestSuite.testgp_PntConstruct
24/48 Test #24: gpTestSuite.testgp_PntConstruct .................   Passed    0.01 sec
      Start 25: gpTestSuite.testgp_PntDistance
25/48 Test #25: gpTestSuite.testgp_PntDistance ..................   Passed    0.00 sec
      Start 26: gpTestSuite.testBuildThousandsPoints
26/48 Test #26: gpTestSuite.testBuildThousandsPoints ............   Passed    0.00 sec
      Start 27: gpTestSuite.testGCC42gp_Trsf
27/48 Test #27: gpTestSuite.testGCC42gp_Trsf ....................   Passed    0.00 sec
      Start 28: IGESImportTestSuite.testImportIGES_1
28/48 Test #28: IGESImportTestSuite.testImportIGES_1 ............***Exception: SegFault  0.30 sec
      Start 29: OCAFExportTestSuite.testExportAscii
29/48 Test #29: OCAFExportTestSuite.testExportAscii .............   Passed    0.04 sec
      Start 30: OCAFExportTestSuite.testExportNonAscii
30/48 Test #30: OCAFExportTestSuite.testExportNonAscii ..........   Passed    0.02 sec
      Start 31: OCAFExportTestSuite.testOverflow
31/48 Test #31: OCAFExportTestSuite.testOverflow ................   Passed    0.02 sec
      Start 32: SortToolsTestSuite.testQuickSortOfInteger
32/48 Test #32: SortToolsTestSuite.testQuickSortOfInteger .......   Passed    0.01 sec
      Start 33: Standard_StdAllocatorTestSuite.testTraits
33/48 Test #33: Standard_StdAllocatorTestSuite.testTraits .......   Passed    0.01 sec
      Start 34: Standard_StdAllocatorTestSuite.testContainers
[... 28 lines cut ...]
48/48 Test #48: TestSuite.testIntegerLighter ....................   Passed    0.00 sec

73% tests passed, 13 tests failed out of 48

Total Test time (real) =   4.74 sec

The following tests FAILED:
	 12 - BRepImportTestSuite.testImportBRep1 (SEGFAULT)
	 13 - BRepImportTestSuite.testImportBRep2 (SEGFAULT)
	 14 - BRepImportTestSuite.testImportBRep3 (SEGFAULT)
	 15 - BRepImportTestSuite.testImportBRep4 (SEGFAULT)
	 18 - BRepMeshTestSuite.testMeshTorus (SEGFAULT)
	 19 - BRepOffsetAPITestSuite.testEvolvedShape (SEGFAULT)
	 20 - BRepOffsetAPITestSuite.testOffsetBox (SEGFAULT)
	 28 - IGESImportTestSuite.testImportIGES_1 (SEGFAULT)
	 35 - STEPImportTestSuite.testImportAP203_1 (SEGFAULT)
	 36 - STEPImportTestSuite.testImportAP203_2 (SEGFAULT)
	 37 - STEPImportTestSuite.testImportAP214_1 (SEGFAULT)
	 38 - STEPImportTestSuite.testImportAP214_2 (SEGFAULT)
	 39 - STEPImportTestSuite.testImportAP214_3 (SEGFAULT)
Errors while running CTest

That will make debugging more difficult.  Will post anything I can find.

Samit

chrisidefix, 2015-09-21:
Yes, thanks for also checking in debug mode - I think it would be worth extending the travis testing options to include a build under OS X (apparently they can generate VMs for that now) to make issues like this one surface more prominently. I am starting to test this here (https://github.com/heartvalve/oce/blob/osx-test/.travis.yml).

samitbasu, 2015-09-21:
I've narrowed it down (I believe) to the following idiom. See in BSplCLib.cxx, on line 301:
```
 const TColStd_Array1OfInteger& Mults,
 const Standard_Real            U,
 const Standard_Boolean         Periodic,
 Standard_Integer&              KnotIndex,
 Standard_Real&                 NewU) 
{
  Standard_Integer first,last;
  if (&Mults) { // <-- This test is always true, even when Mults is null.
    if (Periodic) {
      first = Knots.Lower();
```

During compilation, clang complains that because `Mults` is a reference, taking it's address must yield something that is not-NULL.  Here is the warning during build:

>[  5%] Building CXX object >adm/cmake/TKMath/CMakeFiles/TKMath.dir/__/__/__/src/BSplCLib/BSplCLib.cxx.o
>/Users/sbasu/Devel/trimesh/oce/src/BSplCLib/BSplCLib.cxx:301:8: warning: reference cannot be bound >to dereferenced null
>      pointer in well-defined C++ code; pointer may be assumed to always convert to true [-Wundefined->bool-conversion]
>  if (&Mults) {
>  ~~   ^~~~~
>/Users/sbasu/Devel/trimesh/oce/src/BSplCLib/BSplCLib.cxx:1437:8: warning: reference cannot be bound to dereferenced null
>      pointer in well-defined C++ code; comparison may be assumed to always evaluate to false
>      [-Wtautological-undefined-compare]
>  if (&Mults == NULL) {
>       ^~~~~    ~~~~

Furthermore, the calling code is a bit odd to me - 

>-> 1223	  BSplCLib::LocateParameter(deg,
>    1224				    (flatknots->Array1()),
>    1225				    (BSplCLib::NoMults()),
>    1226				    Parameter,

From the backtrace in LLDB, note the call to `BSplCLib::NoMults()`, which I gather is supposed to be the knot multiplicity vector when the spline has no define multiplicities (equivalent to all unit-multiplicities, I guess).  This is an interesting function.  It is defined in BSplCLib.hxx (around line 380)

```
  //! Used as argument for a non rational curve.
    static   TColStd_Array1OfReal& NoWeights() ;
  
  //! Used as argument for a flatknots evaluation.
    static   TColStd_Array1OfInteger& NoMults() ;
```
But the function itself is actually undefined - or at least I could not find the definition anywhere (including searching through the symbols in the libraries).  As a result, at run time, you get a unique phenomenon - a const reference bound to a null value.  I don't know if that's the intent, but Clang doesn't like it (and it seems to violate the requirements on references never pointing to NULL addresses).  I suspect that GCC accepts the condition (for now).  But I would be surprised if this doesn't crop up in a future version of GCC as well.

Anyone have a suggestion for how this _should_ be fixed?  This particular idiom (of checking a const-ref for NULL-ness) happens in a bunch of places.  I checked the code for OpenCascade 6.9.0, and the issue appears there as well.

a-betenev, 2015-09-21:
Hello,

This is a known issue, see 
http://tracker.dev.opencascade.org/view.php?id=26042
There is a work-around there; the correct fix naturally is to avoid this 
kind of code; to be done in OCCT 7.0.
Hope this helps.

Andrey

samitbasu, 2015-09-21:
Yes - thanks for pointing out the issue, Andrey.  Would it make sense to patch the CMake build system to disable optimized builds for Clang until the issue is fixed?

a-betenev, 2015-09-22:
On 21.09.2015 18:57, samitbasu wrote:
>
> Yes - thanks for pointing out the issue, Andrey. Would it make sense 
> to patch the CMake build system to disable optimized builds for Clang 
> until the issue is fixed?
>
I believe that all fixes (including local disabling of optimization) 
should already be present in relevant commits in OCCT repo.

thatcadguy, 2015-11-19:
http://tracker.dev.opencascade.org/view.php?id=26042

rainman110, 2016-01-20:
I don't think it is an option to build only in debug mode due to performance reasons.

I backported the OCCT 7.0 patch for this fix. This patch is currently OCE-0.16.1 only but can be easily adapted to newer oce releases. The patch can be downloaded here:
https://github.com/DLR-SC/tigl-conda/blob/master/oce/oce-nullrefs-fix.patch

jf---, 2016-01-25:
@rainman110 , wow, that backport of the Clang compiler fix from upstream OCC is an impressive effort!
That's an important contribution :+1: , I think this also affects builds on linux with Clang 3.4+ .

Finally, are your builds on Conda already built with the Clang patch?

rainman110, 2016-01-25:
Hi @jf--- 

I only build it for OSX for now with the patch. The linux builds on conda are not affected, since I used an old gcc which does not do these compiler optimizations.

So far, all the conda builds on dlr-sc should not include these problems anymore.

jf---, 2016-01-25:
OK, thanks for the heads up, that's good to know...
Regarding linux, building with a recent Clang compiler leads to similar issues, right?

rainman110, 2016-01-25:
@jf--- I didn't try to compile it with clang on linux, but the problems should be identical.

It would make sense to integrate this patch into the next oce release.

jf---, 2016-01-25:
Very much in favor of that, its critical fix actually, so many thanks for your efforts backporting this!

jf---, 2016-01-27:
@rainman110 , regarding [the Patch](https://github.com/DLR-SC/tigl-conda/blob/master/oce/oce-nullrefs-fix.patch), is it OK to ask you for a PR at the OCE repo? Its such an important fix :trophy: 

Would you mind how you've gone about creating this patch?
Was there a particular OCC branch you could derive it from?

rainman110, 2016-01-27:
@jf---  I just followed the mantis entry by @a-betenev / https://github.com/tpaviot/oce/issues/576#issuecomment-142023714.

First I backported the proper fix from http://git.dev.opencascade.org/gitweb/?p=occt.git;a=commitdiff;h=0e14656b3052946192bbb08a998e9c23a3be8752. This however changed the public API of some classes and therefore @tpaviot  didn't really like it.

Then, I just implemented a work-around inspired by the workaround attached to this mantis issue. I just looked at the clang compiler warnings and figured out, were this patch is required.

This workaround won't be included in OCCT 7.0. Instead they will provide a proper fix including a changed API.

I can do a PR. How do you actually integrate changes to OCE, when a new OCCT release is targeted?

rainman110, 2016-05-24:
@tpaviot I think this issue can be closed.

closed 2016-05-24 by tpaviot