tpaviot/oce #580: Make Error at ExprIntrp.c:2204

yokoyamy, 2015-10-02:
Hi

I’m trying to build OCE on MinGW with followin instrcution.

https://github.com/tpaviot/oce/blob/master/BUILD.MINGWw64.md

I generated Makefile and did make.

After 9% make has run It stopped with message below.

Does anyone know how to fix this problem?

[  9%] Building C object adm/cmake/TKMath/CMakeFiles/TKMath.dir/__/__/__/drv/Exp
rIntrp/ExprIntrp.tab.c.obj
[  9%] Building C object adm/cmake/TKMath/CMakeFiles/TKMath.dir/__/__/__/drv/Exp
rIntrp/lex.ExprIntrp.c.obj 
In file included from lex.ExprIntrp.c:2204:0:
c:\mingw\include\unistd.h:109:5: error: conflicting types for 'nanosleep'
 int nanosleep( const struct timespec *period, struct timespec *residual )
     ^
c:\mingw\include\unistd.h:105:5: note: previous declaration of 'nanosleep' was h
ere
 int nanosleep( const struct timespec *, struct timespec * );
     ^
c:\mingw\include\unistd.h: In function 'nanosleep':
c:\mingw\include\unistd.h:112:13: error: dereferencing pointer to incomplete typ
e
     residual->tv_sec = (long long)(residual->tv_nsec = 0);
             ^
c:\mingw\include\unistd.h:112:44: error: dereferencing pointer to incomplete typ
e
     residual->tv_sec = (long long)(residual->tv_nsec = 0);
                                            ^
c:\mingw\include\unistd.h:113:41: error: dereferencing pointer to incomplete typ
e
   return __mingw_sleep((unsigned)(period->tv_sec), (period->tv_sec < 0LL)
                                         ^
c:\mingw\include\unistd.h:113:59: error: dereferencing pointer to incomplete typ
e
   return __mingw_sleep((unsigned)(period->tv_sec), (period->tv_sec < 0LL)
                                                           ^
c:\mingw\include\unistd.h:114:41: error: dereferencing pointer to incomplete typ
e
     ? (unsigned)(-1) : (unsigned)(period->tv_nsec));
                                         ^
make[2]: *** [adm/cmake/TKMath/CMakeFiles/TKMath.dir/__/__/__/drv/ExprIntrp/lex.
ExprIntrp.c.obj] Error 1
make[1]: *** [adm/cmake/TKMath/CMakeFiles/TKMath.dir/all] Error 2
make: *** [all] Error 2

Thanks.

tpaviot, 2015-10-02:
which mingw version do you use ? can you please paste your g++-v ?

yokoyamy, 2015-10-02:
Sorry,g++-v returns
sh: g++-v: command not found

but
gcc --version returns...
gcc.exe (GCC) 4.8.1

And another informaion is that  make for OCE-0.13 is running  through [32%] now,hasn't finished yet.

tpaviot, 2015-10-02:
Latest oce release is oce-0.17. Many issues were fixed since 0.13 was
released. I suggest you move up to the latest release, we can't provide
support for such an outdated version.

yokoyamy, 2015-10-02:
I think I did make for oce-0.17, which stopped at 9%.
Because READM.md includes release information of oce-0.17

Oce-0.13 still running reached at 39%.

yokoyamy, 2015-10-02:
I re-installed OCE-0.17 explicitly and did make.

However, this time make stopped at 1% with following massage.

[  1%] Building CXX object adm/cmake/TKernel/CMakeFiles/TKernel.dir/__/__/__/src
/OSD/OSD_File.cxx.obj
c:/src/products/oss/oce/oce-OCE-0.17/src/OSD/OSD_File.cxx: In member function 'S
tandard_Size OSD_File::Size()':
c:/src/products/oss/oce/oce-OCE-0.17/src/OSD/OSD_File.cxx:1650:46: error: 'GetFi
leSizeEx' was not declared in this scope
  retVal = GetFileSizeEx (myFileHandle, &aSize);
                                              ^
make[2]: *** [adm/cmake/TKernel/CMakeFiles/TKernel.dir/__/__/__/src/OSD/OSD_File
.cxx.obj] Error 1
make[1]: *** [adm/cmake/TKernel/CMakeFiles/TKernel.dir/all] Error 2
make: *** [all] Error 2

Also OCE-0.13 stopped at 42%

[ 42%] Building CXX object adm/cmake/TKService/CMakeFiles/TKService.dir/__/__/__
/src/Font/Font_FontMgr.cxx.obj
c:/src/products/oss/oce/oce-OCE-0.13/src/Font/Font_FontMgr.cxx: In member functi
on 'void Font_FontMgr::InitFontDataBase()':
c:/src/products/oss/oce/oce-OCE-0.13/src/Font/Font_FontMgr.cxx:238:56: error: 'G
etSystemWindowsDirectoryA' was not declared in this scope
   UINT aStrLength = GetSystemWindowsDirectoryA (NULL, 0);
                                                        ^
make[2]: *** [adm/cmake/TKService/CMakeFiles/TKService.dir/__/__/__/src/Font/Fon
t_FontMgr.cxx.obj] Error 1
make[1]: *** [adm/cmake/TKService/CMakeFiles/TKService.dir/all] Error 2
make: *** [all] Error 2

Do you have any idea for these issues?

tpaviot, 2015-10-02:
Windows version ?

yokoyamy, 2015-10-02:
I thought It was.
I down loaded it from oce-history section of following page
https://github.com/tpaviot/oce

My Windows is 7Por 64bit

thatcadguy, 2015-12-14:
Is it an option for you to try Mingw-w64? Everything is much more up-to-date. You can still compile 32-bit with it if you need to.

thatcadguy, 2015-12-19:
@yokoyamy If you are still having this issue and don't want to try Mingw-w64, please try with a fresh install of MinGW, using the official installer: http://sourceforge.net/projects/mingw/files/latest/download?source=files
This should not be happening in the latest stable version of MinGW, I am only seeing that warning message reported for older versions.

closed 2015-12-26 by thatcadguy