tpaviot/oce #584: RHEL 6 & OCE >= 0.16 Test Fail

hobbes1069, 2015-10-13:
I have not been able to build official packages of OCE for RHEL/CentOS 6 since 0.16. The build completes fine but many tests fail:

```
Running tests...
/usr/bin/ctest --force-new-ctest-process
Test project /builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test
      Start  1: BRepAlgoAPITestSuite.testCutBox
 1/48 Test  #1: BRepAlgoAPITestSuite.testCutBox .................***Failed    0.01 sec
/builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test/BRepAlgoAPI_test/boolean_ops_test: error while loading shared libraries: libTKPrim.so.10: cannot open shared object file: No such file or directory
      Start  2: BRepAlgoAPITestSuite.testCutCylSphere
 2/48 Test  #2: BRepAlgoAPITestSuite.testCutCylSphere ...........***Failed    0.00 sec
/builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test/BRepAlgoAPI_test/boolean_ops_test: error while loading shared libraries: libTKPrim.so.10: cannot open shared object file: No such file or directory
      Start  3: BRepPrimAPITestSuite.testCreateBox
 3/48 Test  #3: BRepPrimAPITestSuite.testCreateBox ..............***Failed    0.00 sec
/builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test/BRepPrimAPI_test/BRepPrimAPI_test: error while loading shared libraries: libTKPrim.so.10: cannot open shared object file: No such file or directory
      Start  4: BRepPrimAPITestSuite.testCreateFlatBox
 4/48 Test  #4: BRepPrimAPITestSuite.testCreateFlatBox ..........***Failed    0.00 sec
/builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test/BRepPrimAPI_test/BRepPrimAPI_test: error while loading shared libraries: libTKPrim.so.10: cannot open shared object file: No such file or directory
      Start  5: BRepPrimAPITestSuite.testCreateSmallestBox
 5/48 Test  #5: BRepPrimAPITestSuite.testCreateSmallestBox ......***Failed    0.00 sec
/builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test/BRepPrimAPI_test/BRepPrimAPI_test: error while loading shared libraries: libTKPrim.so.10: cannot open shared object file: No such file or directory
      Start  6: OCAFExportTestSuite.testExportAscii
 6/48 Test  #6: OCAFExportTestSuite.testExportAscii .............***Failed    0.00 sec
/builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test/OCAFExport_test/OCAFExport_test: error while loading shared libraries: libTKCAF.so.10: cannot open shared object file: No such file or directory
      Start  7: OCAFExportTestSuite.testExportNonAscii
 7/48 Test  #7: OCAFExportTestSuite.testExportNonAscii ..........***Failed    0.00 sec
/builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test/OCAFExport_test/OCAFExport_test: error while loading shared libraries: libTKCAF.so.10: cannot open shared object file: No such file or directory
      Start  8: OCAFExportTestSuite.testOverflow
 8/48 Test  #8: OCAFExportTestSuite.testOverflow ................***Failed    0.00 sec
/builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test/OCAFExport_test/OCAFExport_test: error while loading shared libraries: libTKCAF.so.10: cannot open shared object file: No such file or directory
      Start  9: BRepGPropTestSuite.testComputeBoxVolume
 9/48 Test  #9: BRepGPropTestSuite.testComputeBoxVolume .........***Failed    0.00 sec
/builddir/build/BUILD/oce-7a8128711e768f5b72fd9c4e68ad1893aa0db61f/build/test/BRepGProp_test/BRepGProp_test: error while loading shared libraries: libTKG2d.so.10: cannot open shared object file: No such file or directory
```

I'm not sure what's different from RHEL 7 (which completes fine). Could this be some sort of weird rpath issue?

jf---, 2015-10-13:
This bit me too, yes, rpath is handled different with CMake < 3.x.x than from the 2.x.x version.
What could be an idea is OCE to test which version of CMake is used, such that this behaviour is not software dependend.

hobbes1069, 2015-10-13:
That would be best I don't have time to come up with a patch at the moment. For now I think I'll disable tests on RHEL 6, if everything checks out fine for RHEL 7 and Fedora 21+ it's probably OK.

dbarbier, 2015-10-14:
@hobbes1069 Are verbose build logs available somewhere?

hobbes1069, 2015-10-14:
That's as verbose as it gets. That's the output from running the test executable unless it has a verbose option:
```
# BRepAlgoAPI_test/boolean_ops_test
BRepAlgoAPI_test/boolean_ops_test: error while loading shared libraries: libTKPrim.so.10: cannot open shared object file: No such file or directory
# ldd BRepAlgoAPI_test/boolean_ops_test ops_test
        linux-vdso.so.1 =>  (0x00007fff79990000)
        libTKPrim.so.10 => not found
        libTKBO.so.10 => not found
        libTKTopAlgo.so.10 => not found
        libTKBRep.so.10 => not found
        libTKMath.so.10 => not found
        libTKernel.so.10 => not found
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f66b34fb000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007f66b31f4000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f66b2fde000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f66b2c4a000)
        /lib64/ld-linux-x86-64.so.2 (0x000055e45e126000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f66b29c5000)
```
Definitely looks like a rpath issue...

dbarbier, 2015-10-14:
@hobbes1069 I want to see how these shared libraries are created, i.e. which linker flags are used.  An hyperlink to online build logs is fine, if available.

hobbes1069, 2015-10-14:
Here's the linker line, I'll have to put the log on dropbox as it's huge.

```
Linking CXX executable boolean_ops_test
cd /builddir/build/BUILD/oce-OCE-0.16.1/build/test/BRepAlgoAPI_test && /usr/bin/cmake -E cmake_link_script CMakeFiles/boolean_ops_test.dir/link.txt --verbose=1
/usr/lib64/ccache/c++   -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=4 -m64 -mtune=generic  -O2 -g -DNDEBUG   -Wl,--as-needed CMakeFiles/boolean_ops_test.dir/boolean_ops.cpp.o  -o boolean_ops_test -rdynamic ../../Unix/x86_64-RelWithDebInfo-64/libTKPrim.so.9.0.0 ../../Unix/x86_64-RelWithDebInfo-64/libTKBO.so.9.0.0 ../../Unix/x86_64-RelWithDebInfo-64/libgtest.a ../../Unix/x86_64-RelWithDebInfo-64/libTKTopAlgo.so.9.0.0 ../../Unix/x86_64-RelWithDebInfo-64/libTKGeomAlgo.so.9.0.0 ../../Unix/x86_64-RelWithDebInfo-64/libTKBRep.so.9.0.0 ../../Unix/x86_64-RelWithDebInfo-64/libTKGeomBase.so.9.0.0 ../../Unix/x86_64-RelWithDebInfo-64/libTKG3d.so.9.0.0 ../../Unix/x86_64-RelWithDebInfo-64/libTKG2d.so.9.0.0 ../../Unix/x86_64-RelWithDebInfo-64/libTKMath.so.9.0.0 ../../Unix/x86_64-RelWithDebInfo-64/libTKernel.so.9.0.0 -ldl -lm -lrt -lpthread
/usr/bin/cmake -E cmake_progress_report /builddir/build/BUILD/oce-OCE-0.16.1/build/CMakeFiles
```

Well thank goodness for xz compression, 290MB to <1MB...
https://dl.dropboxusercontent.com/u/34775202/OCE-0.17-EL6.tar.xz

dbarbier, 2015-10-14:
You build with ` -DCMAKE_SKIP_RPATH:BOOL=ON`, so it is normal that loader does not find shared libraries in your $DESTDIR, you must set `LD_LIBRARY_PATH` when running tests.

hobbes1069, 2015-10-14:
Bah... It's in the cmake RPM macro on EL6, I'm not explicitly calling it. I'll try to override first. Thanks.

hobbes1069, 2015-10-14:
Ok, overriding the skip rpath option fixes the issue. Thanks.

closed 2015-10-14 by hobbes1069

dbarbier, 2015-10-14:
If some RPM tool complains about rpath being set to system libraries, you may try `-DOCE_RPATH_FILTER_SYSTEM_PATHS:BOOL=ON` option, it has been added for you ;) 
See #207.