tpaviot/oce #585: OCE 0.17 release removes TKAdvTools still used by smesh?

hobbes1069, 2015-10-14:
I was able to build 0.17 but when I tried to rebuild smesh against it I got the following error:

```
Linking CXX shared library Unix-x86_64-Release-64/libDriver.so
cd /builddir/build/BUILD/smesh-6.5.3.1/build/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/Driver.dir/link.txt --verbose=1
/usr/bin/c++  -fPIC -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches   -m64 -mtune=generic  -DNDEBUG  -Wl,--as-needed -shared -Wl,-soname,libDriver.so.2 -o Unix-x86_64-Release-64/libDriver.so.2.0.0 CMakeFiles/Driver.dir/Driver/Driver_Document.cpp.o CMakeFiles/Driver.dir/Driver/Driver_Mesh.cpp.o CMakeFiles/Driver.dir/Driver/Driver_SMDS_Mesh.cpp.o CMakeFiles/Driver.dir/Driver/Driver_SMESHDS_Mesh.cpp.o /usr/lib64/libTKernel.so.10.0.0 /usr/lib64/libTKMath.so.10.0.0 /usr/lib64/libTKBRep.so.10.0.0 /usr/lib64/libTKG2d.so.10.0.0 /usr/lib64/libTKG3d.so.10.0.0 /usr/lib64/libTKMeshVS.so.10.0.0 /usr/lib64/libTKPrim.so.10.0.0 /usr/lib64/libTKGeomBase.so.10.0.0 /usr/lib64/libTKGeomAlgo.so.10.0.0 /usr/lib64/libTKTopAlgo.so.10.0.0 /usr/lib64/libTKShHealing.so.10.0.0 -lTKAdvTools /usr/lib64/libTKV3d.so.10.0.0 /usr/lib64/libTKMesh.so.10.0.0 /usr/lib64/libTKHLR.so.10.0.0 /usr/lib64/libTKService.so.10.0.0 /usr/lib64/libTKShHealing.so.10.0.0 -lGLU -lGL -lSM -lICE -lX11 -lXext -lfreetype /usr/lib64/libTKTopAlgo.so.10.0.0 /usr/lib64/libTKGeomAlgo.so.10.0.0 /usr/lib64/libTKBRep.so.10.0.0 /usr/lib64/libTKGeomBase.so.10.0.0 /usr/lib64/libTKG3d.so.10.0.0 /usr/lib64/libTKG2d.so.10.0.0 /usr/lib64/libTKMath.so.10.0.0 /usr/lib64/libTKernel.so.10.0.0 -lpthread -ldl -lm -ltbbmalloc -ltbb 
/usr/bin/ld: cannot find -lTKAdvTools
collect2: error: ld returned 1 exit status
```

dbarbier, 2015-10-15:
Sorry, but there is nothing we can do about that, smesh must be updated.

closed 2015-10-15 by dbarbier

hobbes1069, 2015-10-15:
I wasn't sure if I should file the issue with OCE or your fork of smesh... Can we expect a new release of smesh to correct?