tpaviot/oce #590: CI Issues

thatcadguy, 2015-12-17:
Travis seems to be erroring a lot. There seem to be 2 issues right away: exceeding the 50 minute build limit (which is supposed to be 120 minutes according to the website), and package errors. Trying a few branches with `sudo apt-get update` added to see if this fixes the package errors (this is recommended by Travis: https://docs.travis-ci.com/user/installing-dependencies/).

Appveyor also seems to be hitting the 60 minute limit there.

Getting a paid plan for either won't fix the timeouts, the time limits are still there for paid packages I believe.

If Travis and Appveyor continue to be problematic, I'd like to set up our own servers to do the compiling. I have spare computers to do this and we could also take the opportunity to provide nightly builds of master.

thatcadguy, 2015-12-17:
Also, we should think about using a PPA to get a newer version of GCC and Clang. GCC 4.6.3 is extremely old, current version on Ubuntu 15.10 is 5.2.1.

thatcadguy, 2015-12-18:
#591

thatcadguy, 2015-12-21:
#593

thatcadguy, 2015-12-22:
The image link for AppVeyor build status in README.md is also broken. Tried to find what it should be but I think you have to log in to the account. It should be in the form of "https://ci.appveyor.com/api/projects/status/<account-key-string>"

tpaviot, 2015-12-22:
@thatcadguy The appveyor badge url fixed by commit a3aee81

thatcadguy, 2015-12-26:
Going to close this for now, on my to-do list to make a nightly/weekly server but likely won't get to it for months. For now, the build scripts for Travis CI and AppVeyor appear to be fixed.

closed 2015-12-26 by thatcadguy