tpaviot/oce #594: cross-section?

greysAcademicCode, 2015-12-23:
Can I use oce to cross-section a model and export some 2D vector art (dxf/svg) that contains the lines where the model's edges intersect with the cross-sectioning plane?

thatcadguy, 2015-12-23:
Yes, but you will have to write the code for the dxf/svg part.

jf---, 2015-12-23:
there's [some code here](https://github.com/tpaviot/pythonocc-core/commit/933e88249cab5a1661dccec580117ff552359fa0), that slices a model in parallel with pythonocc. that might be of interest

tpaviot, 2015-12-23:
AFAIK nobody achieved 2D to SVG export. If ever you get something, please let us know. FYI @charles-sharman did the opposite (SVG to 2D) see https://github.com/charles-sharman/ccad/blob/master/model.py

thatcadguy, 2015-12-24:
FreeCAD has become my unofficial OCE/OCCT example bible, they have code to export SVG and DXF. If you start out at https://github.com/FreeCAD/FreeCAD/tree/master/src/Mod/Drawing and dig through you will find code you can use.

thatcadguy, 2015-12-24:
P.S. Just mind the licenses if you take code from FreeCAD. Good luck.

@greysAcademicCode No problem asking this here, but for future reference, the mailing list (https://groups.google.com/forum/#!forum/oce-dev) is the better place to ask questions.

closed 2015-12-26 by thatcadguy