tpaviot/oce #595: Bounding box not correctly calculated for scaled shapes

thatcadguy, 2015-12-27:
Created a new issue for @tpaviot 's scaling issue here instead of cluttering up #547. I have reported the bug and given the test C++ program to OCCT: http://tracker.dev.opencascade.org/view.php?id=27042 so they can fix this. In the meantime, if we find a fix let's of course merge it and we can merge OCCT's fix later when they release it after a few versions.

tpaviot, 2015-12-27:
Thank you @thatcadguy for opening the issue and registering the ticket on the occt issue tracker. ok to fix asap.

thatcadguy, 2015-12-28:
Looks like the solution is to use gp_Trsf and BRepBuilderAPI_Transform instead of gp_GTrsf and BRepBuilderAPI_GTransform:

```
The same result we can obtain using the following draw script:

psphere s 5
puts [bounding s]
scalexyz r s 2 2 2
puts [bounding r]

The bounding boxes are the following:

-5.0000001000000003 -5.0000001000000003 -5.0000001000000003 5.0000001000000003 5.0000001000000003 5.0000001000000003
-20.000000199999995 -17.320508275688777 -10.000000200000001 10.000000200000001 17.320508275688766 10.000000200000001

This result is as expected. It is because the algorithm BRepBuilderAPI_GTransform generates b-spline geometries, and the bounding box on b-splines is constructed using its poles. The poles of this spheric b-spline are shown in the attached picture.

In order to keep the geometry form of the original shape you need to use isotropic transformation that is done with the classes gp_Trsf and BRepBuilderAPI_Transform. For that, the piece of transforming code should look like this:

    gp_Trsf scale_trsf;
    scale_trsf.SetScale(gp::Origin(), 2);
    TopoDS_Shape sp2 = BRepBuilderAPI_Transform(sp, scale_trsf).Shape();

Or, using draw script:

tscale s 0 0 0 2

After doing this, the bounding box is the following:

-10.000000200000001 -10.000000200000001 -10.000000200000001 10.000000200000001 10.000000200000001 10.000000200000001

And the program puts:

10, 10, 10
20, 20, 20

As you expected.
```

tpaviot, 2016-01-04:
@thatcadguy thank's for the hint. However this solution does not allow to scale the shape with different factors along x,y and z. For instance, it's not possible to apply this shape transformation to a sphere in order to get an ellipsoid.

thatcadguy, 2016-01-04:
@tpaviot Understood. I was simply relaying the reply from OCCT, they say this is as intended because of the bounding box being constructed using the b-spline poles.

tpaviot, 2016-01-04:
Yep, strange answer. The well known "it's not a bug it's a feature" ;)

thatcadguy, 2016-01-04:
Yes indeed. Strange that the poles would be included.

closed 2016-04-23 by thatcadguy