tpaviot/oce #598: please update some legacy VTK features

silentmissile, 2015-12-30:
vtkFloatingPointType was obsoleted, but I'm not sure from which version of VTK.

In OCCT-6.9.1 and OCE-master until today 2015-12-30, the definition is still in source tree, so if we compile it with VTK plugin, compiling error would happen.

http://vtk.1045678.n5.nabble.com/What-is-vtkFloatingPointType-td5717022.html
here is some temporary solution. If you like, you can study more to find a better solution

this issue exists in at least
src\IVtkVTK\IVtkVTK_View.cxx
inc\IVtkTools_ShapePicker.hxx
I didn't meet trouble in other files, but I can't promise it doesn't exist in other files.

during make process, build.cmake file in
adm\cmake\TKIVtk\CMakeFiles\TKIVtk.dir
adm\cmake\TKIVtkDraw\CMakeFiles\TKIVtkDraw.dir
link against VTK-6.1 libs only, it can't recognize VTK-6.3
please update it

thatcadguy, 2016-01-02:
This has already been reported to OCCT:
http://tracker.dev.opencascade.org/view.php?id=26511
In the meantime, is that the only thing that needs to be done? `#define vtkFloatingPointType double` basically?

thatcadguy, 2016-01-02:
@silentmissile Could you try the fix on my branch ja/vtkFloatingPointType-fix, or confirm that these changes are indeed correct?

silentmissile, 2016-01-03:
I tested it, here is the log

it seems that the VTK issue was resolved, but some other problem was imported.

[ 97%] Building RC object adm/cmake/TKXCAFSchema/CMakeFiles/TKXCAFSchema.dir/__/__/__/Version.rc.res
NMAKE : fatal error U1095: 扩展命令行“C:\PROGRA~2\WI3CF2~1\10\bin\x64\rc.exe  -DCSFDB -DHAVE_CONFIG_H -DHAVE_OCE_PATHS_H -DOCCT_DEBUG=1 -DTKXCAFSchema_EXPORTS -DWIN32 -DWNT -D_DEBUG -D_OCC64 -D_WINDOWS -D__MXCAFDoc_DLL -D__PXCAFDoc_DLL -D__XCAFDrivers_DLL -D__XCAFSchema_DLL -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\build\build_inc -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\inc -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\MXCAFDoc -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\MXCAFDoc -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\PXCAFDoc -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\PXCAFDoc -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\XCAFDrivers -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\XCAFDrivers -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\XCAFSchema -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\XCAFSchema -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\PDF -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\PDF -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\PDataStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\PDataStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\PFunction -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\PFunction -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\PDocStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\PDocStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\MDF -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\MDF -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\MDataStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\MDataStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\MFunction -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\MFunction -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\MDocStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\MDocStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\CDM -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\CDM -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\PCDM -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\PCDM -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\CDF -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\CDF -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\UTL -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\UTL -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\LDOM -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\FSD -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\FSD -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\MMgt -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\MMgt -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\OSD -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\OSD -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\Plugin -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\Plugin -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\Quantity -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\Quantity -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\Resource -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\Resource -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\SortTools -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\SortTools -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\Standard -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\Standard -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\StdFail -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\StdFail -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\Storage -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\Storage -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\TColStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\TColStd -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\TCollection -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\TCollection -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\TShort -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\TShort -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\Units -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\Units -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\UnitsAPI -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\UnitsAPI -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\IncludeLibrary -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\Dico -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\Dico -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\NCollection -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\Message -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\Message -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\DBC -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\drv\DBC -II:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\src\PCollection -II:\Source\OpenCAS [... cut ...] a-vtkFloatingPointType-fix\drv\StdDrivers  /DWIN32    /D_CRT_SECURE_NO_WARNINGS /D_CRT_NONSTDC_NO_WARNINGS /foCMakeFiles\TKXCAFSchema.dir\__\__\__\Version.rc.res I:\Source\OpenCASCADE\oce-ja-vtkFloatingPointType-fix\build\Version.rc” too long
Stop.
NMAKE : fatal error U1077: “"D:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"”: return “0x2”
Stop.
NMAKE : fatal error U1077: “"D:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"”: return “0x2”
Stop.

thatcadguy, 2016-01-03:
That's normal, as it says at the end, path names are too long. Try with a
shorter (higher-level) folder, like I:\oce or something.

silentmissile, 2016-01-04:
yes, as your suggestion, if I put it in a shorter path, everything goes well.

This issue could be closed, if you want to wait OCCT to update then close it, that's also ok

closed 2016-03-16 by thatcadguy