tpaviot/oce #602: IntCurveSurface_Polygon::Init may thow unhandled exception

mikedstrong, 2016-01-12:
When initializing and IntCurveSurface_Polygon, a deflection is calculated.  If two consecutive points are coincident, and unhandled Standard_ConstructionError exception is thrown when the the gp_Lin is created using a zero-length gp_Dir in line 109 of IntCurveSurface_Polygon.gxx.  I believe the correct thing to do is to skip zero-length segments, so I placed a try block around this code as follows:

    do {
      gp_Pnt Pm=TheCurveTool::Value(C,u);
      gp_Pnt P1=ThePnts.Value(i);
      gp_Pnt P2=ThePnts.Value(i+1);
      // ignore this segment if zero-length (throws exception)
      try {
        gp_Lin L(P1,gp_Dir(gp_Vec(P1,P2)));
        Standard_Real t=L.Distance(Pm);
      
        if(t>TheDeflection) {
                  TheDeflection = t;
        }
      }
      catch(Standard_ConstructionError) {
      }
      u+=du;
      i++;
    }
    while(i<NbPntIn);

thatcadguy, 2016-01-12:
Instead of using exception handling, I think this would be better handled by checking to see if the distance is practically zero before constructing the line, i.e.
```
if (P1.Distance(P2) < Precision::Confusion())
    // do something
```

thatcadguy, 2016-01-12:
Could also use
```
if (P1.IsEqual(P2, Precision::Confusion()))
    // do something
```

mikedstrong, 2016-01-12:
That's certainly true.  However, I think it is important that the tolerance has the same value that is used to throw the exception (gp::Resolution()).  If Precision::Confusion() is greater that gp::Resolution, the exception could still get thrown.  For reference the code in gp_Dir.lxx is:

  Standard_Real D = sqrt(X * X + Y * Y + Z * Z);
  Standard_ConstructionError_Raise_if (D <= gp::Resolution(), "");

thatcadguy, 2016-01-12:
Agreed, I was just wondering what the more appropriate tolerance would be.

So something like:
```
    do {
      gp_Pnt Pm=TheCurveTool::Value(C,u);
      gp_Pnt P1=ThePnts.Value(i);
      gp_Pnt P2=ThePnts.Value(i+1);
      if (P1.Distance(P2) > gp::Resolution()) {
        gp_Lin L(P1,gp_Dir(gp_Vec(P1,P2)));
        Standard_Real t=L.Distance(Pm);
      
        if(t>TheDeflection) {
          TheDeflection = t;
        }
      }
      u+=du;
      i++;
    }
    while(i<NbPntIn);
```

thatcadguy, 2016-01-12:
Try https://github.com/tpaviot/oce/tree/ja/IntCurveSurface_Polygon-fix

mikedstrong, 2016-01-13:
Thanks, Jacob.  Looks good.  Mike

closed 2016-03-16 by thatcadguy