tpaviot/oce #605: OCE 0.16.1 Tests segfaulting with GCC 6.0.0

hobbes1069, 2016-02-03:
I'm assuming this is a GCC 6.0 issue since this wasn't happening before. 

Here's the list of failed tests:
```
19:IGESImportTestSuite.testImportIGES_1
22:STEPImportTestSuite.testImportAP203_1
23:STEPImportTestSuite.testImportAP203_2
24:STEPImportTestSuite.testImportAP214_1
25:STEPImportTestSuite.testImportAP214_2
26:STEPImportTestSuite.testImportAP214_3
29:BRepOffsetAPITestSuite.testEvolvedShape
30:BRepOffsetAPITestSuite.testOffsetBox
31:BRepImportTestSuite.testImportBRep1
32:BRepImportTestSuite.testImportBRep2
33:BRepImportTestSuite.testImportBRep3
34:BRepImportTestSuite.testImportBRep4
37:BRepMeshTestSuite.testMeshTorus
```
The full testlog is here:
https://dl.dropboxusercontent.com/u/34775202/LastTest.log

And the stack trace of one of the tests:
https://dl.dropboxusercontent.com/u/34775202/BRepImport_test_gdb.txt

thatcadguy, 2016-02-03:
Is GCC 6 even considered stable yet?

hobbes1069, 2016-02-03:
I'd have to ask but Fedora Rawhide (future 24 version) is rebuilding all packages against it.

hobbes1069, 2016-02-16:
I tried adding "-fno-delete-null-pointer-checks" to the test binaries per https://github.com/google/googletest/issues/705 but no luck, but I only added it to the binaries, not the test library.

hobbes1069, 2016-02-17:
Adding "-fno-delete-null-pointer-checks" as a global CXXFLAG worked but it's still just a work around.

tpaviot, 2016-02-17:
@hobbes1069 it might be related to issue #604 and #576 . clang used to perform optimizations that makes oce segfault, maybe it's the same with recent gcc versions. Don't know if it's possible to cherry-pick commit 61cb965b9ffeca419005bc15e635e67589c421dd that fixes this issue over oce-0.17 but i's worth trying.

hobbes1069, 2016-02-17:
It applied cleanly to 0.17 so I'm doing a test build however Fedora is still stuck on 0.16.1 until smesh is updated. Hopefully it applies cleanly to it as well. Thanks for the pointer.

hobbes1069, 2016-02-17:
On a side note, does this fix create an ABI change? If so I'll need to rebuild all dependencies.

hobbes1069, 2016-02-17:
I went ahead and did a new build. Everything seems OK but the ARM build is still going, but that's to be expected.

closed 2016-02-17 by hobbes1069