tpaviot/oce #608: CMAKE fail related to pthread.

yokoyamy, 2016-02-19:
Hi.

I tried to build OCE-0.17

When I generate Makefile by `cmake -G "MSYS Makefiles"` and got following messages.

```
CMakeFiles/cmTC_a8353.dir/objects.a(CheckSymbolExists.c.obj):CheckSymbolExists.c
:(.rdata$.refptr.pthread_create[.refptr.pthread_create]+0x0): undefined referenc
e to `pthread_create'^M
collect2.exe: error: ld returned 1 exit status^M
```

```
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/../../../../x86_64-w64-mingw3
2/bin/ld.exe: cannot find -lpthreads^M
collect2.exe: error: ld returned 1 exit status^M

```
I have pthread in `C:\mingw64`

Does anyone know how to tell `CMAKE` where is pthrad?

thatcadguy, 2016-03-16:
When you install the MSYS version of MinGW, you need to use the MSYS command prompt that sets all the paths up correctly, have you tried that?

thatcadguy, 2016-04-23:
@yokoyamy Is this still an issue?

yokoyamy, 2016-04-24:
I haven't solved yet.I've been trying to change latest version of MinGW64 but I couldn't.

However I've been able to build by MinGW32  on another PC.

yokoyamy, 2016-07-21:
This problem doesn't happen so far.
I'm not sure why. But I changed PC .

closed 2016-07-21 by yokoyamy