tpaviot/oce #610: set RPATH or install name for all installed libs?

davydden, 2016-03-27:
i would like to install OCE on Darwin in custom location and use it without setting DYLD_LIBRARY_PATH. My understanding is that one of these solutions would do it:
(i) all libs have install names as full path: (/path/to/oce/lib/libABC.dylib) instead of libABC.dylib
(ii) or there is RPATH set.

Is there any `-DOCE_XYZ` CMake flag that would help to achieve this?

jf---, 2016-03-27:
hi @davydden ,

IIRC there's a difference in [how .dylib's are built]() with cmake 3.x.x and 2.8.x
Are you on a recent version of cmake?

davydden, 2016-03-27:
@jf--- yes, I am on 3.xx.

davydden, 2016-03-31:
nevermind, i ended up using a python script with `install_name` tool on OSX.

closed 2016-03-31 by davydden