tpaviot/oce #611: [0.17.1] More issues with NULL pointers and OSX Clang@7.0.2 and failing tests

davydden, 2016-03-31:
It seems this PR https://github.com/tpaviot/oce/pull/604 did not fix all the issues. I applied this patch to 0.17.1 and have the following warnings:
```
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/Standard/Standard_ErrorHandler.cxx:143:40: warning: comparison of array 'anActive->myLabel' not equal to a null pointer is always true [-Wtautological-pointer-compare]
  return anActive != NULL && anActive->myLabel != NULL;
                             ~~~~~~~~~~^~~~~~~    ~~~~
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/Standard/Standard_ErrorHandler.cxx:157:36: warning: comparison of array 'anActive->myLabel' equal to a null pointer is always false [-Wtautological-pointer-compare]
  if(anActive == NULL || anActive->myLabel == NULL) {
                         ~~~~~~~~~~^~~~~~~    ~~~~
```
```
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/LDOM/LDOM_CharacterData.cxx:62:8: warning: reference cannot be bound to dereferenced null pointer in well-defined C++ code; comparison may be assumed to always evaluate to false [-Wtautological-undefined-compare]
  if (&aText == NULL)
       ^~~~~    ~~~~
```

Additionally i have a couple of others warnings, related to `PLUGINFACTORY`:
```
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/FWOSDriver/FWOSDriver.cxx:26:1: warning: 'PLUGINFACTORY' has C-linkage specified, but returns user-defined type 'Handle_Standard_Transient' which is incompatible with C [-Wreturn-type-c-linkage]
PLUGIN(FWOSDriver)
^
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/Plugin/Plugin_Macro.hxx:21:56: note: expanded from macro 'PLUGIN'
extern "C" {Standard_EXPORT Handle(Standard_Transient) PLUGINFACTORY(const Standard_GUID&);} \
            
```
```
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/BinLDrivers/BinLDrivers.cxx:95:1: warning: 'PLUGINFACTORY' has C-linkage specified, but returns user-defined type 'Handle_Standard_Transient' which is incompatible with C [-Wreturn-type-c-linkage]
PLUGIN(BinLDrivers)
^
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/Plugin/Plugin_Macro.hxx:21:56: note: expanded from macro 'PLUGIN'
extern "C" {Standard_EXPORT Handle(Standard_Transient) PLUGINFACTORY(const Standard_GUID&);} \
```
```
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/XmlLDrivers/XmlLDrivers.cxx:129:1: warning: 'PLUGINFACTORY' has C-linkage specified, but returns user-defined type 'Handle_Standard_Transient' which is incompatible with C [-Wreturn-type-c-linkage]
PLUGIN(XmlLDrivers)
^
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/Plugin/Plugin_Macro.hxx:21:56: note: expanded from macro 'PLUGIN'
extern "C" {Standard_EXPORT Handle(Standard_Transient) PLUGINFACTORY(const Standard_GUID&);} \
                
```
```
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/StdLDrivers/StdLDrivers.cxx:71:1: warning: 'PLUGINFACTORY' has C-linkage specified, but returns user-defined type 'Handle_Standard_Transient' which is incompatible with C [-Wreturn-type-c-linkage]
PLUGIN(StdLDrivers)
^
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/Plugin/Plugin_Macro.hxx:21:56: note: expanded from macro 'PLUGIN'
extern "C" {Standard_EXPORT Handle(Standard_Transient) PLUGINFACTORY(const Standard_GUID&);} \
           
```

Most of the tests fail:
```
Test project /private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1
      Start  1: BRepAlgoAPITestSuite.testCutBox
 1/44 Test  #1: BRepAlgoAPITestSuite.testCutBox .................***Exception: Other  0.01 sec
      Start  2: BRepAlgoAPITestSuite.testCutCylSphere
 2/44 Test  #2: BRepAlgoAPITestSuite.testCutCylSphere ...........***Exception: Other  0.01 sec
      Start  3: BRepExtremaTestSuite.testDistSphereSphere
 3/44 Test  #3: BRepExtremaTestSuite.testDistSphereSphere .......***Exception: Other  0.01 sec
      Start  4: BRepExtremaTestSuite.testDistSphereVertex
 4/44 Test  #4: BRepExtremaTestSuite.testDistSphereVertex .......***Exception: Other  0.01 sec
      Start  5: BRepExtremaTestSuite.testDistVertexEdge
 5/44 Test  #5: BRepExtremaTestSuite.testDistVertexEdge .........***Exception: Other  0.01 sec
      Start  6: BRepExtremaTestSuite.testDistEdgeVertex
 6/44 Test  #6: BRepExtremaTestSuite.testDistEdgeVertex .........***Exception: Other  0.01 sec
      Start  7: BRepExtremaTestSuite.testDistVertexFace
 7/44 Test  #7: BRepExtremaTestSuite.testDistVertexFace .........***Exception: Other  0.01 sec
      Start  8: BRepExtremaTestSuite.testDistFaceVertex
 8/44 Test  #8: BRepExtremaTestSuite.testDistFaceVertex .........***Exception: Other  0.01 sec
      Start  9: BRepGPropTestSuite.testComputeBoxVolume
 9/44 Test  #9: BRepGPropTestSuite.testComputeBoxVolume .........***Exception: Other  0.01 sec
      Start 10: BRepGPropTestSuite.testComputeBoxSurface
[... 45 lines cut ...]
      Start 33: STEPImportTestSuite.testImportAP203_2
33/44 Test #33: STEPImportTestSuite.testImportAP203_2 ...........***Exception: Other  0.01 sec
      Start 34: STEPImportTestSuite.testImportAP214_1
34/44 Test #34: STEPImportTestSuite.testImportAP214_1 ...........***Exception: Other  0.01 sec
      Start 35: STEPImportTestSuite.testImportAP214_2
35/44 Test #35: STEPImportTestSuite.testImportAP214_2 ...........***Exception: Other  0.02 sec
      Start 36: STEPImportTestSuite.testImportAP214_3
36/44 Test #36: STEPImportTestSuite.testImportAP214_3 ...........***Exception: Other  0.01 sec
      Start 37: TCollectionTestSuite.testSequenceOfPnt
37/44 Test #37: TCollectionTestSuite.testSequenceOfPnt ..........***Exception: Other  0.01 sec
      Start 38: UnitsAPITestSuite.testCurrentUnits
38/44 Test #38: UnitsAPITestSuite.testCurrentUnits ..............***Exception: Other  0.00 sec
      Start 39: UnitsAPITestSuite.testMDTVCurrentUnits
39/44 Test #39: UnitsAPITestSuite.testMDTVCurrentUnits ..........***Exception: Other  0.00 sec
      Start 40: TestSuite.testNullPointer
40/44 Test #40: TestSuite.testNullPointer .......................   Passed    0.01 sec
      Start 41: TestSuite.testFloatEq
41/44 Test #41: TestSuite.testFloatEq ...........................   Passed    0.00 sec
      Start 42: TestSuite.testFloatNeq
42/44 Test #42: TestSuite.testFloatNeq ..........................   Passed    0.01 sec
      Start 43: TestSuite.testBoolean
43/44 Test #43: TestSuite.testBoolean ...........................   Passed    0.01 sec
      Start 44: TestSuite.testIntegerLighter
44/44 Test #44: TestSuite.testIntegerLighter ....................   Passed    0.01 sec
```

thatcadguy, 2016-03-31:
The bit about PLUGINFACTORY has been a warning for a long time, OCCT is working on removing those by 7.0.0 or 7.1.0 I believe.

rainman110, 2016-05-24:
The only warning that should be taken serious is the second one:
```
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/src/LDOM/LDOM_CharacterData.cxx:62:8: warning: reference cannot be bound to dereferenced null pointer in well-defined C++ code; comparison may be assumed to always evaluate to false [-Wtautological-undefined-compare]
  if (&aText == NULL)
       ^~~~~    ~~~~
```

This is the same issue as in #604 and we should fix it.

thatcadguy, 2016-07-26:
This is fixed in OCCT master, a.k.a. as soon as we get up-to-date here this will get fixed for OCE.

closed 2016-07-26 by thatcadguy