tpaviot/oce #612: Unit tests can't find just compiled libraries

davydden, 2016-03-31:
subj.
```
1/44 Testing: BRepAlgoAPITestSuite.testCutBox
1/44 Test: BRepAlgoAPITestSuite.testCutBox
Command: "/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/test/BRepAlgoAPI_test/boolean_ops_test" "--gtest_filter=BRepAlgoAPITestSuite.testCutBox"
Directory: /private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/test/BRepAlgoAPI_test
"BRepAlgoAPITestSuite.testCutBox" start time: Mar 31 10:15 CEST
Output:
----------------------------------------------------------
dyld: Library not loaded: libTKernel.10.dylib
  Referenced from: /private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/test/BRepAlgoAPI_test/boolean_ops_test
  Reason: image not found
<end of output>
Test time =   0.01 sec
```
One could of course hack a `DYLD_LIBRARY_PATH` to make it work, but ideally unit tests should work without.

davydden, 2016-03-31:
Interestingly enough, most of the libs are linked correctly in that particular unit test:
```
$ otool -L boolean_ops_test
boolean_ops_test:
	/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/Darwin/x86_64-Release-64/libTKBO.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/Darwin/x86_64-Release-64/libTKPrim.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/Darwin/x86_64-Release-64/libTKTopAlgo.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/Darwin/x86_64-Release-64/libTKGeomAlgo.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/Darwin/x86_64-Release-64/libTKBRep.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/Darwin/x86_64-Release-64/libTKGeomBase.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/Darwin/x86_64-Release-64/libTKG3d.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/Darwin/x86_64-Release-64/libTKG2d.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-dRo5ln/oce-OCE-0.17.1/Darwin/x86_64-Release-64/libTKMath.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	libTKernel.10.dylib (compatibility version 10.0.0, current version 10.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	@rpath/libtbbmalloc.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libtbb.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
```
and it's only `libTKernel.10.dylib` which isn't and gives troubles...

davydden, 2016-03-31:
i should also say that I configure with
```
-DCMAKE_INSTALL_PREFIX=MYPREFIX
-DCMAKE_INSTALL_RPATH_USE_LINK_PATH=FALSE 
-DCMAKE_INSTALL_RPATH=MYPREFIX/lib
```

rainman110, 2016-05-24:
@davydden This is normal. On OS X, cmake does not set any rpath to the produced binaries.

To run the tests, you have to set the DYLD_LIBRARY_PATH environment variable to the directory of the shared libraries (i.e. libTKernel.dylib)

davydden, 2016-05-24:
`DYLD_LIBRARY_PATH` is not the solution as it does not work on modern OSX (El Capitan) unless System Integrity Protection is disabled. I hope there is another way.

rainman110, 2016-05-24:
Interesting, I didn't know that. On my hacktintosh, SIP is always disabled :smirk: 

Try setting the following (not tested):
```
-DCMAKE_SKIP_BUILD_RPATH=FALSE
-DCMAKE_BUILD_WITH_INSTALL_RPATH=FALSE
```

In theory this should add the build rpaths into the testing executable.

rainman110, 2016-05-24:
You might even need the following variable:
-DCMAKE_MACOSX_RPATH=ON

davydden, 2016-07-06:
sorry for a late reply, i fixed the issue by setting `-DCMAKE_INSTALL_NAME_DIR:PATH=` to point to the target's lib.

closed 2016-07-06 by davydden