tpaviot/oce #615: Scaled cylinder yields invalid shape

aothms, 2016-05-01:
Hi all,

Please find attached a trivial example program that yields an invalid shape by using `TopoDS_Shape::Move()` on a shape with a `Geom_CylindricalSurface` and a `gp_Trsf` with scale factor.

~~~
#include <gp_Trsf.hxx>
#include <Geom_Circle.hxx>

#include <BRepBuilderAPI_MakeWire.hxx>
#include <BRepBuilderAPI_MakeEdge.hxx>
#include <BRepBuilderAPI_MakeFace.hxx>
#include <BRepBuilderAPI_Transform.hxx>

#include <BRepPrimAPI_MakePrism.hxx>

#include <TopoDS_Iterator.hxx>

#include <BRepCheck.hxx>
#include <BRepCheck_Result.hxx>
#include <BRepCheck_Analyzer.hxx>
#include <BRepCheck_ListOfStatus.hxx>
#include <BRepCheck_ListIteratorOfListOfStatus.hxx>

#include <BRepTools.hxx>

int main(int argc, char** argv) {
	Handle(Geom_Circle) circle = new Geom_Circle(gp_Ax2(), 400.);
	BRepBuilderAPI_MakeWire wire;
	wire.Add(BRepBuilderAPI_MakeEdge(circle));
	TopoDS_Shape cylinder = BRepPrimAPI_MakePrism(BRepBuilderAPI_MakeFace(wire, false), gp_Vec(0, 0, 4000)).Shape();
	
	{
		BRepCheck_Analyzer valid(cylinder);
		bool validity = valid.IsValid();
		std::cout << validity << std::endl;
	}
	
	gp_Trsf trsf;
	trsf.SetScaleFactor(0.1);
	TopoDS_Shape scaled_cylinder = cylinder.Moved(trsf);

	{
		BRepCheck_Analyzer valid(scaled_cylinder);
		bool validity = valid.IsValid(); 
		std::cout << validity << std::endl;
	}

	scaled_cylinder = BRepBuilderAPI_Transform(cylinder, trsf);

	{
		BRepCheck_Analyzer valid(scaled_cylinder);
		bool validity = valid.IsValid();
		std::cout << validity << std::endl;
	}
}
~~~

Results in:
~~~
1
0
1
~~~

(in other words, the unscaled cylinder is valid, the scaled cylinder with `TopoDS_Shape::Move()` is not, `BRepBuilderAPI_Transform` does yield a valid shape on the same operands)

I did some debugging and it boils down to

1. a `BRepCheck_UnorientableShape` being issued for the 'cylindrical' `TopoDS_Face`
2. This is due to a `BRepCheck_InvalidCurveOnSurface` being issued for one of its edges
https://github.com/tpaviot/oce/blob/c0cdb53fdf7cc708eb303bec49b18e27edd7c74b/src/BRepCheck/BRepCheck_Analyzer.cxx#L257
3. This is due to the distance between points on the pcurve and 3d curve being a factor 10 away from each other in the Z-axis. Which seems to indicate the somehow the transformation does get applied to only one of them.
https://github.com/tpaviot/oce/blob/c0cdb53fdf7cc708eb303bec49b18e27edd7c74b/src/BRepCheck/BRepCheck_Edge.cxx#L718

I'm in way over my head here. Appreciate some help. Thanks all.

a-betenev, 2016-05-01:
Hello Thomas,

Unfortunately, transformations with non-unit scale factor are not 
supported in shape locations.
The reason is that elementary surfaces and curves (such as line or 
cylinder) in OCCT are parametrized 'by length'.
When you apply scale factor, parametrization changes, and all the data 
that refer to it need to be recomputed (such as p-curves on surfaces).
Though this potentially could be accounted for, it is not done in 
practice (would be overly complex).
Thus, just do not use non-unit scales in this way (use Transform to get 
new shape instead).

Andrey

aothms, 2016-05-01:
Dear Andrey,

Thanks for clarifying this. I wasn't aware of this, although admittedly the fact that the method is called `Move()` should have made me think longer. Maybe for future ignorant users like me it might be good to throw an exception when a location with `gp_Trsf::ScaleFactor() != 1.` is encountered, but I'm sure you have your reasons. Thanks again!

Kind regards,
Thomas

a-betenev, 2016-05-01:
Dear Thomas,

I agree, now recorded as issue in OCCT Mantis: 
http://tracker.dev.opencascade.org/view.php?id=27457

Best Regards,
Andrey

aothms, 2016-05-01:
Dear Andrey,

Great! Thanks.

Kind regards,
Thomas

closed 2016-05-01 by aothms