tpaviot/oce #619: OCE only builds if placed under a short absolute path (windows)

viric, 2016-05-18:
I have had troubles building OCE in VC11 (Visual Studio 2012) with msbuild because of too long argument list (lots of /I+absolutepath). It is not that the compiler reported that, but the compiler crashes reliably in that situation, and does not crash if I place it under shorter paths.

Specifically, the error I see is cl.exe dying because of STATUS_STACK_BUFFER_OVERRUN=-1073740791=0xC0000409 
```
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V110\Microsoft.CppCommon.targets(249,5): error MSB6006: "CL.exe" exited with code -1073740791. [C:\cygwin64\tmp\nix-build-oce-0.17.1.drv-1\oce-OCE-0.17.1\build\adm\cmake\TKXDEIGES\TKXDEIGES.vcxproj]
```

Notice that I'm building in this path: `C:\cygwin64\tmp\nix-build-oce-0.17.1.drv-1\oce-OCE-0.17.1`, subdirectory `build`.

If I shorten that to: `C:\cygwin64\tmp\nix-build-o.drv-1\o` (with the same `build` cmake subdirectory`) it builds fine.

I'm using a build system that automatically places things under at least `C:\cygwin64\tmp\nix-build-x.drv-0`, so this is an annoying situation for me.

Isn't there any way to make the build recipes less dependant on absolute path length? Maybe using a single include directory or so.

a-betenev, 2016-05-19:
Hello,

I suggest you try OCCT 7.0: it uses much less number of include dirs 
than previous versions, and thus shall work with much longer paths.
I have just tried building it in the path 130 symbols long and it seems 
to build fine (even if CMake warns on path being too long for VS 2010 
projects).

Andrey

viric, 2016-05-20:
I checked OCCT 7.0... as a start, it seems not to allow a freeimage.lib static (without dll), and I can't see any option to produce shared/static libs. I will test it more, though.

thatcadguy, 2016-06-12:
This has been a longstanding issue, not with OCE or MSVC, but a limitation of Windows' command prompt itself. There was recently an update to Windows 10 that changed this limitation. Otherwise, nothing to be done, just how compilation on Windows is.

closed 2016-06-12 by thatcadguy