tpaviot/oce #621: Crash/Illegal instruction with Clang on OS X

rainman110, 2016-05-24:
In release mode, Clang creates illegal code in ```AdvApp2Var_ApproxF2var::mma2fnc_```

This is a meta issue containing the information from
 - https://github.com/tpaviot/pythonocc-core/issues/273
 - http://tracker.dev.opencascade.org/view.php?id=26778

To reproduce:
```cpp
   std::ifstream file("fillet_crash.brep");
   TopoDS_Shape shape;
    BRep_Builder b;
    BRepTools::Read(shape, file, b);
    for (TopExp_Explorer exp(shape, TopAbs_EDGE); exp.More(); exp.Next())
    {
        try
        {
            BRepFilletAPI_MakeFillet filletMaker(shape);
            filletMaker.Add(0.165405, TopoDS::Edge(exp.Current()));
            filletMaker.Build();
        }
        catch (...)
        {
        }
    }
```

The file fillet_crash.brep was downloaded from the occt mantis: [fillet_crash.zip](https://github.com/tpaviot/oce/files/279598/fillet_crash.zip)

Possible fixes:
 - Compile ```src/AdvApp2Var/AdvApp2Var_ApproxF2var.cxx``` only with optimization level -O1
 - or add ```__attribute__ ((noinline))``` to the static functions of ```AdvApp2Var_ApproxF2var``` as suggested by Roman Lygin
 - use the newest Clang (not tested yet) or a Clang older than 3.4

thatcadguy, 2016-06-12:
Repeat of #520, but we'll leave it open for the time being anyway. #520 was put over to OCCT due to the size of the issue.

thatcadguy, 2016-07-26:
This is fixed in OCCT master, closing.

closed 2016-07-26 by thatcadguy