tpaviot/oce #622: Problems when building OCE

zack11235, 2016-05-25:
Hello,
When trying to build OCE on Windows as described here (https://github.com/tpaviot/oce/wiki/Build). First cmake requires Freetype library, so I downloaded the binaries online from http://gnuwin32.sourceforge.net/packages/freetype.htm and used freetype-binaries\include for FREETYPE_INCLUDE_DIRECTORY_ft2build and freetype-binaries\include\freetype2 for FREETYPE_INCLUDE_DIRECTORY_freetype2 and also \freetype-binaries\lib\freetype.lib for FREETYPE_LIBRARY and unchecked FREETYPE_LIBRARY_DEBUG.  When cmake finishes generating I was not able to find the OCE-libraries.cmake and OCE-libraries-release.cmake files that I will need later on when build OccJava.  Then went on as suggested in the link and build the OCE.sln with my selected generator (Visual Studio 10 Win 64 - Release). 
After a long build I get errors related to FreeTypeLibrary!

Regards,
Mahmoud Zaki

zack11235, 2016-05-25:
in a nutshell I have two problems:
- I do not have OCE-libraries.cmake and OCE-libraries-release.cmake files in the generated code.
- I do not know what do to do with freetype libraries!

thatcadguy, 2016-05-25:
Don't need to double post on this and the group, look up oce-win-bundle, it's all you need.

closed 2016-05-25 by thatcadguy

zack11235, 2016-05-25:
Sorry I posted here because I thought there is not the right place to post. 

Thank you!