tpaviot/oce #623: MSYS2 MinGW64 Compilation error

Fredounet, 2016-05-28:
Hi,

I used not to experience this issue. Using the cmake toolchain on MSYS2 MinGW64 shell on Windows, I previously managed to compile OCE. After a few months, I'm back at it but it now fails. The error happens at the linking stage of TKernel (the first library). I get:

/usr/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: syntax error: unexpected end of file
adm/cmake/TKernel/CMakeFiles/TKernel.dir/build.make:10077: recipe for target 'win64/bin/Release/libTKernel.dll' failed
mingw32-make[2]: *** [win64/bin/Release/libTKernel.dll] Error 1
CMakeFiles/Makefile2:1013: recipe for target 'adm/cmake/TKernel/CMakeFiles/TKernel.dir/all' failed
mingw32-make[1]: *** [adm/cmake/TKernel/CMakeFiles/TKernel.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

This seems to be having something to do with a missing ' " ', but I really cannot figure out where. Or could it be a wrong message, directing me in the wrong direction?

Fredounet, 2016-05-28:
OK. Got it. I was using "mingw32-make" instead of "make" for the make command. 
But, I'm honestly completely confused, because my CMakeLists file includes:

`CMAKE_MAKE_PROGRAM:FILEPATH=C:/msys64/mingw64/bin/mingw32-make.exe`
instead of:
`CMAKE_MAKE_PROGRAM:FILEPATH=C:/msys64/usr/bin/make.exe`

I feel I'm getting something wrong here...

Fredounet, 2016-07-10:
I got that sorted. It should be:
`CMAKE_MAKE_PROGRAM:FILEPATH=C:/msys64/usr/bin/make.exe`

Fredounet, 2016-07-10:
But, now if I use Gl2ps and Freeimage, I get the error that I reported in #630 .

tpaviot, 2016-07-10:
@Fredounet Ok, I guess you figured out how to deal with mingw-64. Can u please share your experience (you can for instance post to pythonocc-users or open an issue here) ? Especially, it would be good to know about some tricky points that are not obvious.

nickoe, 2016-07-15:
@tpaviot  I guess this issue here can be closed, it seems that @Fredounet tried to describe it in https://github.com/tpaviot/oce/issues/631

closed 2016-07-15 by Fredounet