tpaviot/oce #626: Test installation: OCE Draw Test Harness

ivanoterouvigo, 2016-06-15:
How to test this step.

[image]
[image]
[image]

tpaviot, 2016-06-15:
Did you make/make install oce ? DRAWEXE is a binary that is installed to the path you provided

ivanoterouvigo, 2016-06-15:
Hi Thomas!

I follow the steps of INSTALL.Unix of oce-OCE-0.16 version.

Then I follow steps:

 mkdir build
 cd build

cmake ..

I reviewed definition of flags:

All are ok.
Then I modify like:
----------------------------------------------------

[image]

-----------------------------

[image]

----------------------------------------------------

Then I type:

cmake $flags ..

Console show this error: ....

[image]

Sorry but I don´t understand this error.

tpaviot, 2016-06-15:
@ivanoterouvigo tcl/tk must be installed

You can download/install all dependencies on debian
``` sudo apt-get install tcl8.5-dev tk8.5-dev libgl2ps-dev libfreeimage-dev libtbb-dev```

If you want to run pythonocc, the best is certainly to download/install binaries using conda. It's only 5 minutes to get a full installation ready to use

ivanoterouvigo, 2016-06-15:
I compiled oce-OCE-0.16 and I after compiled pythonocc.

It pass test ok 
Then I try test examples of examples directory.

[image]

I am installing PySide but it require to many time.

My computer has 8 MB RAM and 4 CPU´s

ivanoterouvigo, 2016-06-15:
Well I install PySide and tested again.

It work fine.