tpaviot/oce #630: FreeType, FreeImage, Gl2ps and CMAKE_CONFIGURATION_OPTIONS

Fredounet, 2016-07-08:
I'm recompiling OCE on MSYS2 (Mingw-64b) but encountered some strange error with Cmake.
When I added the checked options, Gl2ps and FreeImage then `configure` failed with:

`get_filename_component has incorrect number of arguments.`

This seems to be about the Debug paths. Looking at FreeImage, a if statement is used that tests `CMAKE_CONFIGURATION_OPTIONS`. Putting the same if statement for Gl2ps and Freeimage seems to solve the problem (well... `make` is only 2% complete as I write this). 

But, looking at this in more detail, I couldn't find `CMAKE_CONFIGURATION_OPTIONS` defined anywhere before the first line where it is tested. Wasn't maybe `CMAKE_CONFIGURATION_TYPES` meant to be used instead?

Fredounet, 2016-07-10:
Ok. Forget about the error. I had to do a "delete cache" before recompiling with the flags Freeimage and Gl2ps on. 
Nonetheless, I still find the `CMAKE_CONFIGURATION_OPTIONS` flag rather odd.

tpaviot, 2016-07-10:
@Fredounet does it work now ? if yes, can you please close the issue ?

closed 2016-07-10 by Fredounet

ArvinShaffer, 2021-06-30:
i get the same error, can you tell me how to solve it