tpaviot/oce #631: Compiling with MSYS2 MinGW64

Fredounet, 2016-07-10:
**NOTE: This may not be complete, but almost. I'll remove this message once I'm done.**

## MSYS2:
Install MSYS2 by following the procedure at: http://msys2.github.io/

## QT5:
Open MSYS2.exe and install qt5: 
`pacman -Sy mingw-w64-x86_64-qt5`

## CMAKE:
`pacman -Sy cmake` (this includes cmake-gui)

## OCE:
Open MSYS2 and install: git (sys/git), make (sys/make), toolchain (includes gcc), 
`pacman -Sy git`
`pacman -Sy make`
`pacman -Sy ming-w64-x86_64-toolchain`

Continue by installing: boost, tcl, tk, freeimage, libgl2ps, opencollada, swig, freetype, libpng
`pacman -Sy mingw-w64-x86_64-boost`
`pacman -Sy mingw-w64-x86_64-tcl`
`pacman -Sy mingw-w64-x86_64-tk`
`pacman -Sy mingw-w64-x86_64-freeimage`
`pacman -Sy mingw-w64-x86_64-libgl2ps`
`pacman -Sy mingw-w64-x86_64-opencollada-git`
`pacman -Sy mingw-w64-x86_64-swig`
`pacman -Sy mingw-w64-x86_64-freetype` (actually, this is already installed with qt5, I believe)
`pacman -Sy mingw-w64-x86_64-libpng` (actually, this is already installed with qt5, I believe)
Close MSYS2.

### Cmake
Open `mingw64.exe`, follow the "Building" section at https://github.com/tpaviot/oce/wiki/Build-%28MinGW-w64%29. 
**BUT**, you don't need to use the _Bundle_ as all those libraries are already available under MSYS2.
However, you must add to `CMAKE_NEED_RESPONSE=YES` to Cmake before launching the _Configure_. If you correct this after the error has occured (it's a "linking error, argument list too long" that happens after approx. 40% of the compilation ), then compilation will restart from scratch. So, I recommend adding this upfront, and not wait for the error. Otherwise, you'll have wasted a few hours.
When you _Configure_, use `MSYS Makefiles` and set `CMAKE_MAKE_PROGRAM` to `mingw64/usr/bin/make.exe`. The rest should be straight-forward.
Once _Configure_ is successful, run _Generate_ and finally close Cmake-gui.

### Make
Run `make` (inside the _build_ folder you created in the cmake process).
Remember that it's a good idea to use `make -j4` (or more, if you have). 

As usual, compilation takes forever, but it goes smoothly, only 1 test failed at the very end (no big deal).

Stinkfist0, 2016-07-10:
FYI, I recently made a simple dependency build script for IfcOpenShell for MSYS2 + MinGW which also includes very straightforward steps for building OCE too https://github.com/Tridify/IfcOpenShell/blob/msys/win/build-deps.sh

Fredounet, 2016-07-11:
Nice.
What about the CMAKE_NEED_RESPONSE flag? You didn't need that?

Stinkfist0, 2016-07-11:
Nope, so far everything has worked OK without it.

nickoe, 2016-07-15:
Hi, I have been working on getting this to MSYS2 too. I have a PKGBUILD that is almost complete, I just need disable OCE_AUTOINSTALL_DEPENDENT_LIBS and then I think it is good to go. I also see that createprocess error sometimes. I don't understand the issue, it happens sometimes.

thatcadguy, 2016-07-26:
@Fredounet Hi, if you fork the current master and add this guide as a readme (name it BUILD.MSYS2.md) or something like that, submit a PR and I'll merge.

Fredounet, 2016-09-10:
@thatcadguy Done. Sorry for the delay.

thatcadguy, 2016-11-05:
@Fredounet merged your PR

closed 2016-11-05 by thatcadguy

mahaidong, 2018-03-22:
is it possible to set DRAW=Yes to compile DRAWEXE?

may-tp, 2021-01-12:
`pacman -Sy ming-w64-x86_64-toolchain` should be `pacman -Sy mingw-w64-x86_64-toolchain`