tpaviot/oce #635: Installation in Debug build on VS is broken if CMAKE_PDB_OUTPUT_DIRECTORY is specified

GyrosGeier, 2016-07-31:
As I archive the PDB files on a separate symbol server, I install them into a different path than the regular installation prefix. When this is specified, the install target breaks in Debug builds:

         -- Installing: c:/OCE-0.18-dev/Win32/lib/TKerneld.lib
         -- Installing: c:/OCE-0.18-dev/Win32/bin/TKerneld.dll
         CMake Error at adm/cmake/TKernel/cmake_install.cmake:290 (file):
           file INSTALL cannot find
           "C:/Jenkins/workspace/windows-oce-msvc-head/878e2a29/bin//Debug/TKerneld.pdb".

thatcadguy, 2016-07-31:
Hmm, try removing the trailing slash in ```set(LIBRARY_OUTPUT_PATH ${CMAKE_BINARY_DIR}/bin/)``` in line 773 of CMakeLists.txt

GyrosGeier, 2016-07-31:
The problem is that if I set CMAKE_PDB_OUTPUT_DIRECTORY, the PDB files are created in that directory instead of the normal project output directory, so the install step cannot find them.

I've worked around the problem by asking OCE to not install the PDBs. It seems to me that the OCE build scripts duplicate part of CMake's base functionality here.