tpaviot/oce #636: Warnings for type conversions in 64 bit build

GyrosGeier, 2016-07-31:
I 've built OCE for both 32 and 64 bit using VS 2015, and the 64 bit build has a bunch of pretty ominous warnings:

    C:\Jenkins\oce\1078749c\src\OSD\OSD_File.cxx(1127): warning C4312: 'type cast': conversion from 'DWORD' to 'Standard_CString' of greater size [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_File.cxx(1184): warning C4312: 'type cast': conversion from 'DWORD' to 'Standard_CString' of greater size [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_File.cxx(2002): warning C4311: 'type cast': pointer truncation from 'Standard_PCharacter' to 'DWORD' [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_File.cxx(2002): warning C4302: 'type cast': truncation from 'Standard_PCharacter' to 'DWORD' [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_File.cxx(2015): warning C4311: 'type cast': pointer truncation from 'Standard_PCharacter' to 'DWORD' [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_File.cxx(2015): warning C4302: 'type cast': truncation from 'Standard_PCharacter' to 'DWORD' [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_File.cxx(2036): warning C4311: 'type cast': pointer truncation from 'Standard_PCharacter' to 'DWORD' [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_File.cxx(2036): warning C4302: 'type cast': truncation from 'Standard_PCharacter' to 'DWORD' [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_FileNode.cxx(851): warning C4244: 'return': conversion from 'intptr_t' to 'Standard_Integer', possible loss of data [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_FileNode.cxx(885): warning C4244: 'return': conversion from 'intptr_t' to 'Standard_Integer', possible loss of data [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_Host.cxx(218): warning C4996: 'GetVersionExA': was declared deprecated [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_Process.cxx(307): warning C4244: 'return': conversion from 'intptr_t' to 'Standard_Integer', possible loss of data [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_Thread.cxx(143): warning C4244: 'return': conversion from 'intptr_t' to 'DWORD', possible loss of data [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_Thread.cxx(249): warning C4312: 'type cast': conversion from 'DWORD' to 'Standard_Address' of greater size [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]
    C:\Jenkins\oce\1078749c\src\OSD\OSD_Thread.cxx(279): warning C4312: 'type cast': conversion from 'DWORD' to 'Standard_Address' of greater size [C:\Jenkins\oce\1078749c\adm\cmake\TKernel\TKernel.vcxproj]

    C:\Jenkins\oce\1078749c\src\LDOM\LDOM_OSStream.cxx(106): warning C4244: 'return': conversion from 'std::streamsize' to 'int', possible loss of data [C:\Jenkins\oce\1078749c\adm\cmake\TKCDF\TKCDF.vcxproj]
    C:\Jenkins\oce\1078749c\src\LDOM\LDOM_OSStream.cxx(128): warning C4244: 'initializing': conversion from 'std::streamsize' to 'int', possible loss of data [C:\Jenkins\oce\1078749c\adm\cmake\TKCDF\TKCDF.vcxproj]
    C:\Jenkins\oce\1078749c\src\LDOM\LDOM_OSStream.cxx(156): warning C4244: '+=': conversion from 'std::streamsize' to 'Standard_Integer', possible loss of data [C:\Jenkins\oce\1078749c\adm\cmake\TKCDF\TKCDF.vcxproj]

    C:\Jenkins\oce\1078749c\src\Graphic3d\Graphic3d_ArrayOfPrimitives.lxx(286): warning C4312: 'reinterpret_cast': conversion from 'Standard_Integer' to 'const Graphic3d_Vec4ub *' of greater size [C:\Jenkins\oce\1078749c\adm\cmake\TKV3d\TKV3d.vcxproj]

[Full build log](http://ci.kicad-pcb.org/job/windows-oce-msvc-head/build=debug,cpu=amd64,label=windows/13/consoleFull) and [warning summary](http://ci.kicad-pcb.org/job/windows-oce-msvc-head/build=debug,cpu=amd64,label=windows/13/warnings39Result/) are available as well.

thatcadguy, 2016-07-31:
Hi again Simon,

This is a known outstanding issue, we get all the same on the *nix platforms. My philosophy on most issues dealing with the code directly now is to go straight to the OCCT's bug tracker. It makes no sense for us to fix it one way, and then OCCT fix it another way, and then face headaches when we go to merge upstream. There have been disagreements on fixes in the past but everything is really good now, we can basically just assist them in finding more things to fix. In my mind (nowadays), our scope is to basically deal with our own CMake build scripts, and modeling/functionality issues that OCCT won't address.

thatcadguy, 2016-11-05:
This is fixed in OCCT 7.0.0: https://tracker.dev.opencascade.org/view.php?id=26585, we just need to catch up

closed 2016-11-05 by thatcadguy