tpaviot/oce #641: Only 3/4 of spline surface from imported STEP model is meshed

billtsalamander, 2016-09-07:
I loaded and meshed the attached step file, 
[badmesh.zip](https://github.com/tpaviot/oce/files/460361/badmesh.zip)
save as zip, and triangles were only generated for 3/4 of the spline surface at the base of the pocket.  The face is fine when the STEP file is loaded in Solidworks and OnShape. Furthermore, it meshes perfectly when re-exported to STEP from OnShape, but that's beside the point. I used the following code
`
    STEPControl_Reader reader;
    IFSelect_ReturnStatus stat = reader.ReadFile(filename);
    if (stat == IFSelect_RetDone) {
        Standard_Integer NbRoots = reader.NbRootsForTransfer();
        Standard_Integer NbTrans = reader.TransferRoots();
        Handle(TColStd_HSequenceOfTransient) list = reader.GiveList();
        Standard_Integer NBTrans = reader.TransferList(list); 
        
        //Load the solid from the file
        TopoDS_Shape ocesolid = reader.OneShape();
        //Write directly to STL
        StlAPI_Writer writer;
        writer.ASCIIMode() = false;
        writer.Write(ocesolid, "exported.stl");
     }
`

Alternative versions of code, that give identical results, involve calling `BRepMesh_IncrementalMesh(ocesolid, .01);`  and `BRep_Tool::Triangulation` to extract the triangles one by one from each face. The results are the same.
[image]

billtsalamander, 2016-09-08:
Quadrant is also missing when the file is opened in FreeCAD, so I don't think it's my error.  It looks like the spline face shares an edge with itself (this file was created by Siemans NX), and I'm curious if that is part of the issue.

thatcadguy, 2016-09-08:
Hi, I really don't have much time anymore to help out here, you should go
to the OCCT forum or the OCCT bugtracker for this type of problem as it is
related to algorithms made by them. They will have better advice than us
IMO.

billtsalamander, 2016-09-09:
OK, I will do that. Thanks for getting back to me.