tpaviot/oce #643: enumerator name conflict on macOS Sierra

davydden, 2016-09-21:
`oce` fails to compile on the most recent macOS due to name conflict:
```
In file included from /private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/OSD_Chronometer.cxx:55:
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/gettime_osx.h:81:5: error: redefinition of enumerator '_CLOCK_REALTIME'
    CLOCK_REALTIME,
    ^
/usr/include/time.h:154:24: note: expanded from macro 'CLOCK_REALTIME'
#define CLOCK_REALTIME _CLOCK_REALTIME
                       ^
/usr/include/time.h:153:1: note: previous definition is here
_CLOCK_REALTIME __CLOCK_AVAILABILITY = 0,
^
In file included from /private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/OSD_Chronometer.cxx:55:
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/gettime_osx.h:82:5: error: redefinition of enumerator '_CLOCK_MONOTONIC'
    CLOCK_MONOTONIC,
    ^
/usr/include/time.h:156:25: note: expanded from macro 'CLOCK_MONOTONIC'
#define CLOCK_MONOTONIC _CLOCK_MONOTONIC
                        ^
/usr/include/time.h:155:1: note: previous definition is here
_CLOCK_MONOTONIC __CLOCK_AVAILABILITY = 6,
^
In file included from /private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/OSD_Chronometer.cxx:55:
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/gettime_osx.h:83:5: error: redefinition of enumerator '_CLOCK_PROCESS_CPUTIME_ID'
    CLOCK_PROCESS_CPUTIME_ID,
    ^
/usr/include/time.h:168:34: note: expanded from macro 'CLOCK_PROCESS_CPUTIME_ID'
#define CLOCK_PROCESS_CPUTIME_ID _CLOCK_PROCESS_CPUTIME_ID
                                 ^
/usr/include/time.h:167:1: note: previous definition is here
_CLOCK_PROCESS_CPUTIME_ID __CLOCK_AVAILABILITY = 12,
^
In file included from /private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/OSD_Chronometer.cxx:55:
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/gettime_osx.h:84:5: error: redefinition of enumerator '_CLOCK_THREAD_CPUTIME_ID'
    CLOCK_THREAD_CPUTIME_ID
    ^
/usr/include/time.h:170:33: note: expanded from macro 'CLOCK_THREAD_CPUTIME_ID'
#define CLOCK_THREAD_CPUTIME_ID _CLOCK_THREAD_CPUTIME_ID
                                ^
/usr/include/time.h:169:1: note: previous definition is here
_CLOCK_THREAD_CPUTIME_ID __CLOCK_AVAILABILITY = 16
^
In file included from /private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/OSD_Chronometer.cxx:55:
/private/var/folders/5k/sqpp24tx3ylds4fgm13pfht00000gn/T/davydden/spack-stage/spack-stage-7eGSIl/oce-OCE-0.17.2/src/OSD/gettime_osx.h:85:3: error: typedef redefinition with different types ('enum clockid_t' vs 'enum clockid_t')
} clockid_t;
  ^
/usr/include/time.h:171:3: note: previous definition is here
} clockid_t;
  ^
```

schoeps, 2016-10-06:
Homebrew is using this work around in Homebrew/homebrew-science#4406:
```
diff -ruN a/adm/cmake/TKernel/CMakeLists.txt b/adm/cmake/TKernel/CMakeLists.txt
--- a/adm/cmake/TKernel/CMakeLists.txt	2016-06-02 14:18:16.000000000 +0200
+++ b/adm/cmake/TKernel/CMakeLists.txt	2016-10-05 19:25:14.000000000 +0200
@@ -29,7 +29,7 @@
 	set(TOOLKIT_LIBS ${TOOLKIT_LIBS} ${CSF_SOCKETLibs_LIB} ${CSF_advapi32_LIB} ${CSF_gdi32_LIB} ${CSF_user32_LIB} ${CSF_kernel32_LIB} ${CSF_psapi_LIB})
 else(WIN32)
 	#  An implementation for Mac OS X has been added in src/OSD/gettime_osx.h
-	if(NOT APPLE)
+	if(APPLE)
 		include( CheckFunctionExists )
 		check_function_exists( clock_gettime CLOCK_GETTIME_IN_LIBC )
 		if(NOT CLOCK_GETTIME_IN_LIBC)
@@ -40,7 +40,7 @@
 			endif(NOT CLOCK_GETTIME_IN_LIBRT)
 			set(TOOLKIT_LIBS ${TOOLKIT_LIBS} rt)
 		endif(NOT CLOCK_GETTIME_IN_LIBC)
-	endif(NOT APPLE)
+	endif(APPLE)
 endif(WIN32)
 
 # Adde-DHAVE_TBB in TKernel in order to benefit from Standard_MMgrTBBalloc
Binary files a/src/.DS_Store and b/src/.DS_Store differ
diff -ruN a/src/OSD/OSD_Chronometer.cxx b/src/OSD/OSD_Chronometer.cxx
--- a/src/OSD/OSD_Chronometer.cxx	2016-06-02 14:18:16.000000000 +0200
+++ b/src/OSD/OSD_Chronometer.cxx	2016-10-05 19:26:01.000000000 +0200
@@ -51,7 +51,7 @@
   #include <mach/mach.h>
 #endif
 
-#if defined(__APPLE__) && defined(__MACH__)
+#if defined(__OPPLE__) && defined(__MACH__)
 #include "gettime_osx.h"
 #endif
```

davydden, 2016-10-06:
@schoeps thanks. I will give it a try...

davydden, 2017-01-17:
@tpaviot this issue is still there in `0.18` but now requires a different patch to workaround. Can you please check if a proper fix is possible? Currently, everyone on `macOS` `Sierra` are out of luck... :-(

xzcvczx, 2017-01-17:
@davydden whats the problem with this patch in 0.18? i am currently building it but my machine is not the fastest

xzcvczx, 2017-01-17:
I have managed to get it build (without that other patch) with

```
diff --git a/src/OSD/OSD_Chronometer.cxx b/src/OSD/OSD_Chronometer.cxx
index f7374fb..63ac140 100644
--- a/src/OSD/OSD_Chronometer.cxx.old
+++ b/src/OSD/OSD_Chronometer.cxx
@@ -51,7 +51,7 @@
   #include <mach/mach.h>
 #endif
 
-#if defined(__APPLE__) && defined(__MACH__)
+#if defined(__APPLE__) && !defined(__MAC_10_12)
 #include "gettime_osx.h"
 #endif
 
```

tpaviot will likely come up with something better but this can work for now, (and is perfectly valid currently as OSX 10.12 is the only one with clock_gettime

davydden, 2017-01-17:
@xzcvczx you are right, the patch from @schoeps still applies fine, i just had a slightly modified version when extra white space was removed and this, AFAICT, made my patch broken.

davydden, 2017-01-17:
@xzcvczx will try your patch now (which is much cleaner) on `0.17.2` and `0.18`...

tpaviot, 2017-01-18:
@davydden @xzcvczx thank you guys for the quick fix. I'll push a pr with a fix and request your help for testing since I don't have access to any osx sierra machine