tpaviot/oce #653: msys2/mingw64 build fails

sanderboer, 2017-01-16:
Hi,

trying to build the git repo clone freshly on 16janu 2017, the build fails complaining about "external linkage required" : 
 
`[ 42%] Building CXX object adm/cmake/TKTopAlgo/CMakeFiles/TKTopAlgo.dir/__/__/__/src/BRepBuilderAPI/BRepBuilderAPI_FastSewing.cxx.obj
In file included from D:/OpenCascade/oce/inc/Handle_TopoDS_TShape.hxx:10:0,
                 from D:/OpenCascade/oce/inc/TopoDS_Shape.hxx:13,
                 from D:/OpenCascade/oce/drv/BRepBuilderAPI/BRepBuilderAPI_Copy.jxx:1,
                 from D:/OpenCascade/oce/drv/BRepBuilderAPI/BRepBuilderAPI_Copy.ixx:6,
                 from D:/OpenCascade/oce/src/BRepBuilderAPI/BRepBuilderAPI_Copy.cxx:17:
D:/OpenCascade/oce/src/Standard/Standard_DefineHandle.hxx:94:54: error: external linkage required for symbol 'const Handle_Standard_Type& {anonymous}::BRepBuilderAPI_Copy_Modification::DynamicType() const' because of 'dllexport' attribute
 Standard_EXPORT virtual const Handle(Standard_Type)& DynamicType() const;
                                                      ^
D:/OpenCascade/oce/src/BRepBuilderAPI/BRepBuilderAPI_Copy.cxx:139:3: note: in expansion of macro 'DEFINE_STANDARD_RTTI'
   DEFINE_STANDARD_RTTI(BRepBuilderAPI_Copy_Modification)
   ^~~~~~~~~~~~~~~~~~~~
D:/OpenCascade/oce/src/Standard/Standard_DefineHandle.hxx:76:37: error: external linkage required for symbol 'static {anonymous}::Handle_BRepBuilderAPI_Copy_Modification {anonymous}::Handle_BRepBuilderAPI_Copy_Modification::DownCast(const Handle_Standard_Transient&)' because of 'dllexport' attribute
   Standard_EXPORT static Handle(C1) DownCast(const Handle(BC)& AnObject); \
                                     ^
D:/OpenCascade/oce/src/Standard/Standard_DefineHandle.hxx:81:39: note: in expansion of macro 'DEFINE_STANDARD_HANDLECLASS'
 #define DEFINE_STANDARD_HANDLE(C1,C2) DEFINE_STANDARD_HANDLECLASS(C1,C2,Standard_Transient)
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/OpenCascade/oce/src/BRepBuilderAPI/BRepBuilderAPI_Copy.cxx:146:1: note: in expansion of macro 'DEFINE_STANDARD_HANDLE'
 DEFINE_STANDARD_HANDLE(BRepBuilderAPI_Copy_Modification, BRepTools_Modification)
 ^~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [adm/cmake/TKTopAlgo/CMakeFiles/TKTopAlgo.dir/build.make:4339: adm/cmake/TKTopAlgo/CMakeFiles/TKTopAlgo.dir/__/__/__/src/BRepBuilderAPI/BRepBuilderAPI_Copy.cxx.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1420: adm/cmake/TKTopAlgo/CMakeFiles/TKTopAlgo.dir/all] Error 2
make: *** [Makefile:161: all] Error 2`

Any suggestions ?

gr.
Sander

sanderboer, 2017-01-16:
I just succesfully compiled 0.17.2 from the packaged source zip with this msys2/mingw64 setup. 
Trying to compile 0.18 from the packaged zip results in the same error, however.

tpaviot, 2017-01-17:
@sanderboer thank you for your report

strange, mingw64 uses to work see https://ci.appveyor.com/project/tpaviot/oce

which gcc version is it ? can you please copy/paste the full trace

tpaviot, 2017-01-17:
@sanderboer did u have a look at https://github.com/tpaviot/oce/issues/631 ?

mat127, 2017-05-22:
Hi, I have encoutered nearly the same error:

[ 23%] Building CXX object adm/cmake/TKTopAlgo/CMakeFiles/TKTopAlgo.dir/__/__/__/src/BRepBuilderAPI/BRepBuilderAPI_Copy.cxx.obj
In file included from C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/inc/Handle_TopoDS_TShape.hxx:10:0,
                 from C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/inc/TopoDS_Shape.hxx:13,
                 from C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/drv/BRepBuilderAPI/BRepBuilderAPI_Copy.jxx:1,
                 from C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/drv/BRepBuilderAPI/BRepBuilderAPI_Copy.ixx:6,
                 from C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/src/BRepBuilderAPI/BRepBuilderAPI_Copy.cxx:17:
C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/src/Standard/Standard_DefineHandle.hxx:94:54: error: external linkage required for symbol 'const Handle_Standard_Type& {anonymous}::BRepBuilderAPI_Copy_Modification::DynamicType() const' because of 'dllexport' attribute
 Standard_EXPORT virtual const Handle(Standard_Type)& DynamicType() const;
                                                      ^
C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/src/BRepBuilderAPI/BRepBuilderAPI_Copy.cxx:139:3: note: in expansion of macro 'DEFINE_STANDARD_RTTI'
   DEFINE_STANDARD_RTTI(BRepBuilderAPI_Copy_Modification)
   ^~~~~~~~~~~~~~~~~~~~
C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/src/Standard/Standard_DefineHandle.hxx:76:37: error: external linkage required for symbol 'static {anonymous}::Handle_BRepBuilderAPI_Copy_Modification {anonymous}::Handle_BRepBuilderAPI_Copy_Modification::DownCast(const Handle_Standard_Transient&)' because of 'dllexport' attribute
   Standard_EXPORT static Handle(C1) DownCast(const Handle(BC)& AnObject); \
                                     ^
C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/src/Standard/Standard_DefineHandle.hxx:81:39: note: in expansion of macro 'DEFINE_STANDARD_HANDLECLASS'
 #define DEFINE_STANDARD_HANDLE(C1,C2) DEFINE_STANDARD_HANDLECLASS(C1,C2,Standard_Transient)
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/Petr.Matousek/Documents/NetBeansProjects/oce/src/BRepBuilderAPI/BRepBuilderAPI_Copy.cxx:146:1: note: in expansion of macro 'DEFINE_STANDARD_HANDLE'
 DEFINE_STANDARD_HANDLE(BRepBuilderAPI_Copy_Modification, BRepTools_Modification)
 ^~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [adm/cmake/TKTopAlgo/CMakeFiles/TKTopAlgo.dir/build.make:4339: adm/cmake/TKTopAlgo/CMakeFiles/TKTopAlgo.dir/__/__/__/src/BRepBuilderAPI/BRepBuilderAPI_Copy.cxx.obj] Chyba 1
make[1]: *** [CMakeFiles/Makefile2:1420: adm/cmake/TKTopAlgo/CMakeFiles/TKTopAlgo.dir/all] Chyba 2
make: *** [Makefile:163: all] Chyba 2

I have a fresh MSYS2 installation, the version of gcc is:

mat127, 2017-05-23:
According to my exploration this could be related to the version of the gcc as I was not able to compile under mingw-w64 with gcc 6.x too. With gcc 5.x the compilation was successful.

Unfortunately I was not able to downgrade the gcc under the MSys2 as nearly everything (including the core components like e.g. pacman) are dependent on the gcc-libs that has to be downgraded too.

On the other hand the compilation with the gcc 6.3.1 was successful on my Linux machine.

ArvinShaffer, 2021-06-29:
> @sanderboer thank you for your report
> 
> strange, mingw64 uses to work see https://ci.appveyor.com/project/tpaviot/oce
> 
> which gcc version is it ? can you please copy/paste the full trace

I tried the above method to successfully compile Oce, but these libraries are missing  -lTKPCAF  -lTKXCAFSchema  -lTKShapeSchema -lTKPLCAF -lTKNIS -lTKStdLSchema -lTKVoxel -lTKPShape -lTKStdSchema
```
PS D:\software\oce-OCE-0.18\build> gcc --version
gcc.exe (x86_64-posix-seh-rev0, Built by MinGW-W64 project) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

ArvinShaffer, 2021-07-11:
I successfully used gcc5.3 to compile oce0.18，solve the error Standard_DefineHandle.hxx:94:54: error: external linkage required for symbol 'const Handle_Standard_Type& {anonymous}::BRepBuilderAPI_Copy_Modification::DynamicType() const' because of 'dllexport' attribute