tpaviot/oce #659: cmake error installing OpenCASCADE

bjdarrer, 2017-03-09:
I came across an error trying to install OpenCASCADE (oce) using `cmake`, where I was following the build command:

```
mkdir build
cd build
cmake -DOCE_INSTALL_PREFIX=/path/to/where/you/want/oce ..
make install
```

I downloaded OpenCASCADE with: `git clone git://github.com/tpaviot/oce.git` from [https://www.dealii.org/8.3.0/external-libs/opencascade.html]

The command line build error is:

```
-- Build 64bit
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find OpenGL (missing: OPENGL_gl_LIBRARY OPENGL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindOpenGL.cmake:172 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:440 (find_package)
```

I checked to see if I had a version of OpenGL, with,

`glxinfo | grep "OpenGL version"`

that showed:

`OpenGL version string: 3.0 Mesa 12.0.6`

So it seems I have `OpenGL` but the build cannot find it. Can anyone understand how to resolve this?

tpaviot, 2017-03-09:
@bjdarrer OpenGl libs are not enough, you also need headers. Be sure you installed the opengl dev package (according to your distro the package name may change)

bjdarrer, 2017-03-09:
Thank you tpaviot for your help. I went to:

http://askubuntu.com/questions/11378/how-do-i-set-up-an-opengl-programming-environment

and followed the instructions second reply (with score <10>), that seems to have resolved the `Could NOT find OpenGL (missing: OPENGL_gl_LIBRARY OPENGL_INCLUDE_DIR)` issue. However, now it cannot find `Freetype`
```
-- Build 64bit
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Freetype (missing: FREETYPE_LIBRARY FREETYPE_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindFreetype.cmake:151 (find_package_handle_standard_args)
  CMakeLists.txt:446 (find_package)
```
I am wondering if it needs a 32 bit Freetype instead of 64 bit one, see:

http://askubuntu.com/questions/846961/wine-cannot-find-the-freetype-font-library/847041

I checked my Ubuntu system and it does have Freetype on it.

bjdarrer, 2017-03-09:
@tpaviot , I installed freetype-dev --> freetype2.git --> see:
https://www.freetype.org/download
http://git.savannah.gnu.org/cgit/freetype/freetype2.git/pe2.git/

and built it with: `make setup ansi` & `make`, but on next building `OpenCASCADE`, I get:

```
$ cmake -DOCE_INSTALL_PREFIX=/home/brendan/bin/oce ..
-- Build 64bit
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Freetype (missing: FREETYPE_LIBRARY FREETYPE_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindFreetype.cmake:151 (find_package_handle_standard_args)
  CMakeLists.txt:446 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/brendan/Documents/OpenCASCADE/oce/build/CMakeFiles/CMakeOutput.log".
See also "/home/brendan/Documents/OpenCASCADE/oce/build/CMakeFiles/CMakeError.log".

```
It doesn't seem to have detected it. Would I need to include freetype-dev in the cmake build? and how would I do that as I don't know where freetype2 is called from. I also tried:

`cmake -DFREETYPE_INCLUDE_DIR=/home/brendan/freetype2 -DOCE_INSTALL_PREFIX=/home/brendan/bin/oce ..`

but with the same building error message. Thank you for your help.

bjdarrer, 2017-03-10:
Problem solved by using `sudo apt-get install libfreetype6-dev` , from last comment at:

https://github.com/matplotlib/matplotlib/issues/3029/

Now OpenCASCADE (oce) is building correctly!

closed 2017-04-15 by tpaviot

iulian787, 2018-01-30:
sudo apt-get install libfreetype6-dev was a life saver