tpaviot/oce #660: OCE_LIBRARIES contains DRAWEXE

janusw, 2017-03-09:
Hi,

I have successfully built OCE 0.18 with -DOCE_DRAW=ON. Now, if I want to link my own application to OCE via

    TARGET_LINK_LIBRARIES(myApp ${OCE_LIBRARIES})

I get:

    /usr/bin/ld: cannot find -lDRAWEXE

It seems that OCE_LIBRARIES wrongly includes DRAWEXE (which obviously is not a library).

Hope it's not a big deal to fix this ...

Cheers,
Janus

janusw, 2017-03-09:
The obvious workaround is of course to manually remove DRAWEXE from OCEConfig.cmake, but certainly it should not get in there in the first place.

For completeness, this is how the relevant part of the generated OCEConfig.cmake looks in my case:

    set(OCE_LIBRARIES "TKernel;TKMath;TKG2d;TKG3d;TKGeomBase;TKBRep;TKGeomAlgo;TKTopAlgo;TKPrim;TKBO;TKHLR;TKMesh;TKShHealing;TKXMesh;TKBool;TKFillet;TKFeat;TKOffset;TKService;TKV3d;TKOpenGl;TKMeshVS;TKNIS;TKVoxel;TKCDF;PTKernel;TKLCAF;FWOSPlugin;TKPShape;TKBinL;TKXmlL;TKPLCAF;TKTObj;TKShapeSchema;TKStdLSchema;TKCAF;TKBin;TKXml;TKPCAF;TKBinTObj;TKXmlTObj;TKStdSchema;TKSTL;TKXSBase;TKSTEPBase;TKIGES;TKSTEPAttr;TKSTEP209;TKSTEP;TKVRML;TKXCAF;TKXCAFSchema;TKXmlXCAF;TKBinXCAF;TKXDEIGES;TKXDESTEP;TKDraw;TKTopTest;TKViewerTest;TKXSDRAW;TKDCAF;TKXDEDRAW;TKTObjDRAW;DRAWEXE")

janusw, 2017-03-10:
Btw, the same problem seems to occur with earlier versions as well (e.g. I see it also with 0.14).

janusw, 2017-03-10:
Also the package "oce-draw 0.17.2-1" as shipped in Ubuntu 16.10 is affected.

gladk, 2017-03-10:
How important is this problem? It looks like the current oce in Debian is also affected. It is theoretically possible to fix it there before the next large stable release. 

Thanks

janusw, 2017-03-12:
Well, as mentioned it's easy to work around it, but it's still annoying. It should definitely be fixed for the next release.

Here is a suggestion for a simple fix:

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index dd56a4d..eba6380 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -990,7 +990,8 @@ macro(process_module modulename modules_std)
                list(APPEND modules ${ARGN})
        endif(NOT OCE_DISABLE_X11)
        foreach(module ${modules})
-               list(APPEND OCE_LIBRARIES ${module})
+               if(NOT module MATCHES ".*EXE")
+                       list(APPEND OCE_LIBRARIES ${module})
                set(TOOLKIT_MODULES "")
                set(TOOLKIT_DEPENDS "")
                set(TOOLKIT_INCLUDE_DIRECTORIES "")
```

janusw, 2017-04-14:
The branch is merged (thanks for that!). Closing the issue ...

closed 2017-04-14 by janusw