tpaviot/oce #667: DRAWEXE only opens a GUI but shows no graphic

calocedrus, 2017-06-08:
Switching from pythonocc to OCE is a bit tough without the convenience of pythonocc's QT based graphical interface. I thought that "Draw" which came with OCE could help but while a GUI opens when I launch `DRAWEXE`, I can't seem to be able to display any graphic.

I've installed OCE with the use of the [unix guide](https://github.com/tpaviot/oce/blob/master/BUILD.Unix.md).

Lauching `CASROOT=/usr/local/share/oce-0.18 DRAWEXE` results in the opening of this frame:
[image]
I can then `pload ALL`:
```
~/Programs/OCE/oce_0p18p1/build$ CASROOT=/usr/local/share/oce-0.18 DRAWEXE
Draw[1]> Draw[2]> pload ALL
1
Draw[3]>
```
But then any choice in any of the dropdown menu from the above GUI frame has no action or result in the following message in my terminal window:
```
bgerror failed to handle background error.
    Original error: no files matched glob pattern "*"
    Error in bgerror:
```
Is there anything in particular to setup so that plots can appear?
Some references I've used to try to solve the issue myself:
[https://github.com/tpaviot/oce/issues/626](https://github.com/tpaviot/oce/issues/626)
[https://dev.opencascade.org/doc/overview/html/occt_user_guides__test_harness.html](https://dev.opencascade.org/doc/overview/html/occt_user_guides__test_harness.html)

aothms, 2017-06-08:
Did you try `mu` in the TCL console? You can use `restore <path>` to load a OCC BRep file and `fit` to fit it in view.

calocedrus, 2017-06-09:
Thanks for your reply. I can't explain what happened so that I couldn't see anything. 
Now if I chose `Views` -> `mu4`, 4 different windows open, and I can then select a shape and plot it. 

For any user wanting to do a "harness test", here's what to do:
Launch the executable `DRAWEXE`
In the GUI that opened, menu `Load` > `pload ALL`
In menu `Views` , select any option (`mu4` shows 4 different views of the XYZ axis).
Then select an object to plot using `Curves` or `Surfaces`.
Below is the example of a torus.
[image]

closed 2017-06-09 by calocedrus