tpaviot/oce #668: OCE 0.15: two test failures with gcc 5 (BRepAlgoAPITestSuite)

janusw, 2017-06-08:
I'm unfortunately bound to using OCE 0.15 (OCCT 6.7.0) at the moment. It works great with gcc versions up to 4.9, but when using gcc 5 (to be precise: 5.4.1) I'm seeing these two test failures:

```
      Start 36: BRepAlgoAPITestSuite.testCutBox
36/48 Test #36: BRepAlgoAPITestSuite.testCutBox .................***Exception: SegFault  0.12 sec
      Start 37: BRepAlgoAPITestSuite.testCutCylSphere
37/48 Test #37: BRepAlgoAPITestSuite.testCutCylSphere ...........***Exception: SegFault  0.13 sec
```

Both of them produce a segfault in `BOPAlgo_PaveFiller::MakeSplitEdges`.

Any insights on what causes this problem are greatly appreciated ...

Cheers,
Janus

janusw, 2017-06-09:
In order to debug this, I tried using -DCMAKE_BUILD_TYPE=Debug, but in fact that makes the segfaults disappear. So this might well be a bug in the compiler optimizations of g++.

Also: OCE versions 0.14 and 0.16 appear to work fine with gcc 5.

janusw, 2017-06-09:
The segfaults occur not only with gcc 5.4.1, but also with 6.3.0 and 7.1.0.

janusw, 2017-06-09:
Using -DCMAKE_BUILD_TYPE=RelWithDebInfo, I got the following backtrace:

```
Program received signal SIGSEGV, Segmentation fault.
BOPAlgo_PaveFiller::MakeSplitEdges (this=this@entry=0x79d3f0) at /home/janus/OCE/oce/src/BOPAlgo/BOPAlgo_PaveFiller_7.cxx:112
112	      bCB=!aCB.IsNull();
(gdb) bt
#0  BOPAlgo_PaveFiller::MakeSplitEdges (this=this@entry=0x79d3f0) at /home/janus/OCE/oce/src/BOPAlgo/BOPAlgo_PaveFiller_7.cxx:112
#1  0x00007ffff78e77d5 in BOPAlgo_PaveFiller::Perform (this=0x79d3f0) at /home/janus/OCE/oce/src/BOPAlgo/BOPAlgo_PaveFiller.cxx:191
#2  0x00007ffff785f1e1 in BRepAlgoAPI_BooleanOperation::Build (this=this@entry=0x7fffffffd370) at /home/janus/OCE/oce/src/BRepAlgoAPI/BRepAlgoAPI_BooleanOperation.cxx:254
#3  0x00007ffff785f359 in BRepAlgoAPI_Cut::BRepAlgoAPI_Cut (this=0x7fffffffd370, S1=..., S2=...) at /home/janus/OCE/oce/src/BRepAlgoAPI/BRepAlgoAPI_Cut.cxx:31
#4  0x0000000000415edb in BRepAlgoAPITestSuite_testCutBox_Test::TestBody (this=<optimized out>) at /home/janus/OCE/oce/test/BRepAlgoAPI_test/boolean_ops.cpp:17
#5  0x000000000043c403 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (location=0x43ede8 "the test body", method=<optimized out>, 
    object=<optimized out>) at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2078
#6  testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=object@entry=0x7959b0, 
    method=(void (testing::Test::*)(testing::Test * const)) 0x415a80 <BRepAlgoAPITestSuite_testCutBox_Test::TestBody()>, location=location@entry=0x43ede8 "the test body")
    at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2114
#7  0x0000000000433b5c in testing::Test::Run (this=this@entry=0x7959b0) at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2151
#8  0x0000000000433d35 in testing::Test::Run (this=0x7959b0) at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2142
#9  testing::TestInfo::Run (this=0x7954d0) at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2326
#10 0x0000000000433e9d in testing::TestInfo::Run (this=<optimized out>) at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2301
#11 testing::TestCase::Run (this=0x7959f0) at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2444
#12 0x000000000043423b in testing::TestCase::Run (this=<optimized out>) at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2430
#13 testing::internal::UnitTestImpl::RunAllTests (this=0x7955e0) at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:4315
#14 0x000000000043c8e3 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (
    location=0x43e1d8 "auxiliary test code (environments or event listeners)", method=<optimized out>, object=<optimized out>)
    at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2078
#15 testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (object=0x7955e0, 
    method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x433f90 <testing::internal::UnitTestImpl::RunAllTests()>, 
    location=location@entry=0x43e1d8 "auxiliary test code (environments or event listeners)") at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:2114
#16 0x0000000000434635 in testing::UnitTest::Run (this=0x64cfc0 <testing::UnitTest::GetInstance()::instance>) at /home/janus/OCE/oce/test/gtest-1.7.0/src/gtest.cc:3926
#17 0x000000000041561f in RUN_ALL_TESTS () at /home/janus/OCE/oce/test/gtest-1.7.0/include/gtest/gtest.h:2288
#18 main (argc=1, argv=<optimized out>) at /home/janus/OCE/oce/test/BRepAlgoAPI_test/boolean_ops.cpp:38
```

janusw, 2017-06-09:
Apparently this is (again!) related to the issue of null-refs, cf. 005f79f18b4c98969442b89fb6887d5f6754ea88 etc.

Seems like `-fno-delete-null-pointer-checks` is not enough to resolve this issue for OCE 0.15 (though this seems to be the case for other releases).

One way to fix it is to turn the reference into an actual pointer and check for it being non-null:

```diff
diff --git a/src/BOPAlgo/BOPAlgo_PaveFiller_7.cxx b/src/BOPAlgo/BOPAlgo_PaveFiller_7.cxx
index 3f978c4..1c7a1ae 100644
--- a/src/BOPAlgo/BOPAlgo_PaveFiller_7.cxx
+++ b/src/BOPAlgo/BOPAlgo_PaveFiller_7.cxx
@@ -108,11 +108,11 @@ static void UpdateVertices(const TopoDS_Edge& aE,
     aItPB.Initialize(aLPB);
     for (; aItPB.More(); aItPB.Next()) {
       aPB=aItPB.Value();
-      const Handle(BOPDS_CommonBlock)& aCB=myDS->CommonBlock(aPB);
-      bCB=!aCB.IsNull();
+      const Handle(BOPDS_CommonBlock) *aCB=&myDS->CommonBlock(aPB);
+      bCB=(aCB && !aCB->IsNull());
       if (bCB) {
-        myDS->SortPaveBlocks(aCB);
-        aPB=aCB->PaveBlock1();
+        myDS->SortPaveBlocks(*aCB);
+        aPB=(*aCB)->PaveBlock1();
       }
       //
       if (aMPB.Add(aPB)) {
@@ -123,7 +123,7 @@ static void UpdateVertices(const TopoDS_Edge& aE,
         nSp = SplitEdge(nE, nV1, aT1, nV2, aT2);
         //
         if (bCB) {
-          aCB->SetEdge(nSp);
+          (*aCB)->SetEdge(nSp);
         }
         else {
           aPB->SetEdge(nSp);

```

tpaviot, 2017-08-08:
@janusw I can't reproduce the issue with current master. Everything is fine with gcc-5.4.1 (see https://travis-ci.org/tpaviot/oce/jobs/262296313). oce-0.15 is quite old, you should update to the latest release (currently 0.18.1)

janusw, 2017-08-09:
> @janusw I can't reproduce the issue with current master. Everything is fine with gcc-5.4.1 (see https://travis-ci.org/tpaviot/oce/jobs/262296313).

@tpaviot: Yes, I know that this problem is specific to 0.15. It's already gone in 0.16 (and not present in 0.14 either).

In any case, I have fixed it on my branch https://github.com/janusw/oce/tree/OCE-0.15-branch, so it's not really an issue for me any more.

> oce-0.15 is quite old, you should update to the latest release (currently 0.18.1)

I would like to do that, but unfortunately I'm bound to 0.15 by now (basically due to a dependency on https://github.com/janusw/geom, which does not work properly with more recent versions). Or are you aware of a stand-alone GEOM version which works with OCC 6.9 / OCE 0.18 ?

Cheers,
Janus

tpaviot, 2017-08-10:
@janusw you should port geom to latest oce by your own. I did it for smesh (https://github.com/tpaviot/smesh), it's not that hard.

closed 2017-08-10 by tpaviot