tpaviot/oce #675: OCE build fails in Fedora Rawhide

hobbes1069, 2017-08-12:
It looks like OCE depends on xlocale.h from glibc-headers but it was removed in the latest release (2.26) which is in Fedora Rawhide because it was supposed to be an internal header only...

tpaviot, 2017-08-13:
@hobbes1069 can you please copy/paste the full trace

tpaviot, 2017-08-13:
@hobbes Standard_CLocaleSEntry.hxx relies on xlocale.hxx, but this should be included if and only if xlocale.h has been found (see https://github.com/tpaviot/oce/blob/master/src/Standard/Standard_CLocaleSentry.hxx#L38)

````
#ifdef HAVE_XLOCALE_H
  #include <xlocale.h>
#endif
````

hobbes1069, 2017-08-13:
Well, kinda, it doesn't test for it's presence, it just assumes it's available on OSX and Linux and sets HAVE_XLOCALE_H...

If my patch works I'll submit a pull request but all I did was comment out the conditional for Linux.

tpaviot, 2017-08-13:
you're right. Maybe adding a line to CMakeList would do the job, something like

```
check_include_file(xlocale.h HAVE_XLOCALE_H)
```

hobbes1069, 2017-08-13:
That's a better solution that my brute force workaround.

yopito, 2017-08-17:
Hello, faced the same problem while packaging oce for VoidLinux distribution.  
The patch I've used:
```
--- src/Standard/Standard_CLocaleSentry.hxx.ORIG	2017-08-11 07:51:11.000000000 +0200
+++ src/Standard/Standard_CLocaleSentry.hxx	2017-08-18 01:07:11.639206272 +0200
@@ -30,7 +30,9 @@
   #endif
 
   //! We check _GNU_SOURCE for glibc extensions here and it is always defined by g++ compiler.
-  #if defined(_GNU_SOURCE) && !defined(__ANDROID__)
+  //! Musl libc does not provide xlocale.h
+  //! glibc 2.26+ does not provide xlocale.h (anymore)
+  #if defined(_GNU_SOURCE) && defined(__GLIBC__) && __GLIBC__ <= 2 && __GLIBC_MINOR__ < 26 && !defined(__ANDROID__)
     #define HAVE_XLOCALE_H
   #endif
 #endif // ifndef HAVE_LOCALE_H
```

janusw, 2017-12-18:
> Maybe adding a line to CMakeList would do the job

@tpaviot This is done in aa1321e68cc004e3debe38d79ae74581a617c767, see pull request #684. Works well for me on Ubuntu 17.10. Ok for master?

janusw, 2018-07-20:
The problem should be fixed by aa1321e. @hobbes1069: Can you confirm that it works for you now? And if yes, can we close this issue?

hobbes1069, 2018-07-20:
It's been so long I can't be 100% sure without digging further, but as i have removed the patch for the brute force method I was using I would assume so.

closed 2018-07-20 by hobbes1069