tpaviot/oce #687: Regression: BRepGProp_Face::Load segfaults

nickoe, 2018-01-02:
This is a segfault  I see in KiCad when going from OCE 0.17.2 to OCE 0.18.2. I have not tested OCE 0.18.1.

The issue appears always for me on two archlinux machines. Is there anything more simple, than running KiCad, that I can run from OCE to test loading this IGES file to create a simpler test scenario for debugging?

Originally reported in https://bugs.launchpad.net/kicad/+bug/1738872

Snippet of backtrace, full backtrace in the attached zip file in above bug report.
```
#0 0x00007f9a3481e65e in BRepGProp_Face::Load(TopoDS_Edge const&) () at /opt/oce/lib/libTKTopAlgo.so.11
#1 0x00007f9a348255c6 in BRepGProp_Gauss::Compute(BRepGProp_Face&, BRepGProp_Domain&, gp_Pnt const&, double&, gp_Pnt&, gp_Mat&) ()
    at /opt/oce/lib/libTKTopAlgo.so.11
#2 0x00007f9a34826e05 in BRepGProp_Sinert::Perform(BRepGProp_Face&, BRepGProp_Domain&) () at /opt/oce/lib/libTKTopAlgo.so.11
#3 0x00007f9a3481bdc5 in BRepGProp::SurfaceProperties(TopoDS_Shape const&, GProp_GProps&) () at /opt/oce/lib/libTKTopAlgo.so.11
#4 0x00007f9a34c0c34d in ShapeAnalysis_Wire::CheckSmallArea(TopoDS_Wire const&, bool) () at /opt/oce/lib/libTKShHealing.so.11
#5 0x00007f9a34c399cb in ShapeFix_Face::FixSmallAreaWire(bool) () at /opt/oce/lib/libTKShHealing.so.11
#6 0x00007f9a34c4840b in ShapeFix_Face::Perform() () at /opt/oce/lib/libTKShHealing.so.11
#7 0x00007f9a34c6da9e in ShapeFix_Shape::Perform(Handle_Message_ProgressIndicator const&) () at /opt/oce/lib/libTKShHealing.so.11
#8 0x00007f9a34c6de90 in ShapeFix_Shape::Perform(Handle_Message_ProgressIndicator const&) () at /opt/oce/lib/libTKShHealing.so.11
#9 0x00007f9a34d019c6 in () at /opt/oce/lib/libTKShHealing.so.11
#10 0x00007f9a34cfe8aa in ShapeProcess::Perform(Handle_ShapeProcess_Context const&, char const*) () at /opt/oce/lib/libTKShHealing.so.11
#11 0x00007f9a35101a86 in XSAlgo_AlgoContainer::ProcessShape(TopoDS_Shape const&, double, double, char const*, char const*, Handle_Standard_Transient&, Handle_Message_ProgressIndicator const&) const () at /opt/oce/lib/libTKXSBase.so.11
#12 0x00007f9a3869b221 in IGESToBRep_Actor::Transfer(Handle_Standard_Transient const&, Handle_Transfer_TransientProcess const&) ()
    at /opt/oce/lib/libTKIGES.so.11
#13 0x00007f9a3502873f in Transfer_ActorOfTransientProcess::Transferring(Handle_Standard_Transient const&, Handle_Transfer_ProcessForTransient const&) () at /opt/oce/lib/libTKXSBase.so.11
#14 0x00007f9a3503a5a2 in Transfer_ProcessForTransient::TransferProduct(Handle_Standard_Transient const&) () at /opt/oce/lib/libTKXSBase.so.11
#15 0x00007f9a3503bcff in Transfer_ProcessForTransient::Transferring(Handle_Standard_Transient const&) () at /opt/oce/lib/libTKXSBase.so.11
#16 0x00007f9a3503c498 in Transfer_ProcessForTransient::Transfer(Handle_Standard_Transient const&) () at /opt/oce/lib/libTKXSBase.so.11
#17 0x00007f9a350bebd6 in XSControl_TransferReader::TransferOne(Handle_Standard_Transient const&, bool) () at /opt/oce/lib/libTKXSBase.so.11
#18 0x00007f9a350b9650 in XSControl_Reader::TransferRoots() () at /opt/oce/lib/libTKXSBase.so.11
#19 0x00007f9a3e33c729 in IGESCAFControl_Reader::Transfer(Handle_TDocStd_Document&) () at /opt/oce/lib/libTKXDEIGES.so.11
#20 0x00007f9a3ebf46f0 in readIGES(Handle_TDocStd_Document&, char const*) (m_doc=..., fname=0x5611b1dbb5b0 "/home/nickoe/test_proj/User Library-Fuse_Holder.IGS") at /home/nickoe/kicad-source-mirror/plugins/3d/oce/loadmodel.cpp:415
#21 0x00007f9a3ebf49d8 in LoadModel(char const*) (filename=0x5611b1dbb5b0 "/home/nickoe/test_proj/User Library-Fuse_Holder.IGS")

```

nickoe, 2018-01-09:
Hi @tpaviot, 

I have been told that there is a way to disable Shape Healing with a config file for OCE. But they guy who told me didn't remember where this config file is and said it was not documented. Do you know this such that we can progress with this?

tpaviot, 2018-01-09:
TkShHealing is a core part of oce. You can set OCE_MODEL to False, it will also disable a lot of other packages.

The BRepGProp_Face issue seems to be related to Kicad (which I don't know). Please report issue to Kicad team.

closed 2018-01-09 by tpaviot

nickoe, 2018-01-09:
@tpaviot I originally reported this on KiCad as mentioned in the original issue description.

I may be a bit naive, but what makes you conclude that BRepGProp_Face is related to KiCad? Looking at frame zero it says:
```
RepGProp_Face::Load(TopoDS_Edge const&) () at /opt/oce/lib/libTKTopAlgo.so.11
```
That points to one of the OCE so-files.

orsonmmz, 2018-02-17:
Hi tpaviot,

Would you give any hints about potential causes of the segfault? Is there a way to gracefully handle an input data error rather than getting a segfault? What puzzles me the most is the fact oce 0.17 did not have any problems loading the file.

orsonmmz, 2018-02-17:
I extracted the code and turned it into a standalone project [(link)](https://github.com/orsonmmz/oce_iges_test) to demonstrate the segfault. I hope you could give us an advice on how to proceed.

sethhillbrand, 2018-02-22:
This issue is in FixSmallFaces().  Running OCE's 0.18 version of this function on the attached file will segfault OCE when RemoveSmallAreaFaceMode is enabled on IGES processing.

[User Library-Fuse_Holder.IGS.zip](https://github.com/tpaviot/oce/files/1748938/User.Library-Fuse_Holder.IGS.zip)

nickoe, 2018-03-11:
@tpaviot Is this something we can report upstream?

a-betenev, 2018-03-12:
There is no sense to report it upstream -- this issue has been fixed in OCCT 7.1.0 (OCCT issue 27781). 
To reproduce in DRAW:

pload ALL
param xstep.cascade.unit M
igesbrep "User Library-Fuse_Holder.IGS" a *

Note that this problem is reproducible with OCCT 6.9.1 and 7.0.0, bun not 6.9.0 not 7.1.0+. As well I cannot reproduce it on any version using file "test2.igs" found in the standalone project shared by orsonmmz.

nickoe, 2018-03-12:
@a-betenev  OK, thank you for the information. I guess we will have to wait for a new OCE release based on OCCT 7.1.0+.  :)

tpaviot, 2018-03-12:
@nickoe Branch upstream-occt-720 contains upstream code for occt720, patches are uploaded to review/occt-720-patches. I will merge into master as soon as travis and appveyor builds pass.

nickoe, 2018-03-12:
@tpaviot Thanks.