tpaviot/oce #689: Fix build error with glibc 2.26, xlocale.h

rsm-gh, 2018-01-23:
While building on archilinux I found an `xlocale.h` problem. The fast solution was to comment the `#include <xlocale.h>` line of the `/src/Standard/Standard_CLocaleSentry.cxx` file.

Here are more details taken from [this thread](https://github.com/c72578/cpprestsdk/commit/d4f707abb4575aacf43ec02d30fa10afed464c7d)

> Do not include xlocale.h on systems, where __GLIBC__ is defined
  xlocale.h has been removed from glibc 2.26
  The include of locale.h in asyncrt_utils.h is sufficient
  Further details:
  https://sourceware.org/git/?p=glibc.git;a=commit;h=f0be25b

Kind regards,
rsm

janusw, 2018-02-20:
aa1321e68cc004e3debe38d79ae74581a617c767 should actually fix this. Which version did you use?

rsm-gh, 2018-02-23:
I'm gonna close this because I'm starting to think that the day I did the compilation I did something wrong. Yesterday I made the compilation without any problem.

OCE version: 0.18.3 
glibc: 2.26-11

Thanks anyways, 
rsm

closed 2018-02-23 by rsm-gh