tpaviot/oce #692: Conda installation under Python 3.6 on Mac

ghareth, 2018-03-14:
I have been running pythonocc successfully installed using conda under Python 2.7 within both my Windows(32-bit Python) and MacOS (64-bit Python) environments.

Upgrading my pythonocc installations to Python 3.6, pythonocc has successfully installed in Windows (using 32-bit Python) while the MacOs installation suffers from the following error:

```
import OCC.gp 

Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
```

danZib, 2018-06-12:
Can second that, have the same problems at the moment

aothms, 2018-06-12:
You probably better report this at https://github.com/tpaviot/pythonocc-core

I also ran into this publishing builds for my own project. What I ended up doing is not to link to the python dylib at compile time and allow the linker to ignore the missing symbols. It's been years since I looked into this, but I remember this to be somewhat common practice on OS X.

* Set flags: https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.6.0/conda/build.sh#L10
* Link conditionally: https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.6.0/src/ifcwrap/CMakeLists.txt#L59

danZib, 2018-06-12:
Thanks @aothms ,

My problem arose when trying to install **pythonOCC v18.01** in a **python v3.5 environment** on miniconda3.
After running `conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1` the problem by @ghareth turned up.

The issue was fixed for me after **switching miniconda3 for miniconda2** : I still have a python v.3.5 environment, but somehow switching those fixed the issue for me...
...no idea why but maybe the same might work for you @ghareth

ghareth, 2018-06-12:
Thanks @danZib, I'll give it a try it out when I get a chance. For the time being I'm comfortable write Python 2/3 compatible code.