tpaviot/oce #712: Converting file formats

miclog, 2019-03-01:
Is there a good way to write a file converter using OCE?  Are there any existing examples to work from?

I would like to convert from STEP to X3D format.

The program SFA by NIST does a good job but might be slowed down by the Python interface and being in a self-contained executable.  I'd like to try using the C++ library directly instead of loading it from Python using PythonOCC.

Stinkfist0, 2019-03-03:
Have you looked into IfcOpenShell/IfcConvert?

tpaviot, 2019-03-04:
Have a look at https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_webgl_STEP_to_X3D.py

carpekd, 2020-06-02:
> Have a look at https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_webgl_STEP_to_X3D.py

Can this be done with OCE alone in C++?

tpaviot, 2020-06-02:
Generally speaking, I don't know any python program that could not be done using c++