tpaviot/oce #724: Add isopycnal curves of neutral density to TSplot

gianmarcoingrosso, 2020-01-23:
Hi,
I would like to add isopycnal curves of neutral density anomaly (http://www.teos-10.org/preteos10_software/neutral_density.html) to a TS plot (instead of potential density anomaly isopycnal levels, as in the figure attached).
Any idea how to do this in R and with the oce package?
In particular, I would like to draw the neutral density isopycnal 28.0 and 28.27.
Thanks in advantage for your help.
Best
Gianmarco

[image]

closed 2020-01-23 by gianmarcoingrosso

tpaviot, 2020-01-23:
You posted the wrong place, is that it ?

gianmarcoingrosso, 2020-01-23:
Yes, sorry. I try to delete the issue.