tpaviot/oce #725: pthread issues during build on EC2 instance (Amazon Linux 2)

carpekd, 2020-04-26:
I am building on an EC2 instance running Amazon Linux 2 (derived from centOS i think).

The build gets up to 99% and fails at this point:
```
[ 99%] Built target TKXDEIGES
[ 99%] Built target TKXDESTEP
[ 99%] Built target gtest
Linking CXX executable boolean_ops_test
../../Unix/x86_64-Release-64/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<testing::TestPartResultReporterInterface*>::~ThreadLocal()':
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEED2Ev[_ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEED5Ev]+0x15): undefined reference to `pthread_getspecific'
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEED2Ev[_ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEED5Ev]+0x2a): undefined reference to `pthread_key_delete'
../../Unix/x86_64-Release-64/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<std::vector<testing::internal::TraceInfo, std::allocator<testing::internal::TraceInfo> > >::~ThreadLocal()':
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEED2Ev[_ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEED5Ev]+0x15): undefined reference to `pthread_getspecific'
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEED2Ev[_ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEED5Ev]+0x2a): undefined reference to `pthread_key_delete'
../../Unix/x86_64-Release-64/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<std::vector<testing::internal::TraceInfo, std::allocator<testing::internal::TraceInfo> > >::GetOrCreateValue() const':
gtest-all.cc:(.text._ZNK7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE16GetOrCreateValueEv[_ZNK7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE16GetOrCreateValueEv]+0x18): undefined reference to `pthread_getspecific'
gtest-all.cc:(.text._ZNK7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE16GetOrCreateValueEv[_ZNK7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE16GetOrCreateValueEv]+0x7c): undefined reference to `pthread_setspecific'
../../Unix/x86_64-Release-64/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<testing::TestPartResultReporterInterface*>::CreateKey()':
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE9CreateKeyEv[_ZN7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE9CreateKeyEv]+0x16): undefined reference to `pthread_key_create'
../../Unix/x86_64-Release-64/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<std::vector<testing::internal::TraceInfo, std::allocator<testing::internal::TraceInfo> > >::CreateKey()':
gtest-all.cc:(.text._ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE9CreateKeyEv[_ZN7testing8internal11ThreadLocalISt6vectorINS0_9TraceInfoESaIS3_EEE9CreateKeyEv]+0x16): undefined reference to `pthread_key_create'
../../Unix/x86_64-Release-64/libgtest.a(gtest-all.cc.o): In function `testing::internal::ThreadLocal<testing::TestPartResultReporterInterface*>::GetOrCreateValue() const':
gtest-all.cc:(.text._ZNK7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE16GetOrCreateValueEv[_ZNK7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE16GetOrCreateValueEv]+0x16): undefined reference to `pthread_getspecific'
gtest-all.cc:(.text._ZNK7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE16GetOrCreateValueEv[_ZNK7testing8internal11ThreadLocalIPNS_31TestPartResultReporterInterfaceEE16GetOrCreateValueEv]+0x7a): undefined reference to `pthread_setspecific'
../../Unix/x86_64-Release-64/libTKernel.so.11.0.0: undefined reference to `pthread_create'
../../Unix/x86_64-Release-64/libTKernel.so.11.0.0: undefined reference to `pthread_mutexattr_destroy'
../../Unix/x86_64-Release-64/libTKernel.so.11.0.0: undefined reference to `pthread_mutex_trylock'
../../Unix/x86_64-Release-64/libTKernel.so.11.0.0: undefined reference to `pthread_mutexattr_init'
../../Unix/x86_64-Release-64/libTKernel.so.11.0.0: undefined reference to `pthread_join'
../../Unix/x86_64-Release-64/libTKernel.so.11.0.0: undefined reference to `pthread_detach'
../../Unix/x86_64-Release-64/libTKernel.so.11.0.0: undefined reference to `pthread_mutexattr_settype'
collect2: error: ld returned 1 exit status
make[2]: *** [test/BRepAlgoAPI_test/boolean_ops_test] Error 1
make[1]: *** [test/BRepAlgoAPI_test/CMakeFiles/boolean_ops_test.dir/all] Error 2
make: *** [all] Error 2
```

How do i link pthread in the Makefile?

aothms, 2020-04-29:
> How do i link pthread in the Makefile?

Generally, you can in CMakeLists.txt

~~~
find_package(Threads)
target_link_libraries(... ${CMAKE_THREAD_LIBS_INIT})
~~~

The answer here provides some additional info https://stackoverflow.com/a/29871891

carpekd, 2020-04-30:
I corrected this by installing cmake version 3.17.2 from the downloaded tar instead of using yum install cmake

closed 2020-04-30 by carpekd