tpaviot/oce #736: Maybe a memory leak in Standard.cxx

labusi, 2020-11-26:
In the Standard.cxx file, I'd like to know whether or not there should be a "delete myFMMgr" in the if block, It seems a memory leak issue.
```c++
Standard_MMgrFactory::~Standard_MMgrFactory()
{
  if (  myFMMgr )
    myFMMgr->Purge(Standard_True);
}

```

tpaviot, 2020-11-26:
Hard to say. ```cppcheck``` does not report any issue

aothms, 2020-11-26:
This is actually documented:

> // Therefore holder currently does not call destructor of the memory manager 
> // but only its method Purge() with Standard_True.

https://github.com/tpaviot/oce/blob/oce/patches/src/Standard/Standard.cxx#L203

Has to deal with construction and destruction of static definitions in different compilation units.

tpaviot, 2020-11-26:
Thank you @aothms

closed 2020-11-26 by tpaviot