tpaviot/oce #741: Issue when building - /usr/bin/ld : cannot find -lXi

Ramisay, 2021-05-16:
Hi, 
I am trying to install OCE under Ubuntu 18.04, but I got a problem during the building. 
First I cloned the directory from : git clone https://github.com/tpaviot/oce.git > /dev/null
Then I installed the 3dparty libraries. And finally when I run "cmake" it is working (even if it is not detecting the versions of Tcl and Tk) :  

```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Info. Detecting doxygen
-- Found Doxygen: /usr/bin/doxygen (found suitable version "1.8.13", minimum required is "1.8.4") found components:  doxygen 
-- Info. Doxygen is found and can be used
-- Info. Overview building is turned on
-- Info: TCL is used by OCCT
-- Found Tclsh: /usr/bin/tclsh (found version "8.6") 
-- Info: TCL version isn't found
-- Info: TK is used by OCCT
-- Info: TK version isn't found
-- Info: Freetype is used by OCCT
-- Found Freetype: /usr/lib/x86_64-linux-gnu/libfreetype.so (found version "2.8.1") 
-- Info: TKIVtk and TKIVtkDraw toolkits excluded due to VTK usage is disabled
-- Info: The directories of 3rdparty headers: 
    /usr/include/tcl
    /usr/include/freetype2
-- Info: The directories of 3rdparty libraries: 
    /usr/lib/x86_64-linux-gnu
-- 
Info: (16:40:22) Start collecting all OCCT header files into /scratch/rsayoud/Documents/Software_Packages/oce_folder/build/oce-last/inc ...
-- Info: (16:40:22) Compare FILES with files in package directories...
-- Info: (16:40:28) Create header-links in inc folder...
-- Info: (16:40:29) Checking headers in inc folder...
-- Info: (16:40:30) End the collecting
-- Info: (16:40:30) OCCT toolkits processed
-- Info: (16:40:30) OCCT configuration files prepared
-- Configuring done
-- Generating done
-- Build files have been written to: /scratch/rsayoud/Documents/Software_Packages/oce_folder/build/oce-last
```

But I have a problem when running "make" command, when I run it it starts building until 17% where it shows a linking problem : 

```
[ 17%] Building CXX object src/TKService/CMakeFiles/TKService.dir/__/Font/Font_SystemFont.cxx.o
[ 17%] Building CXX object src/TKService/CMakeFiles/TKService.dir/__/Font/Font_TextFormatter.cxx.o
[ 17%] Linking CXX shared library ../../lin64/gcc/lib/libTKG3d.so
[ 17%] Built target TKG3d
Scanning dependencies of target TKGeomBase
[ 17%] Linking CXX shared library ../../lin64/gcc/lib/libTKService.so
/usr/bin/ld : cannot find -lXi
collect2: error: ld returned 1 exit status
src/TKService/CMakeFiles/TKService.dir/build.make:2568: recipe for target 'lin64/gcc/lib/libTKService.so.7.6.0' failed
make[2]: *** [lin64/gcc/lib/libTKService.so.7.6.0] Error 1
CMakeFiles/Makefile2:1188: recipe for target 'src/TKService/CMakeFiles/TKService.dir/all' failed
make[1]: *** [src/TKService/CMakeFiles/TKService.dir/all] Error 2
```

I was wondering what could be the solution to properly build and install OCE? 
I am trying to install OEC in order to couple it with a software that I am using (Siconos).  
Thanks in advance.

aothms, 2021-05-16:
It seems you need to install `libxi-dev`. It's probably a newer dependency from a tracking a newer OCCT version that is not yet reflected in the docs.

Ramisay, 2021-05-17:
Thank you @aothms , that was the problem, I didn't have libxi-dev in my laptop. Now It's working and OCE is installed. 

In the other hand, I'm trying to install Python-occ now, and when I run the cmake commande , I get : 
```
CMake Error at CMakeLists.txt:131 (find_package):
  By not providing "FindOCE.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OCE", but
  CMake did not find one.

  Could not find a package configuration file provided by "OCE" (requested
  version 0.18) with any of the following names:

    OCEConfig.cmake
    oce-config.cmake

  Add the installation prefix of "OCE" to CMAKE_PREFIX_PATH or set "OCE_DIR"
  to a directory containing one of the above files.  If "OCE" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!

```
Is   "OCEConfig.cmake" should be generated automatically when we install OCE? or there is a way to generate it? and how to specify it when we run cmake commande during OCC building?
Thanks in advance.

Ramisay, 2021-05-17:
In fact , I solved the last problem by installing the " liboce-foundation-dev " library. and now cmake command works. 
But unfortunetly, the make commande also prevents error messages, when I run make I got : 

``` 
[  0%] Swig source
[  0%] Swig source
[  0%] Swig source
[  0%] Swig source
[  0%] Swig source
[  0%] Swig source
[  0%] Swig source
[  0%] Swig source
Scanning dependencies of target _UTL
[  0%] Building CXX object CMakeFiles/_UTL.dir/Unix/x86_64-Release-64/UTLPYTHON_wrap.cxx.o
/scratch/rsayoud/Documents/Software_Packages/oce_folder/build/pyocc/Unix/x86_64-Release-64/UTLPYTHON_wrap.cxx:3788:9: fatal error: UTL.hxx: Aucun fichier ou dossier de ce type
 #include<UTL.hxx>
         ^~~~~~~~~
compilation terminated.
CMakeFiles/_UTL.dir/build.make:71: recipe for target 'CMakeFiles/_UTL.dir/Unix/x86_64-Release-64/UTLPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_UTL.dir/Unix/x86_64-Release-64/UTLPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/_UTL.dir/all' failed
make[1]: *** [CMakeFiles/_UTL.dir/all] Error 2
make[1]: *** Attente des tâches non terminées....
Scanning dependencies of target _CDF
Scanning dependencies of target _PCDM
[  0%] Building CXX object CMakeFiles/_CDF.dir/Unix/x86_64-Release-64/CDFPYTHON_wrap.cxx.o
[  0%] Building CXX object CMakeFiles/_PCDM.dir/Unix/x86_64-Release-64/PCDMPYTHON_wrap.cxx.o
/scratch/rsayoud/Documents/Software_Packages/oce_folder/build/pyocc/Unix/x86_64-Release-64/CDFPYTHON_wrap.cxx:3924:9: fatal error: CDF_TypeOfActivation.hxx: Aucun fichier ou dossier de ce type
 #include<CDF_TypeOfActivation.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/_CDF.dir/build.make:71: recipe for target 'CMakeFiles/_CDF.dir/Unix/x86_64-Release-64/CDFPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_CDF.dir/Unix/x86_64-Release-64/CDFPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/_CDF.dir/all' failed
make[1]: *** [CMakeFiles/_CDF.dir/all] Error 2
/scratch/rsayoud/Documents/Software_Packages/oce_folder/build/pyocc/Unix/x86_64-Release-64/PCDMPYTHON_wrap.cxx:3924:9: fatal error: PCDM_ReadWriter.hxx: Aucun fichier ou dossier de ce type
 #include<PCDM_ReadWriter.hxx>
         ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/_PCDM.dir/build.make:71: recipe for target 'CMakeFiles/_PCDM.dir/Unix/x86_64-Release-64/PCDMPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_PCDM.dir/Unix/x86_64-Release-64/PCDMPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:215: recipe for target 'CMakeFiles/_PCDM.dir/all' failed
make[1]: *** [CMakeFiles/_PCDM.dir/all] Error 2
Scanning dependencies of target _TDF
[  0%] Building CXX object CMakeFiles/_TDF.dir/Unix/x86_64-Release-64/TDFPYTHON_wrap.cxx.o
/scratch/rsayoud/Documents/Software_Packages/oce_folder/build/pyocc/Unix/x86_64-Release-64/TDFPYTHON_wrap.cxx:4013:9: fatal error: TDF_AttributeList.hxx: Aucun fichier ou dossier de ce type
 #include<TDF_AttributeList.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/_TDF.dir/build.make:71: recipe for target 'CMakeFiles/_TDF.dir/Unix/x86_64-Release-64/TDFPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_TDF.dir/Unix/x86_64-Release-64/TDFPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:326: recipe for target 'CMakeFiles/_TDF.dir/all' failed
make[1]: *** [CMakeFiles/_TDF.dir/all] Error 2
Scanning dependencies of target _TDataStd
[  0%] Building CXX object CMakeFiles/_TDataStd.dir/Unix/x86_64-Release-64/TDataStdPYTHON_wrap.cxx.o
/scratch/rsayoud/Documents/Software_Packages/oce_folder/build/pyocc/Unix/x86_64-Release-64/TDataStdPYTHON_wrap.cxx:4055:9: fatal error: TDataStd_DataMapOfStringHArray1OfInteger.hxx: Aucun fichier ou dossier de ce type
 #include<TDataStd_DataMapOfStringHArray1OfInteger.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/_TDataStd.dir/build.make:71: recipe for target 'CMakeFiles/_TDataStd.dir/Unix/x86_64-Release-64/TDataStdPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_TDataStd.dir/Unix/x86_64-Release-64/TDataStdPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:289: recipe for target 'CMakeFiles/_TDataStd.dir/all' failed
make[1]: *** [CMakeFiles/_TDataStd.dir/all] Error 2
Scanning dependencies of target _IGESCAFControl
[  0%] Building CXX object CMakeFiles/_IGESCAFControl.dir/Unix/x86_64-Release-64/IGESCAFControlPYTHON_wrap.cxx.o
/scratch/rsayoud/Documents/Software_Packages/oce_folder/build/pyocc/Unix/x86_64-Release-64/IGESCAFControlPYTHON_wrap.cxx:4068:9: fatal error: IGESCAFControl_Reader.hxx: Aucun fichier ou dossier de ce type
 #include<IGESCAFControl_Reader.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/_IGESCAFControl.dir/build.make:71: recipe for target 'CMakeFiles/_IGESCAFControl.dir/Unix/x86_64-Release-64/IGESCAFControlPYTHON_wrap.cxx.o' failed
[... 6 lines cut ...]
 #include<XCAFPrs_DataMapOfStyleShape.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/_XCAFPrs.dir/build.make:71: recipe for target 'CMakeFiles/_XCAFPrs.dir/Unix/x86_64-Release-64/XCAFPrsPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_XCAFPrs.dir/Unix/x86_64-Release-64/XCAFPrsPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/_XCAFPrs.dir/all' failed
make[1]: *** [CMakeFiles/_XCAFPrs.dir/all] Error 2
Scanning dependencies of target _STEPCAFControl
[... cut ...]
CMakeFiles/_STEPCAFControl.dir/build.make:71: recipe for target 'CMakeFiles/_STEPCAFControl.dir/Unix/x86_64-Release-64/STEPCAFControlPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_STEPCAFControl.dir/Unix/x86_64-Release-64/STEPCAFControlPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/_STEPCAFControl.dir/all' failed
make[1]: *** [CMakeFiles/_STEPCAFControl.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

``` 
I have python3.6 on my computer, with oce and swig installed. 
I am wondering what could be the problem in this case. 

Thanks in advance.

Ramisay, 2021-05-18:
I don't know if it could be a problem of cmake version or not, cause I'm using : 
 cmake --version
cmake version 3.10.2
is it ok for pyocc? 

Best Regards.