tpaviot/oce #743: BRepOffsetAPI_MakePipeShell 构建放样实体存在问题

MonoKelvin, 2021-05-31:
放样存在问题：

```cpp
TopoDS_Wire BuildPath()
{
    const double dRadius = 5;
    const double dStartAngle = 0;
    const double dEndAngle = M_PI;
    const gp_Pnt center(0, 0, 0);

    // 直线
    const gp_Pnt p1 = center.XYZ();
    const gp_Pnt p2 = center.XYZ() + gp_XYZ(dRadius, 0, 0);
    const TopoDS_Edge lineEdge = BRepBuilderAPI_MakeEdge(p1, p2);
	
    // 圆弧
    const gp_Dir zDir(0., 0., 1.);
    const gp_Dir xDir(1., 0., 0.);
    const gp_Ax2 axis(center, zDir, xDir);
    const gp_Circ circ(axis, dRadius);
    Handle(Geom_Circle) C = new Geom_Circle(circ);
    Handle(Geom_TrimmedCurve) arc = new Geom_TrimmedCurve(C, dStartAngle, dEndAngle);
    const TopoDS_Edge arcEdge = BRepBuilderAPI_MakeEdge(arc);

    BRepBuilderAPI_MakeWire mw;
    mw.Add(lineEdge);
    mw.Add(arcEdge);
    return mw.Wire();
}

TopoDS_Wire BuildProfile()
{
    const double dRadius = 1;
    const gp_Pnt center(0, 0, 0);
    const gp_Dir zDir(1., 0., 0.);
    const gp_Dir xDir(0., -1., 0.);
    const gp_Ax2 axis(center, zDir, xDir);
    const gp_Circ circ(axis, dRadius);
    const TopoDS_Edge arcEdge = BRepBuilderAPI_MakeEdge(circ);
    return BRepBuilderAPI_MakeWire(arcEdge);
}

void Test1()
{
    try 
    {
        TopoDS_Shape path, profile, result;

        path = BuildPath();
        profile = BuildProfile();

        TopoDS_Wire aPathWire = TopoDS::Wire(path);
        TopoDS_Wire aProfileWire = TopoDS::Wire(profile);

        BRepOffsetAPI_MakePipeShell mps(aPathWire);
        mps.Add(aProfileWire);
        mps.SetTransitionMode(BRepBuilderAPI_RightCorner);
        mps.Build();

        result = mps.Shape();
    }
    catch (...)
    {
    }
}
```

我向上面这样构建一个由直线和圆弧组合的路径以及圆形轮廓，使用BRepOffsetAPI_MakePipeShell放样时结果出错。我调试过源码，发现是在IntPatch_ImpPrmIntersection求取实体交线时得到的交线有错误。

MonoKelvin, 2021-09-15:
我后来又测试过了，交线没有错误但对于该算法应该要“打断”，否则在BRepFill_TrimShellCorner::MakeFacesSec方法中筛选不到正确的界面轮廓线。

MonoKelvin, 2021-09-15:
我尝试各种方法将该交线在正确的位置打断后，居然可以正常运行并且得到正确的结果。但测试更多的案例后，我发现存在其它各种小问题，如由于精度导致Edge和Edge求交没有交点（IntTools_EdgeEdge::ComputeLineLine），曲线到曲面的距离求错等，我觉得这不是我改正交线而出现的问题，而是OCC算法中本来就存在的。以下是我对面面交线算法改进的修改，这个改动只针对BRepOffsetAPI_MakePipeShell，所以在打断交线前使用IntTools_FaceFace::GetIsSplitCurves()判断是否需要打断，像布尔运算和Section就不需要打断。
```cpp
// Created on: 2000-11-23
// Created by: Michael KLOKOV
// Copyright (c) 2000-2014 OPEN CASCADE SAS
//
// This file is part of Open CASCADE Technology software library.
//
// This library is free software; you can redistribute it and/or modify it under
// the terms of the GNU Lesser General Public License version 2.1 as published
// by the Free Software Foundation, with special exception defined in the file
// OCCT_LGPL_EXCEPTION.txt. Consult the file LICENSE_LGPL_21.txt included in OCCT
// distribution for complete text of the license and disclaimer of any warranty.
//
// Alternatively, this file may be used under the terms of Open CASCADE
// commercial license or contractual agreement.

#include <IntTools_FaceFace.hxx>

#include <BRepAdaptor_Surface.hxx>
#include <BRepTools.hxx>
#include <BRep_Tool.hxx>
#include <ElCLib.hxx>
#include <ElSLib.hxx>
#include <Geom2dAdaptor_Curve.hxx>
#include <Geom2dInt_GInter.hxx>
#include <Geom2d_Line.hxx>
#include <Geom2d_TrimmedCurve.hxx>
#include <GeomAPI_ProjectPointOnSurf.hxx>
#include <GeomAdaptor_HSurface.hxx>
#include <GeomInt_IntSS.hxx>
#include <GeomInt_WLApprox.hxx>
#include <GeomLib_Check2dBSplineCurve.hxx>
#include <GeomLib_CheckBSplineCurve.hxx>
#include <Geom_BSplineCurve.hxx>
#include <Geom_Circle.hxx>
#include <Geom_Ellipse.hxx>
#include <Geom_Hyperbola.hxx>
#include <Geom_Line.hxx>
#include <Geom_OffsetSurface.hxx>
#include <Geom_Parabola.hxx>
#include <Geom_RectangularTrimmedSurface.hxx>
#include <Geom_TrimmedCurve.hxx>
#include <IntAna_QuadQuadGeo.hxx>
#include <IntPatch_GLine.hxx>
#include <IntPatch_RLine.hxx>
#include <IntPatch_WLine.hxx>
#include <IntRes2d_Domain.hxx>
#include <IntSurf_Quadric.hxx>
#include <IntTools_Context.hxx>
#include <IntTools_Tools.hxx>
#include <IntTools_TopolTool.hxx>
#include <IntTools_WLineTool.hxx>
#include <ProjLib_Plane.hxx>
#include <TopExp_Explorer.hxx>
#include <TopoDS.hxx>
#include <TopoDS_Edge.hxx>
#include <gp_Elips.hxx>
#include <GeomAPI_ExtremaCurveCurve.hxx>
#include <GeomLib_Tool.hxx>
#include <BRepBuilderAPI_MakeEdge.hxx>
#include <map>
#include <algorithm>
#include <GCPnts_AbscissaPoint.hxx>
#include <BRepLProp_CLProps.hxx>
#include <GeomAPI_ProjectPointOnCurve.hxx>
#include <IntTools_Range.hxx>

// 用于标识曲线是否打断过
static std::map<Standard_Integer, Standard_Boolean> splittedMap;
[... 3532 lines cut ...]
//purpose  : 
//=======================================================================
 void CorrectPlaneBoundaries(Standard_Real& aUmin,
                             Standard_Real& aUmax, 
                             Standard_Real& aVmin, 
                             Standard_Real& aVmax) 
{
  if (!(Precision::IsInfinite(aUmin) ||
        Precision::IsInfinite(aUmax))) { 
    Standard_Real dU;
    //
    dU=0.1*(aUmax-aUmin);
    aUmin=aUmin-dU;
    aUmax=aUmax+dU;
  }
  if (!(Precision::IsInfinite(aVmin) ||
        Precision::IsInfinite(aVmax))) { 
    Standard_Real dV;
    //
    dV=0.1*(aVmax-aVmin);
    aVmin=aVmin-dV;
    aVmax=aVmax+dV;
  }
}
```

closed 2021-12-07 by MonoKelvin

ShenWang6688, 2024-07-10:
我也遇到了这个问题，尝试用了三种方式进行扫掠，结果都不太令人满意，想问下您这部分是怎么解决的呢
[image]

ShenWang6688, 2024-07-10:
[image]
[image]