tpaviot/oce #745: [question] is this repo still maintained? (the official OCCT seems to be ahead)

kcris, 2021-12-09:
Hi,

is this repository still maintained?

For a while this used to be ahead of the official OCCT repository.

But now it's quite the opposite:
* there are no updates here
* the official OCCT has released version [7.6.0](https://github.com/Open-Cascade-SAS/OCCT/releases/tag/V7_6_0)

Thank you

bartoszek, 2021-12-12:
Bump the questing, since `oneTBB` drops `tbb::atomic` in favor of `std::atomic` this package quickly becomes obsolescent as more ad more distros moved to the new tbb API.

tpaviot, 2021-12-13:
No, this package is currently unmaintained. It used to live thanks to user contributions, but since the occ team became more open to the community (lgpl license, git repos, open mantis issue tacker) the oce project is less useful. I suggest using the latest official occt version rather than once.